atar-axis/xpadneo #455: xpadneo module not loading correctly when connecting Bluetooth Controller

BoomSaag, 2024-01-28:
## Version of xpadneo
v0.9-144-g9b3b696

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [x] Other: Gulikit KingKong2 Pro

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [x] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [x] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [x] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [x] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [x] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [x] It does not work at all
- [x] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

I use my Gulikit KingKong2 Pro for retro gaming in RetroPie. I previously used xpadneo v0.9.5 which is available under the driver modules in RetroPie Setup. This version has the issue with the constant rumble.
I was told this issue was fixed in the Git version. I deleted the version from RetroPie Setup and installed the git version per the instructions on Github.
Now the controller pairs but does not work. It seems there is an issue loading the module as when the controller is connected, there is no hid_xpadneo module loaded under /sys/module/. Only hid and hidp is present.

modinfo output:
```
filename:       /lib/modules/5.10.103-v7l+/kernel/drivers/hid/hid-xpadneo.ko
version:        v0.9-144-g9b3b696
description:    Linux kernel driver for Xbox ONE S+ gamepads (BT), incl. FF
author:         Kai Krakow <kai@kaishome.de>
author:         Florian Dollinger <dollinger.florian@gmx.de>
license:        GPL
srcversion:     8CA1B882F631A86C86D0009
alias:          hid:b0005g*v0000045Ep00000B13
alias:          hid:b0005g*v0000045Ep00000B22
alias:          hid:b0005g*v0000045Ep00000B05
alias:          hid:b0005g*v0000045Ep00000B20
alias:          hid:b0005g*v0000045Ep000002E0
alias:          hid:b0005g*v0000045Ep000002FD
depends:        ff-memless
name:           hid_xpadneo
vermagic:       5.10.103-v7l+ SMP mod_unload modversions ARMv7 p2v8 
parm:           trigger_rumble_mode:(u8) Trigger rumble mode. 0: pressure, 2: disable. (byte)
parm:           rumble_attenuation:(u8) Attenuate the rumble strength: all[,triggers] 0 (none, full rumble) to 100 (max, no rumble). (array of byte)
parm:           ff_connect_notify:(bool) Connection notification using force feedback. 1: enable, 0: disable. (bool)
parm:           gamepad_compliance:(bool) Adhere to Linux Gamepad Specification by using signed axis values. 1: enable, 0: disable. (bool)
parm:           disable_deadzones:(bool) Disable dead zone handling for raw processing by Wine/Proton, confuses joydev. 0: disable, 1: enable. (bool)
parm:           quirks:(string) Override or change device quirks, specify as: "MAC1{:,+,-}quirks1[,...16]", MAC format = 11:22:33:44:55:66, no pulse parameters = 1, no trigger rumble = 2, no motor masking = 4, use Linux button mappings = 16, use Nintendo mappings = 32, use Share button mappings = 64, reversed motor masking = 128 (array of charp)

```

## Steps to Reproduce

install the git version as per instruction on github.
pair and connect controller as per instruction on github.

## Expected Behavior

Controller connects via bluetooth and xpadneo is successfully loaded

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a

Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/14075494/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/14075495/xpadneo-dmesg.txt)

## Additional Context

kakra, 2024-01-28:
Did you reboot the system? This sounds like the module aliases and/or udev rules may not be loaded properly.

BoomSaag, 2024-01-28:
Yes, I have rebooted a number of times. I also installed and reinstalled the driver a few times. Same result.

BoomSaag, 2024-01-28:
So I tried installing the xpadneo on my other sdcard with a cleaner RaspberryPi OS installation.
And it seems to work fine. The controller pairs, rumbles about three times, then stops.
Running `jstest` shows that the input events all register. I do not have any games on that installation as it is the sdcard I insert when my kids have to use the Pi for their school projects.

```
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
```

Thus I am thinking something is messed up on my RetroPie sdcard.
I had xpadneo v0.9.5 installed from Retro-Pie Setup. This morning I uninstalled it and installed xpadneo from git. Then the issue started. If I uninstall the git version and install the RetroPie Setup version again, the problem is still the same.
I am only able to use the controller if I completely remove xpadneo drivers by uninstalling.

Can I somehow do I "fresh" install of xpadneo? Remove every trace of it? Also reset the udev rules?

BoomSaag, 2024-01-28:
I managed to get it to work.
This seemed to be the issue:
`[  530.435937] hid_xpadneo: `sion' invalid for parameter `rumble_attenuation'`

under the `~/xpadneo/' directory I ran 'configure.sh -r 0'. I set the rumble_attenuation to 0.
rebooted and removed the device via `bluetoothctl` and paired it from fresh.

Started 'emulationstation' and it detected the controller. Tested in game, the rumble at first did not work, but I then just fiddled with the rumble setting on the controller (see Gulikit KingKong2 Pro Manual) then the rumble worked.

I do not know what messed up the rumble attenuation, but it is fixed now.

kakra, 2024-01-28:
> I do not know what messed up the rumble attenuation, but it is fixed now.

Me neither. Maybe an artifact of an old version of the configure script caused it which had a bug.

closed 2024-01-28 by kakra

defanator, 2024-09-27:
@BoomSaag I ran into a similar issue on the same kernel version (5.10.103) booted from RetroPie image. XBox controller was entering into pairing mode, but no devices were detected with `scan on` on my Pi (it's Raspberry Pi 2 model B).

USB Bluetooth adapter I'm using there is TP-Link UB400.

I tried to re-flash SD card with latest "legacy" Debian (bullseye) using Raspberry Pi imager. The kernel version in that one is 6.1.21, and now I'm seeing all bluetooth devices in scan, as well as XBox controller. I have not installed xpadneo yet, just trying to confirm that the issue is in older kernel + UB400 combination.

Posting it here for future references + search engines thing (hello future me from the past!)

@kakra thanks for the project once again! I barely remember I was setting it up successfully on the same RPi 2 few years back, so hopefully I'll make it fly again :)