atar-axis/xpadneo #459: xpadneo 0.9.5 sticks not working properly for RPCS3 on Ubuntu 22.04 Linux kernel 6.5.0

ipkpjersi, 2024-02-12:
## Version of xpadneo
v0.9.5

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [x] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've tried disabling or running without above mentioned software
- [x] It does not work at all
- [x] It used to work in a previous version

## Describe the Bug
The sticks do not work properly, I cannot move around. For example, all the buttons work, but in Skate 3 in RPCS3, the sticks do not behave as expected.

## Steps to Reproduce

Install Ubuntu 22.04, install RPCS3 and xpadneo 0.9.5 and test the sticks - for example in the controller settings, you will see the left stick seems to be far outside of the bounds of the deadzones etc. I recall having this bug before I think.

## Expected Behavior

I expect RPCS3 with my Xbox Series X controller and xpadneo to work for me as expected.

## Screenshots / GIFs / Videos
Here is a picture of RPCS3, check out where the left stick is:
[image]

## System Information

```console
# uname -a
Linux fast-desktop 6.5.0-17-generic #17~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Jan 16 14:32:32 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0

```

## Controller and Bluetooth Information

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/14235180/xpadneo-lsusb.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/14235181/xpadneo-dmesg.txt)

[xpadneo-btmon.txt.gz](https://github.com/atar-axis/xpadneo/files/14235179/xpadneo-btmon.txt.gz)

If you require any additional information, please let me know.

kakra, 2024-02-12:
> The sticks do not work properly, I cannot move around. For example, all the buttons work, but in Skate 3 in RPCS3, the sticks do not behave as expected. Also, when colliding with objects in-game, there is randomly force feedback/rumbling despite having trigger_rumble_mode=2 disable_ff=3 in my /etc/modprobe.d/99-xpadneo-options.conf

`disable_ff` is no longer an option and has been deprecated before, use `rumble_attenuation=100` to dampen rumble effects by 100%. In this case, `trigger_rumble_mode` is no longer needed. Actually, your logs above indicate that xpadneo isn't loaded, and looking into your dmesg log, this is confirmed: The device is taken bei `hid-microsoft`, not `hid-xpadneo`.

> options hid_xpadneo rumble_attenuation=100 trigger_rumble_damping=1 trigger_rumble_mode=2 combined_z_axis=n ff_connect_notify=0 disable_ff=3

Please remove all those funny and creative options: There's no `trigger_rumble_damping` (deprecated before), and no `disable_ff` (deprecated before), this has been combined into `rumble_attenuation=100` (or `0,100` is you want to dampen only trigger rumble).

If it still rumbles after this, RPCS3 is probably using SDL2 for your gamepad and has access to the hidraw device, although it says "evdev" in the screenshot.

Also, check that you actually see a line like this in dmesg before reporting wrong behavior:
```
[ 5880.377281] xpadneo 0005:045E:0B13.0009: input,hidraw7: BLUETOOTH HID v5.07 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:15
#               ^^^^^^^^ - instead of "microsoft"
```

If it doesn't properly bind to xpadneo, your udev rules are not installed or not working, or `hid-microsoft` is statically compiled into the kernel.

Did you ever tried calibrating the stick? If yes, please remove that setting. The sticks are auto-calibrating.

ipkpjersi, 2024-02-12:
Hi,

I've removed the extra options in the config:

```
  cat /etc/modprobe.d/99-xpadneo-options.conf
options hid_xpadneo rumble_attenuation=100
```

You are right, it looks like RPCS3 is using SDL, I changed it from Evdev to SDL in the RPCS3 settings and now my controller works.

With that said, I find it odd that it's using hid-microsoft considering I don't even know what that is, and also dkms status reports this:
```
  dkms status
hid-xpadneo/v0.9.5, 6.5.0-15-generic, x86_64: installed
hid-xpadneo/v0.9.5, 6.5.0-17-generic, x86_64: installed
```

I know that Linux version 6.5 was particularly destructive considering all the posts on NVIDIA forums about all the NVIDIA driver issues with version 6.5 (I also had some issues with that but that's a different story) so it wouldn't surprise me if Linux 6.5 also broke xpadneo somehow. I think they might have changed which version of GCC is being used but maybe I'm mistaken.

I upgraded my kernel same as usual and dkms should have taken care of xpadneo with this kernel version too automatically like it did for all my kernel updates in the past.

I do see this after a fresh reboot (I'm pretty sure a reboot clears dmesg?):
```
  sudo dmesg | grep xpadneo
[   20.425748] loaded hid-xpadneo v0.9.5
```

Is there anything else I could check?

kakra, 2024-02-12:
> With that said, I find it odd that it's using hid-microsoft considering I don't even know what that is, and also dkms status reports this:

`hid-microsoft` is part of the kernel and supports Bluetooth Xbox controllers. This is what SDL2 is fully compatible with, there are some issues with wrong buttons under certain circumstances with SDL2 and xpadneo.

> [   20.425748] loaded hid-xpadneo v0.9.5

It just means the module was loaded, you need device specific messages, too.

> It looks like the right stick activates the triggers in Evdev in RPCS3

There's an extra axis combining both triggers, used for simulators. I'll remove that with v0.10 (or at least make it opt-in). It is problematic, meanwhile I think this should be better handled in user-space, the kernel is not supposed to work around user-space behavior. Maybe RPCS3 confuses that for a right axis.

> so it wouldn't surprise me if Linux 6.5 also broke xpadneo somehow

There's no known problem with kernel 6.5, nothing in the input APIs changed that would affect xpadneo. Actually, I'm using it with kernel 6.6 here. It's more likely that something with your udev rule setup is broken because the driver doesn't bind to the device. If this doesn't work, the driver hasn't been bound to the device:
```bash
ls -ald /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
```

You can instead use
```bash
ls -ald /sys/module/*/drivers/*/0005:045E:*/report_descriptor
```
to see which module cares about the device.

Using SDL2 probably uses rawhid, in which case you don't want to use xpadneo at all, or disable the SDL2 HID API. If Steam is loaded at the same time, its Steam Input virtual controller may steal inputs from the original device. A later xpadneo version will deal with the remaining SDL2 hidraw incompatibilites, until then disable HID API.

ipkpjersi, 2024-02-12:
```
 ls -ald /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
```
```
 ls -ald /sys/module/*/drivers/*/0005:045E:*/report_descriptor
-r--r--r-- 1 root root 4096 Feb 12 09:00 /sys/module/hid_microsoft/drivers/hid:microsoft/0005:045E:0B13.000C/report_descriptor
```

How would I go about disabling HID API? Isn't that something that could possibly have side effects for other devices (I am using a USB keyboard and GPX-SL mouse) I'm using on my PC? Couldn't I just blacklist hid_microsoft so it would be forced to use hid_xpadneo? Or I guess the udev rules are supposed to take care of that?

Is there anything else you'd like me to check?

kakra, 2024-02-12:
Look in `/usr/lub/udev/rules.d` if you installed via package manager, otherwise the rules should be in `/etc/udev/rules.d`. You should find these files: https://github.com/atar-axis/xpadneo/tree/master/hid-xpadneo/etc-udev-rules.d

You can disable HID API with `export SDL_JOYSTICK_HIDAPI=0` in your logon profile. This is distribution specific, see you distribution docs how to set environment variables in your profile or globally. This should have no impact on your keyboard and mouse, SDL doesn't handle those (or if it could, games would probably ignore that).

Blacklisting `hid-microsoft` can only be done if it isn't compiled into the kernel. But rebinding should just work, it shouldn't matter.

ipkpjersi, 2024-02-12:
I think I found the issue.

`/etc/udev/rules.d` did not have 50-xpadneo-fixup-steamlink.rules or 60-xpadneo.rules because I was using a pre-6.5 (or even pre 6.x) kernel, and I continually upgraded my kernel without ever doing a clean install.

Just now I did an uninstall and install of xpadneo 0.9.5 which installed the udev rules, and now xpadneo loads properly as expected and also respects rumble_attenuation=100.

closed 2024-02-12 by ipkpjersi

ipkpjersi, 2024-03-02:
Just for future reference: I also had to add ff_connect_notify=0 to my `/etc/modprobe.d/99-xpadneo-options.conf` so it now looks like this: 
```
cat /etc/modprobe.d/99-xpadneo-options.conf
options hid_xpadneo rumble_attenuation=100 ff_connect_notify=0
```

I also found it interesting that turning off the controller, doing rmmod hid_xpadneo then doing modprobe hid_xpadneo then turning the controller back on was not enough, I had to reboot my computer to get my `/etc/modprobe.d/99-xpadneo-options.conf` ff_connect_notify setting change to be picked up.