atar-axis/xpadneo #461: Controller rumble has stopped working

RobotRoss, 2024-02-18:
## Version of xpadneo

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ X] Other: Gulikit KingKong 2 Pro

### Connection mode

- [ X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ Xz] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ X] Steam Proton games are having issues
- [ X] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [X ] I don't use Lutris, Bottles, wine or did not try
- [ X] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ X] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X ] I've read the docs and the bug reporting instructions
- [ X] I've applied the latest firmware update to the controller
- [ X] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ X] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

Gamepad rumble no longer seems to work as of the past day or so. Not sure what's changed - haven't done any system updates that might cause it to stop working. Tried a clean reinstall of xpadneo using the latest git, same problem.

Rumble support is advertised, but does not work.

## Steps to Reproduce

Connect controller via Bluetooth, test rumble in a game or via gamepad tester.

## Expected Behavior

Rumble to work.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
#Linux fedora 6.7.4-200.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Feb  5 22:21:14 UTC 2024 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

```

## Controller and Bluetooth Information

## Additional Context

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/14324332/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/14324333/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/14324334/xpadneo-lsusb.txt)

kakra, 2024-02-18:
Try `ff_connect_notify=0` as a module parameter, there's a race condition in SDL2 for games using the wrong API for rumble detection. Or turn your controller on before starting the game. If you're using Steam Input, try turning it off. There have been reports where rumble doesn't work when Steam Input is enabled (not limited to xpadneo).

kakra, 2024-02-18:
Closing as duplicate of https://github.com/atar-axis/xpadneo/issues/444

RobotRoss, 2024-02-19:
Sorry - this is an issue separate to that. Rumble no longer works regardless of the status of that setting

kakra, 2024-02-22:
> Sorry - this is an issue separate to that. Rumble no longer works regardless of the status of that setting. The controller is fully charged. Rumble works on the controller when plugged into the computer (i.e. not through xpadneo)

@RobotRoss Can you try the steps outlined in https://github.com/atar-axis/xpadneo/issues/462#issuecomment-1960482157 (hidraw tester), and reply back here?

RobotRoss, 2024-02-23:
I think I may have randomly stumbled into what may be causing the issue, seems like xpadneo isn't always taking control of the controller, so the generic kernel driver is used instead (which doesn't support rumble)

dmesg output when rumble doesn't work:
```
[10271.305322] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[10271.305327] Bluetooth: HIDP socket layer initialized
[10271.305863] hid-generic 0005:045E:02E0.0012: unknown main item tag 0x0
[10271.306027] input: GuliKit Controller XW as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:09:00.1/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0012/input/input51
[10271.306234] hid-generic 0005:045E:02E0.0012: input,hidraw17: BLUETOOTH HID v9.03 Gamepad [GuliKit Controller XW] on 50:eb:71:f8:ce:b4
[10271.356214] microsoft 0005:045E:02E0.0012: unknown main item tag 0x0
[10271.356317] input: GuliKit Controller XW as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:09:00.1/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0012/input/input52
[10271.356433] microsoft 0005:045E:02E0.0012: input,hidraw17: BLUETOOTH HID v9.03 Gamepad [GuliKit Controller XW] on 50:eb:71:f8:ce:b4
[10271.357929] loaded hid-xpadneo 6
[10274.597600] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input53
```

dmesg when rumble does work:
```
[10732.280315] microsoft 0005:045E:02E0.0013: unknown main item tag 0x0
[10732.280477] input: GuliKit Controller XW as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:09:00.1/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0013/input/input54
[10732.280721] microsoft 0005:045E:02E0.0013: input,hidraw17: BLUETOOTH HID v9.03 Gamepad [GuliKit Controller XW] on 50:eb:71:f8:ce:b4
[10732.300156] xpadneo 0005:045E:02E0.0013: buggy firmware detected, please upgrade to the latest version
[10732.300159] xpadneo 0005:045E:02E0.0013: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[10732.300161] xpadneo 0005:045E:02E0.0013: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[10732.300164] xpadneo 0005:045E:02E0.0013: report descriptor size: 307 bytes
[10732.300165] xpadneo 0005:045E:02E0.0013: fixing up report descriptor size
[10732.300226] xpadneo 0005:045E:02E0.0013: battery detected
[10732.300228] xpadneo 0005:045E:02E0.0013: enabling compliance with Linux Gamepad Specification
[10732.300267] input: GuliKit Controller XW as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:09:00.1/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0013/input/input55
[10732.300390] xpadneo 0005:045E:02E0.0013: input,hidraw17: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on 50:eb:71:f8:ce:b4
[10732.300394] xpadneo 0005:045E:02E0.0013: controller quirks: 0x00000083
[10732.300396] xpadneo 0005:045E:02E0.0013: GuliKit Controller XW [98:b6:ea:43:62:34] connected
```

kakra, 2024-02-24:
> ```
> [  288.007132] xpadneo 0005:045E:02FD.0004: controller quirks: 0x00000095
> ```

Try `rmmod hid-xpadneo && modprobe hid-xpadneo quirks=98:b6:ea:40:7b:c0-128`, it should then say `0x00000015` for quirk flags in dmesg. Retry if your controller does rumble now. Maybe they fixed something with the motor mask in their firmwares.

If this works, try the `hidraw` test program to see if the motor masking bits (buttons 1-4) actually match the indicated motors. With early versions of OUI 98:b6:ea we had those bits flipped reverse.

Maybe also try `... quirks=98:b6:ea:40:7b:c0-132` and `...  quirks=98:b6:ea:40:7b:c0-4`. I just discovered I may have introduced a bug with one of the latest commits.

kakra, 2024-02-24:
Please re-open, if the latest commit doesn't fix your problem.

RobotRoss, 2024-02-29:
The bug hasn't been patched by the latest commit.

Vibration in hidraw works fine, but game rumble doesn't work. Occasionally game rumble might work, but I'd say that there's a 90% chance when connecting the controller that it doesn't.

Rumble works consistently on a Dualshock 4 that I also have over bluetooth, and works in games okay.

Another couple dmesg samples:
```
[98944.350298] microsoft 0005:045E:02E0.001C: unknown main item tag 0x0
[98944.350451] input: GuliKit Controller XW as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:09:00.1/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.001C/input/input80
[98944.351208] microsoft 0005:045E:02E0.001C: input,hidraw17: BLUETOOTH HID v9.03 Gamepad [GuliKit Controller XW] on 50:eb:71:f8:ce:b4
[98944.383844] xpadneo 0005:045E:02E0.001C: buggy firmware detected, please upgrade to the latest version
[98944.383848] xpadneo 0005:045E:02E0.001C: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[98944.383851] xpadneo 0005:045E:02E0.001C: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[98944.383854] xpadneo 0005:045E:02E0.001C: report descriptor size: 307 bytes
[98944.383856] xpadneo 0005:045E:02E0.001C: fixing up report descriptor size
[98944.383940] xpadneo 0005:045E:02E0.001C: battery detected
[98944.383942] xpadneo 0005:045E:02E0.001C: enabling compliance with Linux Gamepad Specification
[98944.384009] input: GuliKit Controller XW as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:09:00.1/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.001C/input/input81
[98944.384220] xpadneo 0005:045E:02E0.001C: input,hidraw17: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on 50:eb:71:f8:ce:b4
[98944.384226] xpadneo 0005:045E:02E0.001C: controller quirks: 0x00000083
[98944.384228] xpadneo 0005:045E:02E0.001C: GuliKit Controller XW [98:b6:ea:43:62:34] connected
[98947.732519] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input82
```

```
[13138.164602] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[13138.164607] Bluetooth: HIDP socket layer initialized
[13138.165056] hid-generic 0005:045E:02E0.0012: unknown main item tag 0x0
[13138.165168] input: GuliKit Controller XW as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:09:00.1/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0012/input/input51
[13138.165334] hid-generic 0005:045E:02E0.0012: input,hidraw17: BLUETOOTH HID v9.03 Gamepad [GuliKit Controller XW] on 50:eb:71:f8:ce:b4
[13138.205298] microsoft 0005:045E:02E0.0012: unknown main item tag 0x0
[13138.205397] input: GuliKit Controller XW as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:09:00.1/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0012/input/input52
[13138.205490] microsoft 0005:045E:02E0.0012: input,hidraw17: BLUETOOTH HID v9.03 Gamepad [GuliKit Controller XW] on 50:eb:71:f8:ce:b4
[13138.206776] loaded hid-xpadneo 6
[13143.056812] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input53
[13222.335680] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input54
[13535.936438] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input55
[13579.811228] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input56
```

kakra, 2024-02-29:
> ```
> [13143.056812] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input53
> [13222.335680] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input54
> [13535.936438] input: [...]

This looks like Steam Input is taking over the controller. But this looks the same in the other log, so really the problem is with `hid-microsoft` attaching to the device instead of `hid-xpadneo`.

How did you install xpadneo? Did you use direct dkms install from source repository, or are you using a distribution package?

Ensure that these files (https://github.com/atar-axis/xpadneo/tree/master/hid-xpadneo/etc-udev-rules.d) exist in either of the following directories:

1. If using a distribution package: `/usr/lib/udev/rules.d`
2. If using direct install from source: `/etc/udev/rules.d`
3. The files should not exist in both directories.

After you confirmed your system state, we can proceed with diagnosing your situation.

RobotRoss, 2024-02-29:
I'm using the dkms install via source.

The udev rules are present in the correct directory and aren't duplicated.

kakra, 2024-02-29:
Then please turn the controller off and reboot the system, so we end up with a clean initial state. Now run `udevadm monitor -p | tee udevadm.log` and turn the controller on. While it connects, a bunch of text logs are sent to the terminal and will also be stored in `udevadm.log`. Please attach this file.

The last or one of the last blocks should have something like `DEVPATH=/devices/virtual/misc/uhid/0005:045E:0B13.001C` (and no more sub paths). Copy the path after `DEVPATH=` and run the command `udevadm info -a -p <paste-path-here> | tee udevinfo.log`. Please attach the file `udevinfo.log`, too.

I'm not sure if I can already see a problem here. Maybe you'll have to repeat this for the other case when we see only `hid-microsoft` in dmesg. If you want to create two sets of files, please ensure to record only one connection attempt per file (and keep a backup copy of the first attempt for attaching here). No reboot between the following attempts is needed, just turn the controller off, run the udevadm commands, check dmesg if it recorded the other case, and if yes, we have the other case.

Thanks.

RobotRoss, 2024-03-01:
I've attached the requested files.

I couldn't find anything under `DEVPATH=/devices/virtual/misc`, so I took a swing at one of the devices under /devices/pci0000:00/
[udevadm.log](https://github.com/atar-axis/xpadneo/files/14463140/udevadm.log)
[udevinfo.log](https://github.com/atar-axis/xpadneo/files/14463141/udevinfo.log)

kakra, 2024-03-01:
Is it possible that only the first connect after a reboot ends up in the `hid-microsoft` driver and successive connections properly use `hid-xpadneo`?

If yes, please try adding the following file if your system uses `systemd` to boot:
```
# /etc/modules-load.d/xpadneo.conf
hid-xpadneo
```

Then reboot. I'd expect that now xpadneo correctly binds on the first connect.

Other problems about making successful connections at all should probably attributed to bluez and your Intel Wireless chipset instead and should be reported to the bluez project if you still experience that.

As long as xpadneo correctly binds to the device, rumble should work. Please test.

RobotRoss, 2024-03-02:
> Is it possible that only the first connect after a reboot ends up in the `hid-microsoft` driver and successive connections properly use `hid-xpadneo`?

Unfortunately not. It's seemingly random if rumble decides it wants to work or not, although mostly it doesn't work with occasional times it decides to work.

kakra, 2024-03-03:
Maybe something is working concurrently on your system and races... Is `hid-microsoft` compiled in or is it a module?

RobotRoss, 2024-03-03:
Compiled in kernel by the look of it:

```
modinfo hid-microsoft
filename:       /lib/modules/6.7.6-200.fc39.x86_64/kernel/drivers/hid/hid-microsoft.ko.xz
license:        GPL
rhelversion:    9.99
alias:          hid:b0005g*v0000045Ep000002E0
alias:          hid:b0005g*v0000045Ep00000B22
alias:          hid:b0005g*v0000045Ep00000B05
alias:          hid:b0005g*v0000045Ep00000B13
alias:          hid:b0005g*v0000045Ep00000B20
alias:          hid:b0005g*v0000045Ep000002FD
alias:          hid:b0005g*v0000045Ep0000091B
alias:          hid:b0005g*v0000045Ep00000701
alias:          hid:b0003g*v0000045Ep000000E3
alias:          hid:b0003g*v0000045Ep000007DA
alias:          hid:b0003g*v0000045Ep0000076C
alias:          hid:b0003g*v0000045Ep0000009D
alias:          hid:b0003g*v0000045Ep00000732
alias:          hid:b0003g*v0000045Ep00000750
alias:          hid:b0003g*v0000045Ep000000B4
alias:          hid:b0003g*v0000045Ep00000730
alias:          hid:b0003g*v0000045Ep00000713
alias:          hid:b0003g*v0000045Ep000000F9
alias:          hid:b0003g*v0000045Ep0000071D
alias:          hid:b0003g*v0000045Ep000000DC
alias:          hid:b0003g*v0000045Ep000000DB
alias:          hid:b0003g*v0000045Ep00000048
alias:          hid:b0003g*v0000045Ep0000003B
depends:        ff-memless
retpoline:      Y
intree:         Y
name:           hid_microsoft
vermagic:       6.7.6-200.fc39.x86_64 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Fedora kernel signing key
sig_key:        50:44:7E:08:B8:CC:5D:1C:5C:F0:9C:77:FD:79:BB:07:87:49:71:01
sig_hashalgo:   sha256
```

kakra, 2024-03-03:
No, it's compiled as a module (but yes, as part of the kernel distribution, IOW, not an external module), otherwise it would say:
```
filename:       (builtin)
...
```

So if you don't depend on any other Microsoft input devices, you could try blacklisting the module (as a work-around, not a final solution):
```
# /etc/modprobe.d/no-hid-microsoft.conf
blacklist hid-microsoft
```

Then run `sudo depmod -a` and reboot to test again. Verify that `hid-microsoft` won't be loaded with `lsmod`. If this is an early boot driver (many input drivers are), you may need to recreate your initramfs/reinstall the boot loader.

Devices affected by this would still work via `hid-generic` but they won't use proper Microsoft quirk work-arounds for some functions, so YMMV if you use other genuine Microsoft input devices.

kakra, 2024-03-03:
One other thing, please check if this file (https://github.com/atar-axis/xpadneo/blob/master/hid-xpadneo/etc-modprobe.d/xpadneo.conf) has been installed in these directories without duplicates, and identical contents:

1. `/usr/lib/modprobe.d` or
2. `/etc/modprobe.d`

It should ensure that `hid-xpadneo` is tried first (my loading it) when encountering any of those hardware IDs.

In your logs I can see the modalias showing in udev, so it should trigger loading xpadneo. I'd expect to see it loading as soon as the modalias shows up, but it only loads at bind time when already gone through `hid-generic` and rebinding to `hid-microsoft` - this may be too late. Blacklisting `hid-microsoft` would prevent that, but OTOH, `hid-xpadneo` should already load when the modalias is encountered (you can see that `hidp` is triggered probably because the HID bus is on the Bluetooth bus, immediately `hid-xapdneo` should follow). Maybe try manually loading `hidp` before connecting the controller.

RobotRoss, 2024-03-03:
> One other thing, please check if this file (https://github.com/atar-axis/xpadneo/blob/master/hid-xpadneo/etc-modprobe.d/xpadneo.conf) has been installed in these directories without duplicates, and identical contents:
> 
>     1. `/usr/lib/modprobe.d` or
> 
>     2. `/etc/modprobe.d`
> 
> 
> It should ensure that `hid-xpadneo` is tried first (by loading it) when encountering any of those hardware IDs.
> 
> In your logs I can see the modalias showing in udev, so it should trigger loading xpadneo. I'd expect to see it loading as soon as the modalias shows up, but it only loads at bind time when already gone through `hid-generic` and rebinding to `hid-microsoft` - this may be too late. Blacklisting `hid-microsoft` would prevent that, but OTOH, `hid-xpadneo` should already load when the modalias is encountered (you can see that `hidp` is triggered probably because the HID bus is on the Bluetooth bus, immediately `hid-xapdneo` should follow). Maybe try manually loading `hidp` before connecting the controller.

Turns out there was a duplicate xpadneo.conf in /etc/modprobe.d which didn't match the contents of the github file.

Have deleted the duplicate and will report back.

RobotRoss, 2024-03-15:
Near as I can tell, this issue is now resolved and I haven't had any more issues with the controller in the past couple weeks. Closing!

closed 2024-03-15 by RobotRoss