atar-axis/xpadneo #462: Vibration not work on Gamesir T4 Pro

DiamivaeBro, 2024-02-22:
## Version of xpadneo
9.6 FINAL

## Controller Model
Gamesir T4 Pro SE2

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [x] Other: Gamesir T4 Pro

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [x] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [x] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [x] I don't use native games or did not try
- [x] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [x] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [x] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [x] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
Vibration just doesnt work.No connecting notify,no vibration in games and online tests

## Steps to Reproduce
1.Connect gamepad into xbox controller mode
2.Check vibration in any game or in https://mitinogame.ru/gamepad-tester/

## Expected Behavior
Gamepad must vibrate

## Screenshots / GIFs / Videos

## System Information

```console
Linux catslaptop 6.7.5-1-default #1 SMP PREEMPT_DYNAMIC Sat Feb 17 07:45:40 UTC 2024 (a3bab56) x86_64 x86_64 x86_64 GNU/Linux

```

```console
 xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81  u........$...%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05  ...%.u..............
000000c8: 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07  ..#...%...u.....%.u.
000000dc: 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75  ........!........%.u
000000f0: 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25  .......%.u......p..%
00000104: 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75  du......Pf..U...&..u
00000118: 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00  .........&..u.....e.
0000012c: 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06  U..|..&..u..........
00000140: 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0                    . ..&..u......
1703191670 1558

```

## Controller and Bluetooth Information

## Additional Context
Vibration stop working after 9.6 update,but it still working with dongle who emulating xbox 360 controller
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/14374509/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/14374510/xpadneo-dmesg.txt)

kakra, 2024-02-22:
This looks similar to https://github.com/atar-axis/xpadneo/issues/461 which suggests that it is a third-party bug, not directly caused by xpadneo itself.

Do you know how to compile software from source? If yes, please go to the directory `misc/examples/c_hidraw` and run `make` there. It will create an application `./hidraw`. To run it, it expects the hidraw device of xpadneo as parameter (see `dmesg` after connecting the controller).

It will open a fullscreen text window where you can use different keys to send raw rumble commands to the controller. Please check if that works:
```
--[  MotorEnable  ]---------------[ Strength in % ][  10ms ][cnt]--
03  1   1   1   1   0   0   0   0  040 030 020 020  005 005  003
    |   |   |   |                   |   |   |   |    |   |    |
   STR WEA LTR RTR  -   -   -   -  STR WEA LTR RTR  SUS REL  LOP
   [1] [2] [3] [4]                 q/a w/s e/d r/f  h/j k/l  u/i    <-- keys, or Enter = SEND, Ctrl+C = ABORT
```

Using keys 1-4, you can switch the bits which enable programming one of the motors. Your model will probably ignore that. STR/WEA/LTR/RTR is the strength of the strong, weak, left trigger and right trigger rumble motors individually.

SUS/REL is the duration in steps of 10ms to sustain and release the rumble. Your controller model may ignore that. LOP tells the firmware how often to repeat/loop SUS/REL, it's probably ignored, too, by your model.

The default values should give you a rumble on all motors. Set STR/WEA/LTR/RTR to 0 to make it stop rumbling if your controller ignores the SUS/REL timing.

If this makes the controller rumble, everything is fine. Then we need to figure out why user-space cannot rumble the controller. If it doesn't rumble with this test, it simply doesn't work at the hardware level, and xpadneo has no chance to rumble the controller. Some multi-platform controllers cannot rumble in Bluetooth mode.

DiamivaeBro, 2024-02-23:
Sorry but it doesnt building

DiamivaeBro, 2024-02-23:
Sorry but it doesnt building 
```console
❯ LANG="en" make
cc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -I/usr/include/ncurses  -lncurses -Wl,--push-state,--as-needed -ltinfo -Wl,--pop-state  hidraw.c   -o hidraw
/usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /tmp/cc0MQNDH.o: undefined reference to symbol '_nc_COLS@@NCURSES6_TINFO_5.7.20081102'
/usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /usr/lib64/gcc/x86_64-suse-linux/13/../../../../lib64/libtinfo.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make: *** [<builtin>: hidraw] Error 1
```

kakra, 2024-02-24:
> It vibrating!

Okay, this tells us that the controller can rumble in Bluetooth mode - and xpadneo will support it: xpadneo currently doesn't check if report 0x03 is available in the HID descriptor, it will just enable it unconditionally.

The other issue found that sometimes the kernel doesn't bind the correct driver, and uses hid-microsoft or hid-generic instead. The latter doesn't support rumble, the former may for some models.

There's also a bug in some SDL2 games (that includes Proton) which may not detect rumble support with xpadneo if the controller connects while SDL2 is already running. The work-around is easy: connect the controller before starting the game. More modern implementations of SDL2 games (may include modern Proton, too) may not have this problem. You can also disable the welcome rumble to work around this issue. A later xpadneo version will fix this.

kakra, 2024-02-24:
> It looks like
> 
> ```
> ❯ CPPFLAGS="-ltinfo" make
> ```

Yeah, good find. This is a problem with some ncurses packaging (the Makefile get's the flags from pkg-config). Gentoo fixed this a long time ago, so I didn't have this failure (it has `-ltinfo` in pkg-config). I wonder why you need `CPPFLAGS`. Maybe we should set `CPPFLAGS = $(CFLAGS)` in the Makefile.

kakra, 2024-02-24:
Please re-open, if the latest commit doesn't fix your problem.

closed 2024-02-24 by kakra

DiamivaeBro, 2024-02-24:
Yes,all fixed
Thank you!