atar-axis/xpadneo #465: Did not find any bluetooth devices fedora 39.

larsmartin4, 2024-03-09:
## Version of xpadneo

Package dkms-3.0.12-1.fc39.noarch is already installed.
Package make-1:4.4.1-2.fc39.x86_64 is already installed.
Package bluez-5.72-1.fc39.x86_64 is already installed.
Package bluez-tools-0.2.0-0.21.git20170912.7cb788c.fc39.x86_64 is already installed.
Package kernel-devel-6.7.7-200.fc39.x86_64 is already installed.
Package kernel-headers-6.7.3-200.fc39.x86_64 is already installed.

## Controller Model

sparkfox 
ATLAS PC Wireless Triple Mode Smart Controller
SKU: W60U550

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other: PC

### Connection mode
mars 09 18:32:07 fedora kernel: Bluetooth: Core ver 2.22
mars 09 18:32:07 fedora kernel: NET: Registered PF_BLUETOOTH protocol family
mars 09 18:32:07 fedora kernel: Bluetooth: HCI device and connection manager in>
mars 09 18:32:07 fedora kernel: Bluetooth: HCI socket layer initialized
mars 09 18:32:07 fedora kernel: Bluetooth: L2CAP socket layer initialized
mars 09 18:32:07 fedora kernel: Bluetooth: SCO socket layer initialized
...skipping...
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
mars 09 18:32:07 fedora kernel: Bluetooth: Core ver 2.22
mars 09 18:32:07 fedora kernel: NET: Registered PF_BLUETOOTH protocol family
mars 09 18:32:07 fedora kernel: Bluetooth: HCI device and connection manager in>
mars 09 18:32:07 fedora kernel: Bluetooth: HCI socket layer initialized
mars 09 18:32:07 fedora kernel: Bluetooth: L2CAP socket layer initialized
mars 09 18:32:07 fedora kernel: Bluetooth: SCO socket layer initialized
inxi --bluetooth
Bluetooth:
  Message: No bluetooth data found.
larsmartinhambro@fedora:~$ nmcli radio all
WIFI-HW  WIFI     WWAN-HW  WWAN    
missing  enabled  missing  enabled 

- [ ] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Describe your feature reqeust

- [ ] I'd like to see support for a specific model
- [ ] I'd like to see support for a specific hardware feature
- [ ] I'd like to see support for a specific software feature

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when ...
Did not find any bluetooth.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

larsmartin4, 2024-03-09:
Bus 003 Device 002: ID 045e:028e Microsoft Corp. Xbox360 Controller
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  idVendor           0x045e Microsoft Corp.
  idProduct          0x028e Xbox360 Controller
  bcdDevice            1.10
  iManufacturer           1 Controller
  iProduct                2 XBOX For windows
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x008b
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     93 [unknown]
      bInterfaceProtocol      1 
      iInterface              1 
      ** UNRECOGNIZED:  11 21 10 01 01 25 81 14 03 03 03 04 13 02 08 03 03
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     93 [unknown]
      bInterfaceProtocol      3 
      iInterface              0 
      ** UNRECOGNIZED:  1b 21 00 01 01 01 83 40 01 04 20 16 85 00 00 00 00 00 00 16 05 00 00 00 00 00 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               2
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     93 [unknown]
      bInterfaceProtocol      2 
      iInterface              0 
      ** UNRECOGNIZED:  09 21 00 01 01 22 86 07 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    253 [unknown]
      bInterfaceProtocol     19 
      iInterface              4 
      ** UNRECOGNIZED:  06 41 07 05 81 03

kakra, 2024-03-10:
How do you see the controller with `lsusb`? Did you connect it via USB cable? Then Bluetooth won't work. Also, it won't work for xpadneo with the Xbox dongle.

Try double tapping the connect button to see if the controller switches to Bluetooth mode, and disconnect it from USB, otherwise wireless modes are disabled.

kakra, 2024-03-10:
> Did not find any bluetooth.

This is most likely an issue with your distribution or hardware, xpadneo doesn't interact with Bluetooth directly. You should ask in your distribution forum first why no Bluetooth devices are listed at all if your hardware is supposed to support Bluetooth.

closed 2024-04-10 by github-actions[bot]