atar-axis/xpadneo #467: Packaging issue

guoyunhe, 2024-03-18:
## Version of xpadneo
0.9.6

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Describe your feature request

- [ ] I'd like to see support for a specific model
- [ ] I'd like to see support for a specific hardware feature
- [ ] I'd like to see support for a specific software feature

**Is your feature request related to a problem? Please describe.**

Hi, I packaged xpadneo for openSUSE. However, the initial version doesn't work at beginning. Then I figured out I need to install modprobe and udev rules config shipped with xpadneo. Then it worked!

However, when I submit this change to openSUSE, the request get rejected by maintainers, with the following comment:

> The module already has those very aliases (e.g. the B13 one), I cannot fathom how adding the *same* alias again would fix that.
> https://build.opensuse.org/request/show/1158334

I cannot explain it but it just work. Arch Linux's package also install these files: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=xpadneo-dkms

**Describe the solution you'd like**

Can you explain why it is needed to install modprobe? Thanks!

**Describe alternatives you've considered**

**Additional context**

kakra, 2024-03-18:
The comments thread is gated behind a login requirement, so I cannot check what has been objected exactly.

As a minimum, we require the udev rules being installed. In theory, the modaliases are indeed part of the module already. But other modules also provides those aliases, and it looks like an explicit modalias file can serve as an override (I cannot verify it due to lack of a proper testing environment but it fixed issues loading the driver for some users).

However, the udev rule should be able to rebind the driver from `hid-{generic,microsoft}` to `hid-xpadneo`. You can verify this with `udevadm monitor -p` with both the module already loaded and not loaded while connecting your controller.

If you can resolve the issue using only the udev rules, and this gets accepted by OpenSuSE, please comment and leave this open, so we can amend the packaging documentation.

guoyunhe, 2024-03-19:
@kakra Thanks for explanation! I did try to only install udev rules but it doesn't work in openSUSE. Currently modprobe is still required to make it work. I will try to communicate with openSUSE maintainers about the alias overriding problem. See if I get any luck. 😅

kakra, 2024-03-19:
From `man modprobe.d`:
```man
COMMANDS
       alias wildcard modulename
           This allows you to give alternate names for a module. For example: "alias my-mod really_long_modulename" means you can use "modprobe my-mod" instead of "modprobe really_long_modulename". You can
           also use shell-style wildcards, so "alias my-mod* really_long_modulename" means that "modprobe my-mod-something" has the same effect. You can't have aliases to other aliases (that way lies
           madness), but aliases can have options, which will be added to any other options.

           Note that modules can also contain their own aliases, which you can see using modinfo. These aliases are used as a last resort (ie. if there is no real module, install, remove, or alias command
           in the configuration).
```

So aliases from `modinfo` (as part of the module itself) are only used as last resort, aliases placed in the file system are always taking precedence - which should quite easily answer the question of the downstream maintainers. The problem here is clearly that `hid-microsoft` also provides those modinfo aliases. So overriding in `modprobe.d` is needed.

Talking about config files containing aliases, there's probably also an order of override within the files. `xpadneo.conf` is quite late in the alphabet so it's likely the last file overriding previous aliases. There's nothing in the man page about the order of reading those files, but I'd probably expect that the order of directories is as written in den `SYNOPSIS` part of the man page, and within each directory the order is just by alphabet.

closed 2024-05-23 by github-actions[bot]