atar-axis/xpadneo #469: Gulikit kk3 max recognized as 360 and rumble issues

Haxorzz, 2024-04-01:
## Version of xpadneo

v0.9-168-gbe25f7b
## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [x] Other: Gullikit KK3 Max

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [X] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [x] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [x] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [x] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [x] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [x] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [x] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

## Steps to Reproduce

(Android mode or X-input mode)
Pair Gulikit kk3 max to computer
Open steam -> settings -> controller
Controller classified as an xbox 360 controller
(X-input mode only)
Pair gulikit kk3 max to computer
As soon as connection is made, controller won't stop vibrating
setting quirk 7 to controller's mac address does not apply/does not work (x-input mode only) 

## Expected Behavior

Controller is detected as a modern controller in steam
Controller does not vibrate forever (x-input mode only)

## System Information

```console
# uname -a
# Linux LaptopLinux 6.7.6-201.fsync.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC TKG Wed Feb 28 02:20:59 UTC 2024 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
# 00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
# 00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
# 00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
# 0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
# 00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
# 00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
# 00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
# 0000008c: 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00  ....)...%.u.......%.
# 000000a0: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01  u.................%.
# 000000b4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
# 000000c8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
# 000000dc: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
# 000000f0: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
# 00000104: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
# 00000118: 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75  .u........... ..&..u
# 0000012c: 08 95 01 81 02 c0                                            ......
# 3445511648 1458

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/14827657/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/14827690/xpadneo-dmesg.txt)

no dongle = no lsusb entry

## Additional Context

Running the controller on firmware 3.8
Previously hidraw5 had correct permissions (still showed xbox 360), but after using the git version, now hidraw5 is almost forced to have wrong permissions (user has no rw) after the controller connects?

kakra, 2024-04-02:
> Previously hidraw5 had correct permissions (still showed xbox 360)

The controller should not expose any hidraw permissions to user-space, this is currently incompatible with SDL and Steam Input. xpadneo shows all controllers as Xbox360 controllers, that's normal and not an issue.

> but after using the git version, now hidraw5 is almost forced to have wrong permissions (user has no rw) after the controller connects?

Yes, this is intended behavior.

After connect, the controller should only play a short rumble. If it doesn't, some quirk flags may be needed for motor masking and/or pulse parameters. Unload the module and try loading the module with
```
rmmod hid-xpadneo && modprobe hid-xpadneo quirks=98:b6:ea:4a:21:a4+7
```

Does it fix rumble? You can confirm the quirk flags via dmesg output (it should change from `0x00000083` to  `0x00000087`. You already tried that before according to your report, but the quirks flags were wrong. So please double-check you've put the options correctly, and it's not set twice with different options. You controller OUI should already default to quirks 7 with the version of xpadneo you're using.

If you actually want to run with hidraw, do not use xpadneo. The controller will be supported via hid-generic and hidraw in SDL. Even if hid-generic won't support rumble, rumble is provided via hidraw in SDL. This would also fix the cosmetic issue with the controller being shown as "Xbox 360 controller". The downside is: Non-SDL games probably won't support rumble.

There's a test program for hidraw in `misc/examples/c_hidraw`: run `make` to build it (ncurses-dev is needed), then run it as root and the dev-path of your current hidraw device. It will open a simple GUI for programming and sending raw rumble commands to the controller. Try playing with the parameters to find which one your controller ignores.

rishabh-os, 2024-08-07:
Hi, I am having the same issue here, with the same controller. I have managed to fix the constant rumble with the steps described above, however, this just disables rumble entirely. No games or websites are able to use the rumble in the controller.

The test program for hidraw devices however, is able to send rumble commands to the controller. Using this I can control the strength of the rumble, and turn it on and off. 

xpadneo version: v0.9-172-gfd0cd48

kakra, 2024-08-08:
@rishabh-os Please open a new request describing which commands you send with the test program to stop or start rumble, and also describe which commands should stop the rumble but don't (e.g., there's a timing parameter which should stop rumble after some multiple of 10ms which often doesn't work on 3rd party controllers).

gustav1105, 2024-12-26:
I have a KK3 (std)  edition, rumble can be put on or off using the controller settings plus  up combo buttons, but it does not work in any games for me. tried in ubuntu and steam, controller has some oddities in modes found, No rumble in games. it is switched on and selected option 2 rumble,

kakra, 2024-12-26:
Please use the latest version and maybe open a new bug report.