atar-axis/xpadneo #470: Series controller not found at all

IceYetiWins, 2024-04-13:
Hi, I've been trying to use a series controller with manjaro using xpadneo and I'm facing an issue where the controller doesn't show anywhere in bluetoothctl or any bluetooth guis. The controller's firmware is fully updated and I have ERTM disabled, but no luck. It shows immediately on any other device.

Has anyone experienced similar?

kakra, 2024-04-14:
This is an issue with your Bluetooth installation of your distribution then, maybe the radio is disabled. Please ask in their forums how to check and enable Bluetooth. Also, do not disable ERTM anymore, modern kernels handle it properly and with ERTM disabled, you may run into issues.

IceYetiWins, 2024-04-14:
Other devices work and connect properly, it's only with xbox controllers that I've had issues. That's why I assumed it was an issue with xpadneo.

kakra, 2024-04-14:
xpadneo doesn't use Bluetooth at all. The bluez subsystem just gives us a HID device. But the Bluetooth connection itself is completely handled by bluez and the kernel Bluetooth drivers. xpadneo has no control over it.

Upgrade the controller firmware from Windows or an Xbox. Then retry. If it still doesn't work, try double-clicking the connect button (it will switch between Bluetooth and Xbox dongle). Check if you can connect the controller to your smartphone. Then try again with your PC. You need to press and hold the connect button until it starts to flash fast (around 2-3 seconds). No USB cable must be connected, otherwise the controller doesn't enable its radio (even a charging-only cable would not work).

The Xbox controllers are incompatible with some Bluetooth chipsets. There's no way around it except MS improving the firmware. You should at least see the controller, even if it does not connect properly in that case.

IceYetiWins, 2024-04-14:
Ah, didn't realize the bluetooth wasn't handled by xpadneo. 
- The controller is on the latest firmware
- Can connect to phone and windows on same pc
- I don't have an xbox dongle
- No cable connected
- I don't think chipset is incompatible since it works on windows

kakra, 2024-04-14:
Maybe you tried to connect the controller in the past and it is cached in bluez with invalid attributes?

Stop the bluez service, go to `/var/lib/bluetooth` and clear all files and directories matching the MAC address of your controller. If in doubt, empty the complete folder but keep in mind that you need to reconnect each Bluetooth device. After this, start the bluez service again, or reboot.

IceYetiWins, 2024-04-14:
Still nothing

IceYetiWins, 2024-04-15:
Update: I tried using a random bluetooth usb dongle I had and it worked, so I guess it must be a mobo support issue. Not an issue with xpadneo though so I'll close this thread.

closed 2024-04-15 by IceYetiWins