atar-axis/xpadneo #472: How To Update Controller Firmware through Xbox Accessories and VirtualBox

BobbyBlock, 2024-05-06:
So a bit of context I just spent two days troubleshooting an issue that turned out to be just me neglecting to download the linux-headers because it's not auto installed as a dependency by either xpadneo (which makes sense) or DKMS, during this process I assumed I had a firmware issue so tried updating the controller firmware through VirtualBox and managed to soft-brick my controller, making it not show up on my computer nor phone's Bluetooth while in pairing mode. However through this process I learned a fool-proof way to update the firmware through VirtualBox and thought I would share since I see it's an issue a lot of people have had.

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [X] Xbox Series X|S controller
- [ ] Other:

But I see no reason why it wouldn't work the same with other controller

## Steps to Update Controller Firmware through Xbox Accessories and VirtualBox

1. Download VirtualBox
2. Download Windows 10 ISO (You can get it [here](https://www.microsoft.com/en-us/software-download/windows10ISO) from Microsoft as long as you are not accessing the website from a Windows device, if you have a Windows device then just download Xbox Accessories on that and skip VirtualBox and this guide altogether)
3. Load the Windows ISO on the virtual machine
4. Set up Windows, install all Windows Updates and all updates available in the Microsoft Store to ensure maximum compatibility, also install Xbox Accessories from the Microsoft Store and then shutdown the VM

So far all pretty standard

6. If you have xpad installed (was installed by default for me) you are going to want to disable that by placing a file in `etc/modprobe.d/`, this file can be named whatever as long as it ends in .conf
7. Inside this file write `blacklist xpad`. These 2 steps will stop the xpad drivers being loaded. Otherwise the xpad driver will make the controller look like a 360 controller to your PC, which affects the USB passthrough into Windows and will prevent it from showing up properly in Xbox Accessories 
8. Restart your PC
9. Go into your VirtualBox VM's Settings and go to "USB" on the sidebar
[image]
          
10. Then, while ensuring your controller is plugged in, go to "Add new USB filter with all fields set to the values of
 the selected USB device connected to the host PC" (The little picture of the USB with the plus next to it). Select your controller from the list of devices that pop up
11. Then edit that new filter by double clicking on it in the "USB Device Filters List" and remove all fields except `Port`, this will ensure that the controller stays connected throughout the firmware update and not soft-brick it like I did, and `Name` which you can set to whatever you want. Ensure that you keep to controller connected to the same USB Port throughout as what you have just done will mean anything connected to that port will autoconnect to the VM while it's running. 
12. Launch the VM and open Xbox Accessories, at this point the controller should appear correctly and you will be able to update the firmware successfully
13. At this point if you are planning on using xpad again in future (useful for if you want to play over wire instead of Bluetooth) you will want to reverse the actions taken in steps 6 & 7 by deleting that file you created. 

Hope this helps some people

kakra, 2024-05-07:
> So a bit of context I just spent two days troubleshooting an issue that turned out to be just me neglecting to download the linux-headers because it's not auto installed as a dependency by either xpadneo (which makes sense) or DKMS

Distributions don't install kernel headers as part of DKMS, e.g. Arch documents this as a manual step, and other distributions depend on correct headers if you install xpadneo-dkms through the package manager.

If you install xpadneo from source, you're supposed to also provide the proper kernel headers. This project has no business in installing random packages, especially since this is different per distribution. We provide hints in the docs for that, and I'm pretty sure kernel headers are generally mentioned. ;-)

OTOH, "which makes sense" could just mean "it makes sense for xpadneo to not install the headers". But something similar is true for DKMS itself. I think Arch Linux documents, why...

> during this process I assumed I had a firmware issue so tried updating the controller firmware through VirtualBox and managed to soft-brick my controller, making it not show up on my computer nor [...]

Yes, the controller boots into a firmware upgrade mode which can make it appear soft-bricked. I'm pretty sure this can be reset by removing the batteries. But good you found a way of resolving and documented the whole process.

I'll keep this open so we can include it in the documentation later.

Thanks.

temereo, 2025-02-01:
@BobbyBlock you are my saviour!

P.S. sry for off-topic, kind of, but this solution was extremely helpful for me!

anmolsekhon590, 2025-02-14:
Why do I see this after dual booting Windows.
This needs to be in Arch Wiki 🙏

brianclinkenbeard, 2025-02-18:
For anyone looking to do this with QEMU, I used [quickemu](https://github.com/quickemu-project/quickemu), then started Windows 10 with spice (`quickemu --vm windows-10.conf --display spice`) and enabled the controller passthrough under "Input" (also select "Auto redirect newly plugged in USB devices" for when the update causes reconnects) and this worked. I had to blacklist xpad, as you noted.

innateessence, 2026-01-20:
In case anyone wants another approach to take.

I used `quickemu` to setup my Windows VM, but my steps were a bit different. Here's the small guide I wrote for myself.

```md
## How to create the windows 10 VM
* Refer to quickemu to get a bootable windows VM. Proceed after you can boot into a working windows VM.

## What to do before running the Windows VM
* Blacklist `xpad` and `xpadneo`
* `echo -e "blacklist xpad\nblacklistxpadneo"| sudo tee /etc/modprobe.d/block-controller.conf`
* Reboot
* `systemctl --user stop sunshine` And stop anything else that tries to grab the xbox controller device

## How to run Windows
* Run the VM, but don't connect the viewer yet.
* `quickemu --vm windows-10.conf --display spice --viewer none`
* Now, we connect the viewer as `root` user
* `sudo remote-viewer spice://127.0.0.1:5930`

## What to do in the Windows VM
* Open Xbox Accessories App
* In the Spice Viewer (VM GUI) Select the Xbox Controller for Spice USB Redirection
* Begin firmware update in Xbox Accessories App
* Keep the Spice USB Redirection window open. The controller will disconnect and reconnect many times. Check the box to re-enable redirection everytime.
* You basically babysit the Spice USB Redirection window while the firmware update process happens. It should finish successfully after a couple minutes.

## Post Firmware Update
* `rm /etc/modprobe.d/block-controller.conf`
```

kakra, 2026-01-21:
One other note: One has to be careful with USB auto redirect. If users manage to disconnect and redirect their mouse or keyboard (by accident or intentional), they end up with input periphery jailed inside the VM - with no possible way to stop the auto redirection again except by a hard reboot.

kakra, 2026-03-05:
> Apologies for necroing this old thread, but I was wondering, is using the `quickemu` workaround detailed above something that can be used for reprogramming the Elite Series 2's back paddles in the 'Xbox Accessories' app?

@droenn This should work but I didn't test that yet as part of some reverse engineering efforts. My future goal is to allow back paddles programming directly in Linux, but for now we will be going with software emulation during one of the next release cycles.

I know that there are xpadneo users which actually successfully did that. If you are seeing bugs resulting from it, please open a new bug report.