atar-axis/xpadneo #473: Steam and Wine not seeing any sort of input on the controls

dbob16, 2024-05-09:
## Version of xpadneo
aur/xpadneo-dkms 0.9.6-1
## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ x] Xbox Series X|S controller

- [ ] Other:

### Connection mode

- [x ] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [x] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [x] xboxdrv (user-space gamepad driver)
- [x] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [x] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [x] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool
- [x] Running "wine control" doesn't show any input from controller

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help
- [x] Tried adding my user to the input group.

## Describe the Bug
Controller connects. But it doesn't pick up ANY input from it.

## Steps to Reproduce

## Expected Behavior
I expected to see something in the various tests on Steam and in WINE.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux Dilan-PC 6.8.9-zen1-2-zen #1 ZEN SMP PREEMPT_DYNAMIC Tue, 07 May 2024 22:06:02 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

```

## Controller and Bluetooth Information

## Additional Context

kakra, 2024-05-09:
Please reboot your computer, run `dmesg -w` and then connect the controller. Post the `dmesg` output. Thanks. :-)

dbob16, 2024-05-09:
I got it working by uninstalling xone and xboxdrv, then reverted back to the original arch kernel. Marking as completed as a result.