atar-axis/xpadneo #474: Triggers getting stuck after releasing them

jaimetavares-code, 2024-05-12:
## Version of xpadneo

v0.9.6 FINAL
## Controller Model

- [x] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [x] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [x] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [x] I don't use native games or did not try
- [x] Other software is having issues (describe software and issues below)
- PCSX2 is having issues
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

The triggers (RT and LT) get stuck after pressing them randomly. I will press the triggers in a game, let go of the trigger and the game will still show signs of the trigger still being active, until i press it again. 

## Steps to Reproduce

Press or hold RT or LT
Let go of RT or LT

## Expected Behavior

Game to not detect the buttons being active when I'm not pressing them

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux pop-os 6.8.0-76060800daily20240311-generic #202403110203~1714077665~22.04~4c8e9a0 SMP PREEMPT_DYNAMIC Thu A x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0
```

## Controller and Bluetooth Information

## Additional Context

kakra, 2024-05-13:
Please reboot the system, run `dmesg -w`, then connect the controller. Post the output of `dmesg`. Thanks. :-)

jaimetavares-code, 2024-05-13:
@kakra, here's the output of `dmesg -w`
[dmesg -w.txt](https://github.com/atar-axis/xpadneo/files/15291911/dmesg.-w.txt)

I included the output as a txt file, so it's easier to read. Lemme know if there's anything else I can provide to help you help me.

kakra, 2024-05-13:
Thanks, I wanted to ensure you're actually running your controller with xpadneo because creating the report descriptor fails - but that may be due to outdated instructions which don't work since firmware switched to BLE.

Do you run games through Steam when the problem occurs? Do Lutris games show the same problem if you quit the Steam client?

I've seen this behavior only with the XBE2 controller in the very early firmwares back when the Bluetooth connection was still very unstable. It has long been gone and your firmware is updated. So I'm wondering if we see another instance of "double controller instance" while Steam is running, and controller inputs from both instances race with each other. In that case, disabling Xbox controller support within Steam should fix the issue.

If this is true, you should not be able to see the issue with `evtest` because it will watch just exactly one device instance. But it may show your controller twice (with different names) while Steam is running.

jaimetavares-code, 2024-05-13:
@kakra I run games through Steam, Lutris and PCSX2. The problem still occurs if I quit the Steam client.
When i run `evtest` I see 3 instances of "Xbox Wireless controller":
`/dev/input/event11:	Xbox Wireless Controller`
`/dev/input/event12:	Xbox Wireless Controller Consumer Control`
`/dev/input/event13:	Xbox Wireless Controller Keyboard`

The same happens with Steam open or closed.
Also I have Xbox controller support disabled in Steam, the problem still occurs.

kakra, 2024-05-13:
Those three are not really instances, they are sub-devices: The Xbox controller supports keyboards on the extension port, the XBE2 controller actually always shows being a keyboard, without connecting a keyboard. I think this is because you can remap buttons through the firmware. xpadneo simply shows all features no matter the model, we will implement emulation of some features later.

Your gamepad sub-device is actually on `event11` in that case. The keyboard and CC sub-devices will be ignored by games.

If you run `evtest` for the game sub-device, please check if you can replicate events that don't end up on value 0 if your release the triggers. If you can, we are seeing a low-level problem in the driver or in Bluetooth.

But if releasing the trigger always ends up on a zero value, the problem is somewhere else, maybe in SDL or wine. We can look further from here in that case.

You can run `evtest` along with the game to compare trigger values with behavior in the game.

jaimetavares-code, 2024-05-13:
@kakra I'm sorry for the delayed response. Releasing the trigger always ends up on a zero value, but the game still keeps acting as if i was pressing it
example: In dragon ball z budokai tenkaichi 3, the character keeps charging up (LT button) even after releasing it, until i press it again, and it goes back to normal.

kakra, 2024-05-13:
I've seen previous reports with people using retro game emulators and seeing a similar behavior with the A button registering twice in the game. It may affect other buttons, too, but that's not so easy to discover.

So far, my conclusion is that this happens under certain circumstances if you press multiple buttons (or other inputs) at once. There's nothing in the driver code which could affect it. Well, actually, there is a code path which could affect it and has to do with syncing the report frames to the kernel input driver: It can happen if you use the paddle buttons on the back of the controller but yours doesn't have those.

The sync issue may also occur with the combined z axis feature, tho I've never seen an issue with it. A fix was introduced with 5ab9bcf4953ef4b1c48f330e1340527e683baa62 and that's probably not backported to the 0.9 branch yet.

So I'd like to ask you for two tests:

- **with the 0.9 branch**, the issue should be present only when using the paddles (combined z axis does not report a sync event): check if you can see an sync event in `evtest` between two simultaneous inputs which should be in the same report frame (without an intermediate sync), so you'd need to use the same button/trigger combo as in the game
- **with the 0.10 branch (master)**, the issue should be fixed: Try the latest git version and see if it is fixed, if yes, please report so

Please tell me the exact button/axis/trigger combo you're using to show the behavior.

I'm somewhat convinced that this issue is not triggered by a bug in xpadneo (except the known sync issue with the paddles) but rather something in the Bluetooth stack. You could try a different Bluetooth dongle.

The above commit mentions two other linked issues:

- https://github.com/atar-axis/xpadneo/issues/460, especially https://github.com/atar-axis/xpadneo/issues/460#issuecomment-1975394577
- https://github.com/atar-axis/xpadneo/pull/282 because mouse mode introduces new sync behavior

BTW: You wrote you are using v0.9.6 but your dmesg indicates you're using the master branch (v0.10-dev). The above text is based on the assumption you're really using v0.9.6. In latest v0.10, the combined z axis is turned off by default and marked deprecated but maybe you turned it on via a module parameter? (`enable_rolling_axis=1`, default is `0`)

Whatever version you're using, try both the v0.9 branch and the latest master branch and see if there's a difference.

Also, maybe 5ab9bcf4953ef4b1c48f330e1340527e683baa62 introduced a new bug which I wasn't seeing before. After having checked out master, you can try `git revert 5ab9bcf4953ef4b1c48f330e1340527e683baa62` and see if that fixes your issue. If it does, there may have been an oversight in my code.

Also, if testing with `evtest`: Only a sync event (`SYN_REPORT`) will finalize the input report and send it to user-space, read: If there are multiple button presses and releases within the same frame (aka between to syncs), user-space will only see the final result and can only discover presses and releases of inputs from two different report frames. IOW, if your trigger behavior does not end with a sync, user-space won't see the releasing of the trigger until a sync is sent.

Try `evtest /path/to/device | grep -E 'SYN|ABS_RZ|ABS_Z'` to filter out noise from other inputs and see if you can replicate an issue where the last trigger value isn't zero before the last sync. I'm able to rarely end on a value of 1, 2, or 3, just before the last sync. But this looks more like an issue within the firmware, and your game shouldn't actually see such a low value as an active input - otherwise you may need to enable deadzones (default, but maybe you changed it for better precision in Proton games).

jaimetavares-code, 2024-05-14:
@kakra You're right, this isn't a problem within xpadneo.
After changing deadzone values on my emulator and certain games with the setting, I was unable to replicate my problem. Both on v.09 and v0.10, changing the deadzones has made it so my problem disappeared. 
To make sure, I've also tried on an older laptop that has linux mint installed, installed xpadneo on it too and the problem didn't exist after configuring the deadzones too.
No other button or trigger now seems to get "stuck" or even repeats itself.

kakra, 2024-05-14:
Nice, thanks for confirmation. I'll add that to the documentation, please keep opened.

closed 2025-12-17 by kakra