atar-axis/xpadneo #476: Controllers do not register as joysticks after reboot

AccaliaDeElementia, 2024-06-05:
Pairing an Xbox Elite (1) controller or an Xbox Series X controller via the KDE control panel with xpadneo install works flawlessly.... until i reboot my system. After reboot the controller will connect via bluetooth, but xpadneo does not pick either model of controller up.

## Version of xpadneo
0.9.6 - via AUR package xpadneo-dkms

## Controller Model

- [ ] Xbox One S controller
- [x] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information
Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [x] It mostly works but sometimes it doesn't
- [x] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
Pairing an Xbox Elite (1) controller or an Xbox Series X controller via the KDE control panel with xpadneo install works flawlessly.... until i reboot my system. After reboot the controller will connect via bluetooth, but xpadneo does not pick either model of controller up.

Forgetting the controller and re-pairing will cause the controller to be picked up again and work flawlessly, until the next reboot.

## Steps to Reproduce

Pair controller using KDE system config, observe that the controller works flawlessly, including power cycling the controller and reconnecting.

Reboot the host machine, observe that while the controller will automatically connect, it is not detected as a joystick and xpadneo is not attaching to the device. No lines appear in dmesg when connecting.

Forget the controller in the KDE control panel and re-pair, observe that it is connected successfully and dmesg shows xpadneo grabbing the device

## Expected Behavior

Connecting the controller after reboot works.

I'm not sure if this is a bluez issue or an xpadneo issue, but i cannot seem to figure it out, and attempting the various fixes suggested when googling or reading the docs/issues result in either breaking the bluetooth pairing process entirely or no change in behavior.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux Alexis 6.9.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 31 May 2024 15:14:45 +0000 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0
```

## Controller and Bluetooth Information

xpadneo-reconnect-btmon.txt and xpadneo-reconnect-dmesg.txt are after a fresh boot and connecting the controller and resulted in no detected controller

xpadneo-new-pairing-btmon.txt and xpadneo-new-pairing-dmesg.txt are after forgetting the controller and initiating a pair and resulted in a detected and working controller

[xpadneo-lsusb.txt](https://github.com/user-attachments/files/15597463/xpadneo-lsusb.txt)
[xpadneo-reconnect-btmon.txt](https://github.com/user-attachments/files/15597466/xpadneo-reconnect-btmon.txt)
[xpadneo-reconnect-dmesg.txt](https://github.com/user-attachments/files/15597467/xpadneo-reconnect-dmesg.txt)
[xpadneo-new-pairing-btmon.txt](https://github.com/user-attachments/files/15597464/xpadneo-new-pairing-btmon.txt)
[xpadneo-new-pairing-dmesg.txt](https://github.com/user-attachments/files/15597465/xpadneo-new-pairing-dmesg.txt)

## Additional Context

kakra, 2024-06-05:
From your description, it first sounds like a udev issue. But looking at the logs, this is more likely an issue with bluez. This was most likely fixed in the later firmware versions but your version is up to date. So probably something isn't handled properly for your Bluetooth chipset in bluez.

Please check the troubleshooting for bluez and adjust some settings, to see if it helps: https://github.com/atar-axis/xpadneo/blob/master/docs/TROUBLESHOOTING.md

Also, lately I've discovered that my original Xbox XB1S controller doesn't automatically reconnect. Press and hold 1 second on the connect button fixes that and it connects.

AccaliaDeElementia, 2024-06-06:
Hmm.... 

looks like on arch, or with recent enough versions of bluez the JustWorksRepairing setting is required to properly load xpadneo after a reboot.

after... i honestly lost track of how many reboots it took, but here's the bluetooth config that appears to reliably work, at least with my bluetooth controller:

```
█▓▒░accalia@Alexis░▒▓██▓▒░ Thu Jun 06 11:18:08 
~/ grep -e '^[^#]' /etc/bluetooth/main.conf
[General]
JustWorksRepairing = always
[BR]
[LE]
[GATT]
[CSIS]
[AVDTP]
[Policy]
[AdvMon]
█▓▒░accalia@Alexis░▒▓██▓▒░ Thu Jun 06 11:18:24 
~/ 
```

closed 2024-06-06 by AccaliaDeElementia