atar-axis/xpadneo #483: Can't play with two controllers

kelna, 2024-07-19:
## Version of xpadneo: 0.9.6-1

## Controller Model

- [x] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [x] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

I want to play split screen co-op with two controllers. It doesn't work. Both controllers work individually but not at the same time.

## Steps to Reproduce

## Expected Behavior

It should work. I used to play coop on Linux with these two controllers with an Xbox dongle and xone. Then it stopped working for some reason. Now I want to try xpadneo.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux x 6.9.10-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 18 Jul 2024 18:06:13 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0
```

## Controller and Bluetooth Information

## Additional Context

kakra, 2024-07-20:
If it stopped working, the problem is at another layer, and xpadneo can do nothing about it. I'll try a game that supports multiple controllers to see if it still works for me.

Which game did you try?

kelna, 2024-07-20:
Unravel Two on Steam

kelna, 2024-07-20:
Oh, now it works. I don't know what solved the issue, but I uninstalled and reinstalled both xone and xpadneo several times. Xone doesn't work at all, but xpadneo does and now I can play multiplayer. Thanks and sorry for the issue.

closed 2024-07-20 by kelna

kakra, 2024-07-21:
Then this is probably an issue with bluez keeping devices around that have been disconnected. If you pair the first controller, disconnect it, it will stay as a visible device in the system. If you then connect a USB controller, it will become second. Then pair another wireless controller and it will become the third controller and stay at that position: You game may not recognize it even if you disconnected the USB controller. A reboot usually resolves this, and restarting bluez should also resolve that.