atar-axis/xpadneo #484: Vibration not work on Gamesir T4 Nova Lite

BakaJzon, 2024-07-27:
## Version of xpadneo

xpadneo-dkms-git & xpadneo-dkms 0.9.6-1 from AUR
`-git` means using the latest commit on master branch to build

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [x] Other: Gamesir T4 Nova Lite or 盖世小鸡启明星

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [x] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [x] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [x] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [x] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [x] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [x] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

I am writing to report an issue with the GameSir Nova Lite (T4NL) gamepad that resembles the one described in #462. The gamepad has only a main vibration motor. Here are the details of the issue:

**Connection Modes:**
- Wired Mode: Recognized as a generic gamepad (with vibration) in Linux under Steam Play.
- Bluetooth Mode: Recognized as an Xbox Series X gamepad.
- Bluetooth + Xpadneo: Recognized as an Xbox 360 gamepad.

**`Hidraw` Test:**
- The gamepad responds to control data only when all the STR, WEA, LTR, and RTR parameters are set to 1.
- The SUS, REL, and LOP parameters have no effect. After the last command is issued, the gamepad vibrates for a fixed duration of 20 seconds.
- The STR and WEA parameters are effective.
- There's no trigger motor in this gamepad.

## Steps to Reproduce

1. Connect gamepad into xbox controller mode
2. Check vibration in any game

## Expected Behavior

Gamepad must vibrate

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux bkjzme 6.10.1-zen1-1.1-zen #1 ZEN SMP PREEMPT_DYNAMIC Fri, 26 Jul 2024 03:22:07 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

## Controller and Bluetooth Information

## Additional Context

"Portal 2" is the only game that can trigger the vibration, and it does so only when both the red and blue portal shooting buttons are pressed simultaneously (an action not allowed by the game).

hithere-at, 2024-09-15:
Hello, i am also facing the same issue on firmware v4200 (which is the last firmware to support Bluetooth Xinput mode) and have no vibrations. Did you find a workaround?

BakaJzon, 2024-09-16:
Sorry for not including `dmesg`, you might need this below.
The MAC of this gamepad is `3e:42:6c:84:17:3f`.
```
86208.898123] loaded hid-xpadneo 0.9.r174.gbc99c69
[86210.394584] xpadneo 0005:045E:0B13.0022: classic Bluetooth firmware version 0.23, please upgrade for better stability
[86210.394610] xpadneo 0005:045E:0B13.0022: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
[86210.394615] xpadneo 0005:045E:0B13.0022: working around wrong SDL2 mappings (changed version from 0x00000023 to 0x00001130)
[86210.394622] xpadneo 0005:045E:0B13.0022: report descriptor size: 283 bytes
[86210.394625] xpadneo 0005:045E:0B13.0022: fixing up Rx axis
[86210.394628] xpadneo 0005:045E:0B13.0022: fixing up Ry axis
[86210.394630] xpadneo 0005:045E:0B13.0022: fixing up Z axis
[86210.394632] xpadneo 0005:045E:0B13.0022: fixing up Rz axis
[86210.394634] xpadneo 0005:045E:0B13.0022: fixing up button mapping
[86210.394993] xpadneo 0005:045E:0B13.0022: gamepad detected
[86210.394996] xpadneo 0005:045E:0B13.0022: enabling compliance with Linux Gamepad Specification
[86210.395089] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0022/input/input85
[86210.395603] xpadneo 0005:045E:0B13.0022: input,hidraw3: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on c8:58:c0:23:a0:27
[86210.395671] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B13.0022/input/input86
[86210.395866] xpadneo 0005:045E:0B13.0022: consumer control added
[86210.395981] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B13.0022/input/input87
[86210.396135] xpadneo 0005:045E:0B13.0022: keyboard added
[86210.396145] xpadneo 0005:045E:0B13.0022: controller quirks: 0x00000050
[86210.396148] xpadneo xpadneo_welcome_rumble start
[86211.386211] xpadneo xpadneo_welcome_rumble took 990ms
[86211.386216] xpadneo 0005:045E:0B13.0022: Xbox Wireless Controller [3e:42:6c:84:17:3f] connected
```

kakra, 2024-09-22:
It looks like your controller may have swapped motor enable bits. There's a test program in `misc/examples/c_hidraw` which you can use to check whether the motor enable bits a properly assigned or work at all.

Just run `make hidraw` in that directory, you may need to install some dev libraries, e.g. ncurses-dev, to make it work. Then connect your controller, run `dmesg` to find the current hidraw device name, and run `sudo ./hidraw /dev/<HIDRAWNAMEHERE>`.

hithere-at, 2024-09-22:
I just did the `hidraw` test and found that the controller vibrates when **all** of the `MotorEnable` bits (STR, WEA, LTR, RTR) is turned to 1. The strength bits also affects the strength of the vibration, but only STR and WEA bit works.

I dont know if this information is relevant, but i am using TP-Link UB500 Bluetooth dongle. Also, should i try loading the xpadneo kernel module with some parameters?

kakra, 2024-09-22:
Thanks, this is valuable information. Do you really need ALL motor bits set? Or does any bit correspond to some motor?

We already have quirks flags to reverse the bit mask (128) and disable the motor mask complete (4), but the latter just leaves the bits 0. Maybe we should change them to 1 instead:
```c
// xpadneo.c, line 273
	/* do not send these bits if not supported */
	if (unlikely(xdata->quirks & XPADNEO_QUIRK_NO_MOTOR_MASK))
		r->ff.enable = 0;
```

If you find the remaining information needed, I can create a proper patch. Please leave this open then, I will close this issue via commit.

hithere-at, 2024-09-22:
Based on my testing, all of the motor bits has to be set in order for the vibration to work. It seems that any unset motor bit will NOT vibrate the controller. As for the quirks flag, i have tried loading the xpadneo kernel module with +4, +128, +132, -4, -128, and -132 but vibration is still not working. I am testing the vibration with Wine control panel in the `Game Controllers` menu. Also the controller has dual vibration motor.

This is the commands i used to unload and load the kernel module:
`sudo modprobe -r hid-xpadneo # unload`
`sudo modprobe hid-xpadneo quirks=ed:bc:9a:85:33:6c{+|-}{quirks} # load`

kakra, 2024-09-22:
I'll create a PR for you to test.

kakra, 2024-09-22:
> sudo modprobe hid-xpadneo quirks=ed:bc:9a:85:33:6c{+|-}{quirks} # load

Your MAC OUI seems to differ from the original reporter's. Can you send the dmesg output of the controller connecting? Is it the same manufacturer?

hithere-at, 2024-09-22:
I am pretty sure i have the same controller as the original reporter. I will compare the reporter's dmesg output and mine. This is the dmesg output.
[xpadneo-dmesg-log.txt](https://github.com/user-attachments/files/17091526/xpadneo-dmesg-log.txt)

hithere-at, 2024-09-22:
Oh yes, it is different. If you need more system information, i have attached the output of my controller's device descriptor and my `uname -a` output.

```
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```
```
Linux zhezhi 6.9.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 28 Jun 2024 04:32:50 +0000 x86_64 GNU/Linux
```

kakra, 2024-09-22:
Please try the pull request.

hithere-at, 2024-09-22:
Thank you, but im sorry i cannot test the pull request right now because im at school. I will get back to you when i get back home.

hithere-at, 2024-09-23:
Hello, it is with great news that the controller now vibrates instead of pure silence. Great work guys. Thank you very much :DDD

The only thing i noticed is that the welcome rumble takes some good amount of time (around 10 seconds i think) but that can be disabled using a kernel parameter. Again, thank you very much.

BakaJzon, 2024-09-23:
Thanks. This patch works on my devices too!

Looking forward to it being merged into the master branch! XD

BakaJzon, 2024-09-23:
> I am pretty sure i have the same controller as the original reporter. I will compare the reporter's dmesg output and mine. This is the dmesg output. [xpadneo-dmesg-log.txt](https://github.com/user-attachments/files/17091526/xpadneo-dmesg-log.txt)

It's strange. Did you update the firmware? And the controller LED should be blue.

kakra, 2024-09-23:
> The only thing i noticed is that the welcome rumble takes some good amount of time (around 10 seconds i think)

Okay, then this is another important fix we need to do. It means that the controller doesn't currently correctly handle our rumble stop command in the initial rumble test (the welcome rumble is actually a test to easily see if the controller behaves as expected).

kakra, 2024-09-23:
> It's strange. Did you update the firmware? And the controller LED should be blue.

Looks like this is a cheap Chinese manufacturer who actually uses unofficial unassigned MAC OUIs. I couldn't find your OUIs in any database. I'm not sure if this is official allowed in most countries because the MAC is sent on radio. Using the device in some countries may thus not be covered by laws if unofficial OUIs are in use. But we don't need to care about that in xpadneo, it only means these controllers will probably have a lot of different random OUIs and we cannot cover all in the driver. I've already updated the docs to reflect that. Maybe we can detect this device in a more general manner if it has a specific size of HID report descriptor, or specific bytes at specific positions.

hithere-at, 2024-09-23:
Yes, the controller LED is blue and the firmware is updated to v4200. Its the last firmware to support XInput on Bluetooth. Firmware v5700 and v6900 now changed the Bluetooth XInput mode to DInput (PS4) as far as i know.

Maybe you're on a different firmware than mine?

BakaJzon, 2024-09-23:
> > It's strange. Did you update the firmware? And the controller LED should be blue.
> 
> Yes, the controller LED is blue and the firmware is updated to v4200. Its the last firmware to support XInput on Bluetooth. Firmware v5700 and v6900 now changed the Bluetooth XInput mode to DInput (PS4) as far as i know.
> 
> Maybe you're on a different firmware than mine?

v4200 as well.
So the manufacturer is probably using random MACs. ~~But it is so cheap!~~

hithere-at, 2024-09-23:
Wait, v6300 exists?
I need to check them

hithere-at, 2024-09-23:
Well looks like firmware 6300 is not available on the Gamesir Android app, but its available on the website that you attached. Anyways, controller now vibrates and we can wait for the PR to be merged. Thank you everyone for all of the informations.

kakra, 2024-10-31:
Can you guys check if https://github.com/atar-axis/xpadneo/pull/496 actually fixes your problem?

hithere-at, 2024-11-02:
Do you mean the vibration? Then yes, it now vibrates the controller, but welcome rumble still takes around  ~20 seconds before it stops vibrating.

kakra, 2024-11-02:
> but welcome rumble still takes around ~20 seconds before it stops vibrating.

Okay then we still need some tweaking. I guess we need to send an explicit "strength 0" then to stop it because it doesn't obey the haptic timer parameters.

hithere-at, 2024-11-03:
I just found out that adding the `XPADNEO_QUICK_NO_PULSE` quick makes the welcome rumble behave correctly.

hithere-at, 2024-11-03:
And yes, it does need the strength set to 0 (both STR and WEA on hidraw test) to stop the vibration.

kakra, 2024-11-03:
I'll add that then. Thanks.

hithere-at, 2024-11-04:
You're welcome. Thank you for the Nova Lite support by the way, really amazing work.

BakaJzon, 2024-11-05:
I bought a new controller, and its MAC address is `6A:07:14:8F:07:7B`.
Clearly, this manufacturer is using randomly generated MAC addresses. That's bad.

Anyway, thanks for your awesome work.

kakra, 2024-11-05:
Maybe we can find a common trait of those types of controllers, a specific byte in the HID descriptor, a specific length of the descriptor. It could be useful if can collect some dumps to see if they differ between firmware versions.

I'm not sure if I can integrate such a detection easily into the source code, but we can give it a try.

hithere-at, 2024-11-06:
> It could be useful if we could collect some descriptor dumps to see if they differ between firmware versions
I see. I will check them when i have time.

hithere-at, 2024-11-06:
I just checked the descriptor dump between firmware v3800 to v4200. There are no difference. Even the descriptors are the same with the one from the first comment. 
[dump_v3800.txt](https://github.com/user-attachments/files/17648706/dump_v3800.txt)
[dump_v4200.txt](https://github.com/user-attachments/files/17648710/dump_v4200.txt)
[dump_v4000.txt](https://github.com/user-attachments/files/17648708/dump_v4000.txt)

BakaJzon, 2024-11-25:
I'm not sure if I did something wrong. When I try to package it (either by using ./install.sh or by modifying the [PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=xpadneo-dkms-git)), I get the error like this:
`fatal: No names found, cannot describe anything.`

So I manually added a tag for `10274c0` using `git tag v0.9-178-g10274c0`
```bash
$ git log
commit 10274c064b6e5cd7fd8bfcedbd88c09d7825a52e (HEAD -> hardware/gamesir-t4-nova-lite, tag: v0.9-178-g10274c0, origin/hardware/gamesir-t4-nova-lite)
Author: Kai Krakow <kai@kaishome.de>
Date:   Sun Nov 24 14:01:02 2024 +0100
```

In any case, I installed this driver, but it seems that the vibration is still not working.
```dmesg
[ 2609.753390] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0028/input/input80
[ 2609.753836] hid-generic 0005:045E:0B13.0028: input,hidraw0: BLUETOOTH HID v0.23 Gamepad [Xbox Wireless Controller] on c8:58:c0:23:a0:27
[ 2609.758081] loaded hid-xpadneo 0.9.r178.g10274c0.0.g10274c0
[ 2609.797562] xpadneo 0005:045E:0B13.0028: classic Bluetooth firmware version 0.23, please upgrade for better stability
[ 2609.797572] xpadneo 0005:045E:0B13.0028: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
[ 2609.797576] xpadneo 0005:045E:0B13.0028: working around wrong SDL2 mappings (changed version from 0x00000023 to 0x00001130)
[ 2609.797580] xpadneo 0005:045E:0B13.0028: report descriptor size: 283 bytes
[ 2609.797583] xpadneo 0005:045E:0B13.0028: fixing up Rx axis
[ 2609.797585] xpadneo 0005:045E:0B13.0028: fixing up Ry axis
[ 2609.797586] xpadneo 0005:045E:0B13.0028: fixing up Z axis
[ 2609.797588] xpadneo 0005:045E:0B13.0028: fixing up Rz axis
[ 2609.797589] xpadneo 0005:045E:0B13.0028: fixing up button mapping
[ 2609.797816] xpadneo 0005:045E:0B13.0028: gamepad detected
[ 2609.797819] xpadneo 0005:045E:0B13.0028: enabling compliance with Linux Gamepad Specification
[ 2609.797895] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0028/input/input81
[ 2609.798298] xpadneo 0005:045E:0B13.0028: input,hidraw0: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on c8:58:c0:23:a0:27
[ 2609.798375] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B13.0028/input/input82
[ 2609.798501] xpadneo 0005:045E:0B13.0028: consumer control added
[ 2609.798544] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B13.0028/input/input83
[ 2609.798584] xpadneo 0005:045E:0B13.0028: keyboard added
[ 2609.798589] xpadneo 0005:045E:0B13.0028: controller quirks: 0x00000050
[ 2609.798592] xpadneo xpadneo_welcome_rumble start
[ 2610.789328] xpadneo xpadneo_welcome_rumble took 991ms
[ 2610.789333] xpadneo 0005:045E:0B13.0028: Xbox Wireless Controller [6a:07:14:8f:07:7b] connected
[ 2610.803993] xpadneo 0005:045E:0B13.0028: reverting to original version (changed version from 0x00001130 to 0x00000023)
[ 2610.803996] xpadneo 0005:045E:0B13.0028: reverting to original product (changed PID from 0x028E to 0x0B13)
[ 2610.863509] xpadneo 0005:045E:0B13.0028: classic Bluetooth firmware version 0.23, please upgrade for better stability
[ 2610.863518] xpadneo 0005:045E:0B13.0028: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
[ 2610.863521] xpadneo 0005:045E:0B13.0028: working around wrong SDL2 mappings (changed version from 0x00000023 to 0x00001130)
[ 2610.863525] xpadneo 0005:045E:0B13.0028: report descriptor size: 283 bytes
[ 2610.863528] xpadneo 0005:045E:0B13.0028: fixing up Rx axis
[ 2610.863529] xpadneo 0005:045E:0B13.0028: fixing up Ry axis
[ 2610.863531] xpadneo 0005:045E:0B13.0028: fixing up Z axis
[ 2610.863532] xpadneo 0005:045E:0B13.0028: fixing up Rz axis
[ 2610.863534] xpadneo 0005:045E:0B13.0028: fixing up button mapping
[ 2610.863744] xpadneo 0005:045E:0B13.0028: gamepad detected
[ 2610.863746] xpadneo 0005:045E:0B13.0028: enabling compliance with Linux Gamepad Specification
[ 2610.863812] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0028/input/input84
[ 2610.863987] xpadneo 0005:045E:0B13.0028: input,hidraw0: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on c8:58:c0:23:a0:27
[ 2610.864030] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B13.0028/input/input85
[ 2610.864086] xpadneo 0005:045E:0B13.0028: consumer control added
[ 2610.864144] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B13.0028/input/input86
[ 2610.864220] xpadneo 0005:045E:0B13.0028: keyboard added
[ 2610.864226] xpadneo 0005:045E:0B13.0028: controller quirks: 0x00000050
[ 2610.864229] xpadneo xpadneo_welcome_rumble start
[ 2611.854511] xpadneo xpadneo_welcome_rumble took 990ms
[ 2611.854520] xpadneo 0005:045E:0B13.0028: Xbox Wireless Controller [6a:07:14:8f:07:7b] connected
```
Again, I'm not sure if I did something wrong. And the older gamepad was broken accidentally, the package of the latest usable driver was deleted. :(
The test results for `hidraw` remain unchanged.

kakra, 2024-11-25:
Well, according to this code:
```c
	if ((xdata->original_rsize == 283)
	    && OUI_MASK(xdata->gamepad->uniq[0], XPADNEO_OUI_MASK_GAMESIR_NOVA)
	    && ((xdata->quirks & XPADNEO_QUIRK_SIMPLE_CLONE) == 0)) {
		hid_info(hdev, "enabling heuristic GameSir Nova quirks\n");
		xdata->quirks |= XPADNEO_QUIRK_SIMPLE_CLONE;
	}
```

the following should happen, according to your dmesg log:

- your original descriptor size is 283, match
- the uniq ID (actually the MAC address) first byte should match the bitmask (see below)
- the quirk flags 0x50 masked by the need features should be 0
- so the code should set the required quirk flags

But here's where my brain messed up: `xdata->gamepad->uniq` isn't actually a byte array, it's a null terminated string. So I need to convert that first. Let me fix that.

kakra, 2024-11-25:
Please retry with 10274c0...fb5068e, the PR has been updated.

kakra, 2024-11-25:
> get the error like this:

Yep, this happens if you clone my fork directly because it has no tags. Use `git remote add kakra <my-fork-url>` in the main repository, run `git remote update`, then checkout the branch. The main repository has all the required tags.

BakaJzon, 2024-11-26:
```dmesg
[ 4452.629469] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0013/input/input92
[ 4452.629851] hid-generic 0005:045E:0B13.0013: input,hidraw2: BLUETOOTH HID v0.23 Gamepad [Xbox Wireless Controller] on c8:58:c0:23:a0:27
[ 4452.654370] loaded hid-xpadneo 0.9.r182.gfb5068e
[ 4452.692069] xpadneo 0005:045E:0B13.0013: classic Bluetooth firmware version 0.23, please upgrade for better stability
[ 4452.692080] xpadneo 0005:045E:0B13.0013: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
[ 4452.692085] xpadneo 0005:045E:0B13.0013: working around wrong SDL2 mappings (changed version from 0x00000023 to 0x00001130)
[ 4452.692090] xpadneo 0005:045E:0B13.0013: report descriptor size: 283 bytes
[ 4452.692093] xpadneo 0005:045E:0B13.0013: fixing up Rx axis
[ 4452.692096] xpadneo 0005:045E:0B13.0013: fixing up Ry axis
[ 4452.692097] xpadneo 0005:045E:0B13.0013: fixing up Z axis
[ 4452.692099] xpadneo 0005:045E:0B13.0013: fixing up Rz axis
[ 4452.692101] xpadneo 0005:045E:0B13.0013: fixing up button mapping
[ 4452.692299] xpadneo 0005:045E:0B13.0013: gamepad detected
[ 4452.692302] xpadneo 0005:045E:0B13.0013: enabling compliance with Linux Gamepad Specification
[ 4452.692361] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0013/input/input93
[ 4452.692849] xpadneo 0005:045E:0B13.0013: input,hidraw2: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on c8:58:c0:23:a0:27
[ 4452.692901] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B13.0013/input/input94
[ 4452.692945] xpadneo 0005:045E:0B13.0013: consumer control added
[ 4452.692981] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B13.0013/input/input95
[ 4452.693016] xpadneo 0005:045E:0B13.0013: keyboard added
[ 4452.693020] xpadneo 0005:045E:0B13.0013: bkjzon debug: size 283
[ 4452.693024] xpadneo 0005:045E:0B13.0013: bkjzon debug: kstrtou8 ffffffea
[ 4452.693026] xpadneo 0005:045E:0B13.0013: bkjzon debug: oui hex 0
[ 4452.693029] xpadneo 0005:045E:0B13.0013: bkjzon debug: mask 0
[ 4452.693031] xpadneo 0005:045E:0B13.0013: bkjzon debug: quirk hex 50
[ 4452.693033] xpadneo 0005:045E:0B13.0013: bkjzon debug: uniq[0] as hex 36 and char 6
[ 4452.693036] xpadneo 0005:045E:0B13.0013: bkjzon debug: uniq may be str 6a:07:14:8f:07:7b
[ 4452.693038] xpadneo 0005:045E:0B13.0013: bkjzon debug: force set oui to 0x6a
[ 4452.693040] xpadneo 0005:045E:0B13.0013: enabling heuristic GameSir Nova quirks
[ 4452.693042] xpadneo 0005:045E:0B13.0013: controller quirks: 0x00000056
[ 4452.693045] xpadneo xpadneo_welcome_rumble start
[ 4453.353115] xpadneo xpadneo_welcome_rumble took 660ms
[ 4453.353121] xpadneo 0005:045E:0B13.0013: Xbox Wireless Controller [6a:07:14:8f:07:7b] connected
[ 4453.370773] xpadneo 0005:045E:0B13.0013: reverting to original version (changed version from 0x00001130 to 0x00000023)
[ 4453.370778] xpadneo 0005:045E:0B13.0013: reverting to original product (changed PID from 0x028E to 0x0B13)
[ 4453.436017] xpadneo 0005:045E:0B13.0013: classic Bluetooth firmware version 0.23, please upgrade for better stability
[ 4453.436029] xpadneo 0005:045E:0B13.0013: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
[ 4453.436034] xpadneo 0005:045E:0B13.0013: working around wrong SDL2 mappings (changed version from 0x00000023 to 0x00001130)
[ 4453.436038] xpadneo 0005:045E:0B13.0013: report descriptor size: 283 bytes
[ 4453.436041] xpadneo 0005:045E:0B13.0013: fixing up Rx axis
[ 4453.436043] xpadneo 0005:045E:0B13.0013: fixing up Ry axis
[ 4453.436045] xpadneo 0005:045E:0B13.0013: fixing up Z axis
[ 4453.436047] xpadneo 0005:045E:0B13.0013: fixing up Rz axis
[ 4453.436049] xpadneo 0005:045E:0B13.0013: fixing up button mapping
[ 4453.436264] xpadneo 0005:045E:0B13.0013: gamepad detected
[ 4453.436267] xpadneo 0005:045E:0B13.0013: enabling compliance with Linux Gamepad Specification
[ 4453.436330] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0013/input/input96
[ 4453.436563] xpadneo 0005:045E:0B13.0013: input,hidraw2: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on c8:58:c0:23:a0:27
[ 4453.436601] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B13.0013/input/input97
[ 4453.436683] xpadneo 0005:045E:0B13.0013: consumer control added
[ 4453.436723] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B13.0013/input/input98
[ 4453.436790] xpadneo 0005:045E:0B13.0013: keyboard added
[ 4453.436795] xpadneo 0005:045E:0B13.0013: bkjzon debug: size 283
[ 4453.436799] xpadneo 0005:045E:0B13.0013: bkjzon debug: kstrtou8 ffffffea
[ 4453.436802] xpadneo 0005:045E:0B13.0013: bkjzon debug: oui hex 0
[ 4453.436805] xpadneo 0005:045E:0B13.0013: bkjzon debug: mask 0
[ 4453.436807] xpadneo 0005:045E:0B13.0013: bkjzon debug: quirk hex 50
[ 4453.436809] xpadneo 0005:045E:0B13.0013: bkjzon debug: uniq[0] as hex 36 and char 6
[ 4453.436813] xpadneo 0005:045E:0B13.0013: bkjzon debug: uniq may be str 6a:07:14:8f:07:7b
[ 4453.436815] xpadneo 0005:045E:0B13.0013: bkjzon debug: force set oui to 0x6a
[ 4453.436817] xpadneo 0005:045E:0B13.0013: enabling heuristic GameSir Nova quirks
[ 4453.436819] xpadneo 0005:045E:0B13.0013: controller quirks: 0x00000056
[ 4453.436822] xpadneo xpadneo_welcome_rumble start
[ 4454.096876] xpadneo xpadneo_welcome_rumble took 660ms
[... cut ...]
	hid_info(hdev, "bkjzon debug: uniq may be str %s\n", xdata->gamepad->uniq);
	hid_info(hdev, "bkjzon debug: force set oui to 0x6a");
	oui = 0x6a;

	if ((xdata->original_rsize == 283)
		// && (kstrtou8(xdata->gamepad->uniq, 16, &oui) == 0)
	    && OUI_MASK(oui, XPADNEO_OUI_MASK_GAMESIR_NOVA)
	    && ((xdata->quirks & XPADNEO_QUIRK_SIMPLE_CLONE) == 0)) {
		hid_info(hdev, "enabling heuristic GameSir Nova quirks\n");
		xdata->quirks |= XPADNEO_QUIRK_SIMPLE_CLONE;
	}
```
Now, the rumble works, while welcome rumble still takes about 20 seconds.

kakra, 2024-11-26:
Your debug output is very very helpful. Thanks a lot, it tells me all the values I need to know. Great work.

> Now, the rumble works, while welcome rumble still takes about 20 seconds.

I will look into it.

kakra, 2024-11-26:
Looks like
```
[ 4453.436799] xpadneo 0005:045E:0B13.0013: bkjzon debug: kstrtou8 ffffffea
```
means `-EINVAL` aka "invalid argument"... That's why `&oui` is not modified.

kakra, 2024-11-26:
Following the call trace, `strtou8()` actually trips over the `:` in the OUI although during my research it was documented somewhere that the parser ignores the first invalid char. I will fix this by copying the first two chars into an immediate buffer first. You're setting the OUI statically to `0x6a` which proves that the rest of the code properly works and then identifies your controller correctly. So we "just" need to fix the parsing bug.

kakra, 2024-11-27:
> Now, the rumble works, while welcome rumble still takes about 20 seconds.

@BakaJzon Does it properly stop if you add the no pulse quirk?
```c
#define XPADNEO_QUIRK_NO_PULSE 1
```

SherLock707, 2024-11-27:
My nova lite:
`Bus 001 Device 014: ID 3537:1041 Zikway HID gamepad` 

Connected via 2.5Ghz, light remains yellow and vibration does not work, in wired mode in Green or Red light vibration works.

Yellow mode:
```
fftest /dev/input/event25
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/event25 opened
Features:
  * Absolute axes: 
    [00 00 00 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: 
    Force feedback periodic effects: 
    [00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
  * Number of simultaneous effects: 0

Uploading effect #0 (Periodic sinusoidal) ... Error:: Function not implemented
Uploading effect #1 (Constant) ... Error: Function not implemented
Uploading effect #2 (Spring) ... Error: Function not implemented
Uploading effect #3 (Damper) ... Error: Function not implemented
Uploading effect #4 (Strong rumble, with heavy motor) ... Error: Function not implemented
Uploading effect #5 (Weak rumble, with light motor) ... Error: Function not implemented
Enter effect number, -1 to exit
```

PC:
```
Arch Linux x86_64
Linux 6.12.1-zen1-1-zen
Wayland
```

BakaJzon, 2024-11-27:
> > Now, the rumble works, while welcome rumble still takes about 20 seconds.
> 
> @BakaJzon Does it properly stop if you add the no pulse quirk?
> 
> ```c
> #define XPADNEO_QUIRK_NO_PULSE 1
> ```

ahh, yes!

BakaJzon, 2024-11-27:
@SherLock707 It seems that xpadneo currently only supports Bluetooth controllers and does not support wired or USB dongles...

SherLock707, 2024-11-27:
When I connect in dongle mode.

Green light turns yellow after pairing:
```
[ 8876.515480] usb 1-6: new full-speed USB device number 43 using xhci_hcd
[ 8876.640344] usb 1-6: unable to read config index 0 descriptor/start: -71
[ 8876.640346] usb 1-6: can't read configurations, error -71
[ 8877.227483] usb 1-6: new full-speed USB device number 44 using xhci_hcd
[ 8877.356047] usb 1-6: New USB device found, idVendor=3537, idProduct=1041, bcdDevice= 1.00
[ 8877.356051] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8877.356052] usb 1-6: Product: HID gamepad
[ 8877.356053] usb 1-6: Manufacturer: Zikway
[ 8877.359341] input: Zikway HID gamepad as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:3537:1041.003B/input/input69
[ 8877.410688] input: Zikway HID gamepad Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:3537:1041.003B/input/input70
[ 8877.410729] hid-generic 0003:3537:1041.003B: input,hidraw0: USB HID v1.11 Gamepad [Zikway HID gamepad] on usb-0000:00:14.0-6/input0
[ 8877.411738] input: Zikway HID gamepad Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:3537:1041.003C/input/input71
[ 8877.498551] hid-generic 0003:3537:1041.003C: input,hidraw1: USB HID v1.11 Keyboard [Zikway HID gamepad] on usb-0000:00:14.0-6/input1
[ 8877.499229] hid-generic 0003:3537:1041.003D: hiddev96,hidraw2: USB HID v1.11 Device [Zikway HID gamepad] on usb-0000:00:14.0-6/input2
```

kakra, 2024-11-27:
@SherLock707 Please stick to the other issue report with that topic, thanks. :-)

kakra, 2024-11-28:
> ahh, yes!

Then please retry with the updated PR. Remember to remove any quirk options from modprobe, otherwise the heuristic may not kick in.

BakaJzon, 2024-11-28:
[This line](https://github.com/kakra/xpadneo/commit/abf677c3bc64cfd2c93711ef590e7d15e1e8cb95#diff-8702be7a57fdc76797c2c127143f1745d75514767ff63af50ab253e7a1a07780R1111)
`    && (kstrtou8(xdata->gamepad->uniq, 16, &oui_byte) == 0)`
should be
`    && (kstrtou8(oui, 16, &oui_byte) == 0)`
otherwise rumble won't work. 
```dmesg
[  979.427457] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B13.0007/input/input53
[  979.427641] xpadneo 0005:045E:0B13.0007: keyboard added
[  979.427648] xpadneo 0005:045E:0B13.0007: strscpy(oui, xdata->gamepad->uniq, sizeof(oui)) == -7, -E2BIG == -7, equals? 1
[  979.427652] xpadneo 0005:045E:0B13.0007: oui ==  "6a"
[  979.427654] xpadneo 0005:045E:0B13.0007: kstrtou8(xdata->gamepad->uniq, 16, &oui_byte) == -22
[  979.427654] xpadneo 0005:045E:0B13.0007: oui_byte == 0x00
[  979.427656] xpadneo 0005:045E:0B13.0007: kstrtou8(oui, 16, &oui_byte) == 0
[  979.427658] xpadneo 0005:045E:0B13.0007: oui_byte == 0x6a
[  979.427664] xpadneo 0005:045E:0B13.0007: enabling heuristic GameSir Nova quirks
[  979.427666] xpadneo 0005:045E:0B13.0007: controller quirks: 0x00000057
```
After modifying this line, all things works perfect.

kakra, 2024-11-28:
> This line ... should be ...

Oh wow. Actually, I thought of this and then forgot. I shouldn't program late at night. Will fix and then merge. Thanks for the heads up.

closed 2024-11-28 by kakra

Gustavo17pacheco, 2024-12-23:
I have upgraded to 0.9.7, dmesg shows heuristics as being on but still no rumble. Even if I set the MAC address on /etc/modprobe.conf

JoseskVolpe, 2025-05-02:
It doesn't works for me, this is my MAC: 64:6f:9b:xx:xx:xx
It isn't listed in gamesir-nova.md, doing AND with this sets the mask as 0x20

JoseskVolpe, 2025-05-02:
Commit 3e08597d59f54002d9cfd39031fcc9feb30673c7 works

BakaJzon, 2025-05-02:
> +6

should be `+7`

The previous heuristic GameSir Nova quirks detection used two bits, `0x28`, to identify this controller.
Reducing it to a single bit, `0x20`, might cause false positives with other devices. Would this be the case?
If so, should we remove the heuristic logic and instead provide clear instructions in the documentation on how to modify the rules?

JoseskVolpe, 2025-05-02:
> I'll look into trying the "name quirk" again which we had in some very early versions. This requires that Gamesir properly identifies itself as such.

This controller uses "Xbox Wireless Controller" for it's name on XInput mode, so i don't think it'll work.
Vibration has no problems on Windows though, so they must use some trick.

JoseskVolpe, 2025-05-02:
¿Enabling the triggers rumbles by default won't work, right? (As i expect)

kakra, 2025-05-02:
> Vibration has no problems on Windows though, so they must use some trick.

The problem is: We are using the `ff-memless` force feedback driver which can emulate some force feedback effects (like sine, sawtooth, square etc) through rumble. This requires working haptic parameters (attack, decay, loop) which Gamesir doesn't support. Actually, I've not discovered any recent game that uses it. So the next dev branch v0.11 will drop `ff-memless` and support only native rumble commands.

This still doesn't support haptic parameters then which rumble can use but our heuristics will become much easier then. Firstly, games don't use haptics, they just send start and stop commands which Gamesir supports perfectly fine. Secondly, we can use a simple OUI list of devices which are known to support haptics (all genuine Microsoft controllers do). Thus, with v0.11, I will add support for haptic feedback, e.g. when pushing the triggers or buttons in mouse emulation mode, so you can "feel" the trigger points.

Also, force feedback effects like sine or sawtooth don't work well with rumble motors anyways because they are too inert to properly support it. Also, rumble isn't really a force working against you, it's just a vibration. Even the triggers do not support force, they only vibrate (which is different to the PS5 controllers, I think, which support a real force working against your finger).

But the most important issue for dropping `ff-memless` is an aliasing effect: The `ff-memless` driver is internally working with 20hz (or 50hz, I don't recall) resolution. The gamepad itself has a resolution of up to 125hz for sending commands (Windows seems to max out at 100hz) with a 10ms effect resolution. Games will probably update the gamepad at 30 or 60 hz. The Bluetooth firmware tends to crash in genuine Xbox controllers if we go faster than 100-125hz (so I need to additionally apply a limiter). None of those frequencies align, thus we get weird aliasing effects and rumble just doesn't feel right.

Without haptic parameters, games have to explicitly stop the motors - which all games seem to do. Otherwise, the motors may run for 10 or 20 minutes.

This will simplify things a lot: we can use a simple flag to denote haptic and trigger support. Until then, we have to rely on some quirky heuristics for Gamesir.

**TL;DR:** Even in Windows, Gamesir won't work correctly if any software relied on haptic rumble. It just works, because games don't care and start and stop the rumble motors manually in the game event loop. The xboxneo driver simply tests for this with the initial rumble - and if some quirk isn't set properly, it will either continue rumble or not rumble at all. This was good for planning a haptic feature but it will go away in v0.11.

kakra, 2025-05-02:
> This controller uses "Xbox Wireless Controller" for it's name on XInput mode, so i don't think it'll work. Other modes doesn't seems to support vibration.

Yeah, if `dmesg` shows this:
> `[ 6992.744887] xpadneo 0005:045E:0B20.0022: input,hidraw12: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on aa:bb:cc:dd:ee:ff`

then we cannot use the name quirk. Almost every third party device now identifies with that name (the name in square brackets is taken from the device info). That's probably why I dropped the name-based quirks.

To identify a device, we only have: OUI, name, HID descriptor. For Gamesir, the HID descriptor is a 1:1 copy of the genuine Microsoft descriptor (not all third-party controllers do that). So it has the same checksum and the same length (even including the weird NUL terminator which MS added and which should not be there).

And yes, only Xinput supports rumble. Dinput didn't have that. And other modes are probably for Switch or Playstation emulation, so xpadneo won't detect the device then. But I'm planing to support Switch mode because we have access to the gyro sensors then.

JoseskVolpe, 2025-05-02:
> Actually, I've not discovered any recent game that uses it. So the next dev branch v0.11 will drop `ff-memless` and support only native rumble commands.

¿So will v0.11 break for older games?

JoseskVolpe, 2025-05-02:
> Secondly, we can use a simple OUI list of devices which are known to support haptics (all genuine Microsoft controllers do)

¿How will it work for devices with completely random OUI like this one? It isn't a genuine Microsoft controller despite the GameSir partnership with Microsoft, but it would be recognized as such if using it's name to identify as one.