atar-axis/xpadneo #486: spams connect/disconnect when connecting via bluetooth.

jgilmore, 2024-07-27:
## Version of xpadneo
cgilmore@Marnie:~/xpadneo$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

as of filing this bug report. Where does it show the version number?

## Controller Model
Unknown

### Connection mode
- [ ] Bluetooth connection

## Installed Software
- [ ] Steam Input (enabled by default via Steam Desktop client)

## Protocol Information
Bluetooth

Please describe how it is failing below in the next sections.

## Severity / Impact

- [????where docs???? ] I've read the docs and the bug reporting instructions
- [NO ] I've applied the latest firmware update to the controller
- [ NO] I've tried disabling or running without above mentioned software
- [ YES] It does not work at all
- [NO ] It used to work in a previous version
- [ NO] It mostly works but sometimes it doesn't
- [ NO] I found a work-around

## Describe the Bug

Pairing, trusting, and connecting from the command line succeeds, but it then immediately disconnects.

If I connect using the "blueman applet" it connects, and then starts spamming connect/disconnect notifications. This continues until the controller's "flash quickly" mode times out.

Controller never rumbles, and remains at "flashes quickly" until it would have timed out anyway (as far as I can tell.)

## System Information
```console
Linux Marnie 6.8.0-38-generic #38-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun  7 15:25:01 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
Does not exist
```

## Controller and Bluetooth Information

## Additional Context
lsusb output is from when connected via USB.
[xpadneo-btmon.txt](https://github.com/user-attachments/files/16401410/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/user-attachments/files/16401411/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/user-attachments/files/16401412/xpadneo-lsusb.txt)

kakra, 2024-07-27:
> = Index Info: 80:00:0B:44:04:57 (Intel Corp.)                   [hci0] 0.064844

This Bluetooth chipset is known to not work at all with the older Xbox controller firmwares, and with the newer firmware it still shows instabilities.

See https://github.com/atar-axis/xpadneo/blob/master/docs/BT_DONGLES.md and https://github.com/atar-axis/xpadneo/blob/master/docs/TROUBLESHOOTING.md

guoyunhe, 2024-08-31:
I am also experiencing constantly disconnecting in recent days. It used to work fine.

Logs:

```
8月 31 19:15:31 amd-desktop kernel: xpadneo 0005:045E:0B13.0013: reverting to original version (changed version from 0x00001130 to 0x00000521)
8月 31 19:15:31 amd-desktop kernel: xpadneo 0005:045E:0B13.0013: reverting to original product (changed PID from 0x028E to 0x0B13)
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: BLE firmware version 5.21
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: working around wrong SDL2 mappings (changed version from 0x00000521 to 0x00001130)
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: report descriptor size: 283 bytes
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: fixing up Rx axis
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: fixing up Ry axis
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: fixing up Z axis
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: fixing up Rz axis
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: fixing up button mapping
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: enabling compliance with Linux Gamepad Specification
8月 31 19:15:34 amd-desktop kernel: input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0014/input/input54
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on a6:22:04:18:46:2f
8月 31 19:15:34 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: controller quirks: 0x00000050
8月 31 19:15:35 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: Xbox Wireless Controller [3c:fa:06:70:43:d6] connected
8月 31 19:15:35 amd-desktop kernel: input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input55
8月 31 19:17:41 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: reverting to original version (changed version from 0x00001130 to 0x00000521)
8月 31 19:17:41 amd-desktop kernel: xpadneo 0005:045E:0B13.0014: reverting to original product (changed PID from 0x028E to 0x0B13)
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: BLE firmware version 5.21
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: working around wrong SDL2 mappings (changed version from 0x00000521 to 0x00001130)
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: report descriptor size: 283 bytes
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: fixing up Rx axis
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: fixing up Ry axis
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: fixing up Z axis
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: fixing up Rz axis
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: fixing up button mapping
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: enabling compliance with Linux Gamepad Specification
8月 31 19:17:44 amd-desktop kernel: input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0015/input/input56
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on a6:22:04:18:46:2f
8月 31 19:17:44 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: controller quirks: 0x00000050
8月 31 19:17:45 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: Xbox Wireless Controller [3c:fa:06:70:43:d6] connected
8月 31 19:17:45 amd-desktop kernel: input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input57
8月 31 19:21:13 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: reverting to original version (changed version from 0x00001130 to 0x00000521)
8月 31 19:21:13 amd-desktop kernel: xpadneo 0005:045E:0B13.0015: reverting to original product (changed PID from 0x028E to 0x0B13)
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: BLE firmware version 5.21
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: working around wrong SDL2 mappings (changed version from 0x00000521 to 0x00001130)
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: report descriptor size: 283 bytes
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: fixing up Rx axis
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: fixing up Ry axis
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: fixing up Z axis
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: fixing up Rz axis
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: fixing up button mapping
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: enabling compliance with Linux Gamepad Specification
8月 31 19:21:17 amd-desktop kernel: input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0016/input/input58
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on a6:22:04:18:46:2f
8月 31 19:21:17 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: controller quirks: 0x00000050
8月 31 19:21:18 amd-desktop kernel: xpadneo 0005:045E:0B13.0016: Xbox Wireless Controller [3c:fa:06:70:43:d6] connected
8月 31 19:21:18 amd-desktop kernel: input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input59
```

guoyunhe, 2024-08-31:
I have similar issue with Bluetooth. But I found it only happens with Steam apps, with or without xpadneo. So I think it is probably not a bug of xpadneo, but something wrong with Steam.

guoyunhe, 2024-08-31:
Found a confirmed issue with Steam/Proton that can cause controller disconnect https://github.com/ValveSoftware/Proton/issues/7952

kakra, 2024-09-01:
There are several situations which can cause such issues:

- buggy controller firmware (dmesg will log if a known buggy version is detected, upgrade your controller firmware, this information only works reliable for genuine Microsoft controllers)
- incompatible Bluetooth chipset or buggy chipset firmware (try downgrading the chipset firmware package if it worked previously, otherwise get a supported chipset, there's a list in the docs folder)
- incompatible bluez settings or old bluez version (upgrade bluez, check our troublshooting docs)
- incompatible device flags as artifacts in the bluez device database (stop bluez, remove the device MAC from `/var/lib/bluetooth/HOST-MAC/DEVICE-MAC`, then reboot, pair the controller again)
- too busy air time of the bluetooth connection (we found that kdeconnect and derivatives can cause a lot of BT discovery spamming and destabilize the connection, it's probably still the controller firmware not handling this correctly but stopping kdeconnect can help, turning off other bluetooth devices, e.g. some TVs may also fix this)
- too fast sending of rumble commands causes a crash/race condition in the Xbox controller firmware (the controller supports up to 100 hz of HID commands but we found that rumble should stay below 50 hz or even 20 hz, this is an issue partially with how ff-memless works, partially with games using hidraw, SDL ships with a protection when using hidraw, xpadneo ships with a protection when using ff-memless, future xpadneo version will bypass ff-memless)

Please also check our troubleshooting docs in additional to all issues collected under https://github.com/users/atar-axis/projects/2. Unfortunately, Github migrated all the issues to the wrong categories but it is still helpful.

jgould22, 2024-12-22:
This seems a lot like what is described here  https://github.com/bluez/bluez/issues/605

kakra, 2024-12-22:
Well, the last comments point to UHID and we already check and recommend enabling it in the installer.

closed 2025-02-06 by github-actions[bot]