atar-axis/xpadneo #489: Can't connect more then one xbox controller(?)

sprokkel78, 2024-08-05:
After linking the first xbox-controller, the second xbox-controller is not visible on bluetoothctl.

kakra, 2024-08-05:
It should be, I connected around 6 different controllers, and they can all be connected at the same time.

sprokkel78, 2024-08-06:
Hmm, only one of my three xbox controllers gets recognized by bluetoothctl. How is that possible?

kakra, 2024-08-06:
Please inspect the output of `bluetoothctl` while connecting the controllers.

If only one and always the same controller gets recognized, chances are that the others are simply not paired correctly, not paired at all, or didn't boot in Bluetooth mode. You can switch Xbox controllers between Xbox and Bluetooth mode by double-clicking the connect button.

sprokkel78, 2024-08-06:
That might be the case. But I can't pair them because I can't see the mac-address or any other indiciation to an xbox controller in bluetootctl with scan on.

kakra, 2024-08-06:
Then it is not announcing Bluetooth. Try double-clicking the connect button while it blinks and see if it shows up. xpadneo supports any amount of controllers.

kakra, 2024-12-05:
Press and hold the connect button for 5 seconds, the Guide button should start flashing in a faster mode, indicating that it's ready to pair. It should show up in your Bluetooth list then.

If it doesn't, pair the controller to a different device first, a smartphone will do.

Then repeat the steps for your PC.

sprokkel78, 2024-12-05:
Thanks for the replay, but it isn't working for me.  The specific
xbox-controllers that don't pair also don't show up in bluetooth settings
or bluetoothctl on linux, but I can pair them to the xbox one x without
problems.

kakra, 2024-12-05:
Do your other controllers support Bluetooth? Pre Xbox One S, controllers didn't support Bluetooth.

And if they do but do not show up, there's nothing we can do about it in xpadneo. I can only advice to ask in our Discord community, maybe someone had a similar problem and fixed it, or head over to the bluez project an ask there. But it's not an issue in xpadneo itself, I'm sorry.

sprokkel78, 2024-12-05:
Oke thanks. Must be bluez then.  I've filed a question at bluez previous
month ago, nothing so far.

sprokkel78, 2024-12-05:
yes all my xbox controllers are bluetooth versions.

closed 2024-12-05 by kakra