atar-axis/xpadneo #490: Gulikit KK3 Max rumble issues

rishabh-os, 2024-08-08:
## Version of xpadneo
v0.9-172-gfd0cd48

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [x] Other: Gulikit KK3 Max

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [x] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [x] I don't use native games or did not try
- [x] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [x] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [x] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [x] I found a work-around
- [x] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [x] I can code and I want to help

## Describe the Bug
I am able to use the controller in BT mode without xpadneo. However, rumble does not work. In an attempt to get rumble working, I installed xpadneo. However, upon connecting to the controller, it keeps rumbling forever and does not stop.

Using the fixes in #469, I am able to get the infinite rumble to stop. However, rumble continues to not work in Steam games. 

The rumble functionality only seems to work if I connect the controller with the included USB dongle.

## Steps to Reproduce
 1. Install xpadneo as indicated in the README.
 2. Pair/connect the controller in Windows (X-input) mode

## Expected Behavior
I expect the controller to only vibrate for the welcome vibration and not vibrate forever. I also expect the vibration to work in games.

## System Information

```console
# uname -a
Linux rishabh-os 6.10.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 04 Aug 2024 05:11:32 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01  u.................%.
000000b4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000000c8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
000000dc: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
000000f0: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000104: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000118: 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75  .u........... ..&..u
0000012c: 08 95 01 81 02 c0                                            ......
3445511648 1458

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/user-attachments/files/16552989/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/user-attachments/files/16552990/xpadneo-dmesg.txt)

[xpadneo-lsusb.txt](https://github.com/user-attachments/files/16553013/xpadneo-lsusb.txt)

## Additional Context

### Commands sent by `misc/examples/c_hidraw`

The interface for the program is as follows:
```
--[  MotorEnable  ]---------------[ Strength in % ][  10ms ][cnt]--
03  1   1   1   1   0   0   0   0  040 030 020 020  005 005  003
|   |   |   |                   |   |   |   |    |   |    |
STR WEA LTR RTR  -   -   -   -  STR WEA LTR RTR  SUS REL  LOP
[1] [2] [3] [4]                 q/a w/s e/d r/f  h/j k/l  u/i    <-- keys, or Enter = SEND, Ctrl+C = ABORT
```
The controller vibrates when only when the 3rd and 4th bit are set: corresponding to LTR and RTR.
However, the corresponding strength values for LTR and RTR have no effect. Instead, the strength values that are valid are STR and WEA. The values related to the timings (SUS, REL, LOP) seem to have no effect as far as I can tell.

So, the following combination will set the controller to vibrate forever: 
```
03  0   0   1   1   0   0   0   0  001 001 000 000  000 000  000
```
And this will stop it:
```
03  0   0   1   1   0   0   0   0  000 000 000 000  000 000  000
```

### Other controller modes

The controller is not detected by the OS as a controller when connected in Android/iOS mode despite being connected via Bluetooth. Installing xpadneo fixes this issue and the controller works flawlessly, as far as I can tell, the welcome vibration plays correctly and rumble works in games as well. So, as a workaround, this mode can be used.

The controller also has a Switch mode, which xpadneo does not support, of course. But the rumble issue is present in this mode also. The controller however is detected by the OS in this case.

kakra, 2024-08-08:
So this looks like this is another controller which has the motor enable bits swapped or in reverse order. Can you please check if LTR corresponds to STR or WEA (aka testing it in isolation).

Also, the controller for sure ignores the haptic timing parameters. We already have a quirk bit for it.

rishabh-os, 2024-08-09:
Apologies, should have made it clear in the initial post. LTR corresponds to STR and RTR corresponds to WEA.

I tried setting the quirk bit for pulse parameters, and the issues remain.

0xInstrumentarium, 2024-09-06:
I also have this issue, the Android mode works but the Backpaddle won't.

kakra, 2024-09-22:
> Apologies, should have made it clear in the initial post. LTR corresponds to STR and RTR corresponds to WEA.

Those clones have all sorts of funny quirks. So this one swaps the first two with the last two bits (3,4,1,2). We don't have a quirks flag for that yet, we only have one to reverse the bit mask (4,3,2,1). What a mess. I'll look into that.

RX0FA, 2024-11-16:
Any update? btw same problem happens with the same controller with firmware 5.0

RX0FA, 2024-11-30:
@kakra , I've built the branch from #510 by using `sudo ./install.sh` on Arch Linux with kernel 6.12.1, installation succeeds, hid_xpadneo appears in `lsmod` after reboot, rumble works only when the KK3 Max is paired, then the rumble no longer works, I've also tested an original Xbox controller and rumble works as expected.

Here is how it looks on Plasma when both controllers are connected:
[image]

kakra, 2024-11-30:
You mean the controller rumbles while connecting it? (three short rumbles per each strong motor, weak motor, triggers)

But then in a game, it no longer works?

Please run `dmesg -w` while connecting the controller so we get a log of the process.

RX0FA, 2024-11-30:
> You mean the controller rumbles while connecting it? (three short rumbles per each strong motor, weak motor, triggers) But then in a game, it no longer works?

Yes, exactly.

> Please run dmesg -w while connecting the controller so we get a log of the process.

```
[  331.177385] hid-generic 0005:045E:02E0.0007: unknown main item tag 0x0
[  331.177529] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input24
[  331.177669] hid-generic 0005:045E:02E0.0007: input,hidraw6: BLUETOOTH HID v9.03 Gamepad [GuliKit Controller XW] on e0:e1:a9:...
[  331.183106] loaded hid-xpadneo pr510
[  331.277219] xpadneo 0005:045E:02E0.0007: buggy firmware detected, please upgrade to the latest version
[  331.277224] xpadneo 0005:045E:02E0.0007: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[  331.277226] xpadneo 0005:045E:02E0.0007: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[  331.277229] xpadneo 0005:045E:02E0.0007: report descriptor size: 307 bytes
[  331.277230] xpadneo 0005:045E:02E0.0007: fixing up report descriptor size
[  331.277299] xpadneo 0005:045E:02E0.0007: battery detected
[  331.277301] xpadneo 0005:045E:02E0.0007: gamepad detected
[  331.277303] xpadneo 0005:045E:02E0.0007: enabling compliance with Linux Gamepad Specification
[  331.277340] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input25
[  331.277445] xpadneo 0005:045E:02E0.0007: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on e0:e1:a9:...
[  331.277473] input: GuliKit Controller XW Consumer Control as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input26
[  331.277504] xpadneo 0005:045E:02E0.0007: consumer control added
[  331.277529] input: GuliKit Controller XW Keyboard as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input27
[  331.277561] xpadneo 0005:045E:02E0.0007: keyboard added
[  331.277563] xpadneo 0005:045E:02E0.0007: controller quirks: 0x00000103
[  331.277565] xpadneo xpadneo_welcome_rumble start
[  331.937974] xpadneo xpadneo_welcome_rumble took 660ms
[  331.937978] xpadneo 0005:045E:02E0.0007: GuliKit Controller XW [98:b6:ec:...] connected
[  332.004971] xpadneo 0005:045E:02E0.0007: reverting to original version (changed version from 0x00001130 to 0x00000903)
[  332.004976] xpadneo 0005:045E:02E0.0007: reverting to original product (changed PID from 0x028E to 0x02E0)
[  332.157321] xpadneo 0005:045E:02E0.0007: buggy firmware detected, please upgrade to the latest version
[  332.157325] xpadneo 0005:045E:02E0.0007: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[  332.157327] xpadneo 0005:045E:02E0.0007: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[  332.157330] xpadneo 0005:045E:02E0.0007: report descriptor size: 307 bytes
[  332.157332] xpadneo 0005:045E:02E0.0007: fixing up report descriptor size
[  332.157406] xpadneo 0005:045E:02E0.0007: battery detected
[  332.157408] xpadneo 0005:045E:02E0.0007: gamepad detected
[  332.157410] xpadneo 0005:045E:02E0.0007: enabling compliance with Linux Gamepad Specification
[  332.157447] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input28
[  332.157563] xpadneo 0005:045E:02E0.0007: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on e0:e1:a9:...
[  332.157592] input: GuliKit Controller XW Consumer Control as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input29
[  332.157619] xpadneo 0005:045E:02E0.0007: consumer control added
[  332.157644] input: GuliKit Controller XW Keyboard as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input30
[  332.157671] xpadneo 0005:045E:02E0.0007: keyboard added
[  332.157674] xpadneo 0005:045E:02E0.0007: controller quirks: 0x00000103
[  332.157676] xpadneo xpadneo_welcome_rumble start
[  332.818091] xpadneo xpadneo_welcome_rumble took 660ms
[  332.818095] xpadneo 0005:045E:02E0.0007: GuliKit Controller XW [98:b6:ec:...] connected
```

kakra, 2024-11-30:
Please retry with the updated PR. It should print information about the test rumbles and also fixes a few minor bugs with those. I wonder if you still get the test rumble.

If you do, the problem is with games. Do you run your games with Steam? You may need to check the access permissions of the hidraw devices. `dmesg -w` will output the hidraw device number like so:
```
[  331.277445] xpadneo 0005:045E:02E0.0007: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on e0:e1:a9:...
                                                  ^^^^^^^
```

In this case, run `getfacl /dev/hidraw6`. The number may change with each reconnect.

RX0FA, 2024-11-30:
Installed new version and after reboot, the controller is back to having the rumble on all the time.

Here is the commit I've installed:
```
$ git log -n 1
commit 8cb3057242365df21b0a344691834c7959f31b78 (HEAD -> fixes/issue-490, tag: pr510v2, origin/fixes/issue-490)
```

The dmesg output shows
```
[  101.620804] hid-generic 0005:045E:02E0.0007: unknown main item tag 0x0
[  101.620948] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input24
[  101.621082] hid-generic 0005:045E:02E0.0007: input,hidraw6: BLUETOOTH HID v9.03 Gamepad [GuliKit Controller XW] on e0:e1:a9:...
[  101.626732] loaded hid-xpadneo pr510v2
[  101.720632] xpadneo 0005:045E:02E0.0007: buggy firmware detected, please upgrade to the latest version
[  101.720638] xpadneo 0005:045E:02E0.0007: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[  101.720640] xpadneo 0005:045E:02E0.0007: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[  101.720642] xpadneo 0005:045E:02E0.0007: report descriptor size: 307 bytes
[  101.720644] xpadneo 0005:045E:02E0.0007: fixing up report descriptor size
[  101.720713] xpadneo 0005:045E:02E0.0007: battery detected
[  101.720715] xpadneo 0005:045E:02E0.0007: gamepad detected
[  101.720717] xpadneo 0005:045E:02E0.0007: enabling compliance with Linux Gamepad Specification
[  101.720756] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input25
[  101.720848] xpadneo 0005:045E:02E0.0007: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on e0:e1:a9:...
[  101.720877] input: GuliKit Controller XW Consumer Control as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input26
[  101.720904] xpadneo 0005:045E:02E0.0007: consumer control added
[  101.720929] input: GuliKit Controller XW Keyboard as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input27
[  101.720956] xpadneo 0005:045E:02E0.0007: keyboard added
[  101.720959] xpadneo 0005:045E:02E0.0007: controller quirks: 0x00000103
[  101.720961] xpadneo xpadneo_welcome_rumble start
[  101.720963] xpadneo 0005:045E:02E0.0007: testing weak motor: sustain 00ms release 00ms loop 0 wait 30ms
[  102.051172] xpadneo 0005:045E:02E0.0007: testing strong motor: sustain 00ms release 00ms loop 0 wait 30ms
[  102.381380] xpadneo xpadneo_welcome_rumble took 660ms
[  102.381384] xpadneo 0005:045E:02E0.0007: GuliKit Controller XW [98:b6:ec:...] connected
[  102.441686] xpadneo 0005:045E:02E0.0007: reverting to original version (changed version from 0x00001130 to 0x00000903)
[  102.441692] xpadneo 0005:045E:02E0.0007: reverting to original product (changed PID from 0x028E to 0x02E0)
[  102.567344] xpadneo 0005:045E:02E0.0007: buggy firmware detected, please upgrade to the latest version
[  102.567349] xpadneo 0005:045E:02E0.0007: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[  102.567351] xpadneo 0005:045E:02E0.0007: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[  102.567354] xpadneo 0005:045E:02E0.0007: report descriptor size: 307 bytes
[  102.567356] xpadneo 0005:045E:02E0.0007: fixing up report descriptor size
[  102.567424] xpadneo 0005:045E:02E0.0007: battery detected
[  102.567427] xpadneo 0005:045E:02E0.0007: gamepad detected
[  102.567428] xpadneo 0005:045E:02E0.0007: enabling compliance with Linux Gamepad Specification
[  102.567467] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input28
[  102.567602] xpadneo 0005:045E:02E0.0007: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on e0:e1:a9:...
[  102.567646] input: GuliKit Controller XW Consumer Control as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input29
[  102.567702] xpadneo 0005:045E:02E0.0007: consumer control added
[  102.567743] input: GuliKit Controller XW Keyboard as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input30
[  102.567795] xpadneo 0005:045E:02E0.0007: keyboard added
[  102.567800] xpadneo 0005:045E:02E0.0007: controller quirks: 0x00000103
[  102.567803] xpadneo xpadneo_welcome_rumble start
[  102.567804] xpadneo 0005:045E:02E0.0007: testing weak motor: sustain 00ms release 00ms loop 0 wait 30ms
[  102.898017] xpadneo 0005:045E:02E0.0007: testing strong motor: sustain 00ms release 00ms loop 0 wait 30ms
[  103.228231] xpadneo xpadneo_welcome_rumble took 660ms
[  103.228235] xpadneo 0005:045E:02E0.0007: GuliKit Controller XW [98:b6:ec:...] connected
[  117.178493] xpadneo 0005:045E:02E0.0007: reverting to original version (changed version from 0x00001130 to 0x00000903)
[  117.178501] xpadneo 0005:045E:02E0.0007: reverting to original product (changed PID from 0x028E to 0x02E0)
```

The getfacl output (hidraw6 / Gulikit KK3 Max)
```
# owner: root
# group: root
user::---
group::---
other::---
```

It's a bit odd that getfacl shows the same output for my Xbox controller, but the rumble works just fine with the Xbox controller.

btw I'm not using Steam at all, I can verify the rumble functionality with some retro games, or by using `https://hardwaretester.com/gamepad` , on that site I can trigger the rumble on demand, it works fine with my Xbox controller and hid_xpadneo.

kakra, 2024-12-01:
> Installed new version and after reboot, the controller is back to having the rumble on all the time.

Please elobarate what you mean by this.

RX0FA, 2024-12-01:
Sure, basically I cloned the project and checked out into the respective branch, these are the commands I've used
```bash
cd ~/Desktop
git clone https://github.com/kakra/xpadneo.git
cd xpadneo
git checkout fixes/issue-490
git tag pr510v2
sudo ./uninstall.sh
sudo ./install.sh
sudo reboot
```

kakra, 2024-12-01:
I'd rather asked for "is back to having rumble on all the time"... You mean directly after connection? Or do you mean it now works in games?

RX0FA, 2024-12-01:
The controller's rumble is activated immediately after connection.

kakra, 2024-12-01:
I wonder if the initially reported result of the hidraw tester are really true. Can you try that and check how each masking bit responds to each rumble motor? The behavior should be like this: If the bit isn't set, the motor won't be updated, regardless the motor parameters.

RX0FA, 2024-12-03:
Hi, sorry for the delay. Here are the steps I followed:
* Connected the Gulikit controller via Bluetooth to the PC
* The controller does the initial test rumbles as usual (I only see 2 test rumbles on dmesg unlike my Xbox controller which shows 3 tests)
* The rumble then keeps going on indefinitely (does not stop after 20 seconds)
* I ran misc/examples/c_hidraw anyway
* I tested all combinations from 0000 to 1111 and the controller rumble did not stop in any test

Here is what got printed from misc/examples/c_hidraw
```
03  0   0   0   0   0   0   0   0  040 030 020 020  005 005  003
03  0   0   0   1   0   0   0   0  040 030 020 020  005 005  003
03  0   0   1   0   0   0   0   0  040 030 020 020  005 005  003
03  0   0   1   1   0   0   0   0  040 030 020 020  005 005  003
03  0   1   0   0   0   0   0   0  040 030 020 020  005 005  003
03  0   1   0   1   0   0   0   0  040 030 020 020  005 005  003
03  0   1   1   0   0   0   0   0  040 030 020 020  005 005  003
03  0   1   1   1   0   0   0   0  040 030 020 020  005 005  003
03  1   0   0   0   0   0   0   0  040 030 020 020  005 005  003
03  1   0   0   1   0   0   0   0  040 030 020 020  005 005  003
03  1   0   1   0   0   0   0   0  040 030 020 020  005 005  003
03  1   0   1   1   0   0   0   0  040 030 020 020  005 005  003
03  1   1   0   0   0   0   0   0  040 030 020 020  005 005  003
03  1   1   0   1   0   0   0   0  040 030 020 020  005 005  003
03  1   1   1   0   0   0   0   0  040 030 020 020  005 005  003
03  1   1   1   1   0   0   0   0  040 030 020 020  005 005  003
```

Just in case it helps, here are the test lines I saw on the Gulikit compared to the Xbox controller
```
# Gulikit
[ 2888.517030] xpadneo 0005:045E:02E0.0011: testing weak motor: sustain 00ms release 00ms loop 0 wait 30ms
[ 2888.847277] xpadneo 0005:045E:02E0.0011: testing strong motor: sustain 00ms release 00ms loop 0 wait 30ms

# Xbox
[ 3285.862530] xpadneo 0005:045E:0B20.0012: testing weak motor: sustain 50ms release 50ms loop 2 wait 30ms
[ 3286.192779] xpadneo 0005:045E:0B20.0012: testing strong motor: sustain 50ms release 50ms loop 2 wait 30ms
[ 3286.523028] xpadneo 0005:045E:0B20.0012: testing trigger motors: sustain 50ms release 50ms loop 2 wait 30ms
```

Also, a new firmware (version 5.1) got released today, I wonder if it's wise to update `https://www.gulikit.com/newsinfo/887152.html`

kakra, 2024-12-03:
So I guess the rumble will stop if you set the motor strength explicitly to zero and send the command. In that case: Will this depend on the masking bits set?

- if you set bits 0000, does it stop the rumble with strength set to zero?
- if you set bits 0011, does it stop the rumble with strength set to zero?
- if you set bits 1100, does it stop the rumble with strength set to zero?

kakra, 2024-12-03:
> The controller does the initial test rumbles as usual (I only see 2 test rumbles on dmesg unlike my Xbox controller which shows 3 tests)

This is expected: Your model doesn't have trigger rumble.

RX0FA, 2024-12-03:
Test 1: if you set bits 0000, does it stop the rumble with strength set to zero?
Answer: No
Output:
```
03  0   0   0   0   0   0   0   0  000 000 000 000  005 005  003
```

Test 2: if you set bits 0011, does it stop the rumble with strength set to zero?
Answer: Yes (tested after after test 1 and tested again after reconnecting the controller)
Output
```
03  0   0   1   1   0   0   0   0  000 000 000 000  005 005  003
```

Test 3: if you set bits 1100, does it stop the rumble with strength set to zero?
Answer: No (The rumbles stopped with previous test, so I reconnected it to do this test)
Output:
```
03  1   1   0   0   0   0   0   0  000 000 000 000  005 005  003
```

kakra, 2024-12-04:
@RX0FA Sorry for the noise, I accidentally closed the issue report and PR via commit.

Your last comment looks very valuable but it confuses me even more because according to hidraw, your device needs the no-pulse quirk and the swapped-mask quirk. And that's what your quirk flags actually say.

I've added some debug code. Please install the current PR and load the module with `modprobe hid-xpadneo debug_hid=1`.

With debug enabled, xpadneo will print the rumble reports to dmesg. It will show the packet contents right before sent to the hardware. So you should observe the following during rumble tests by the module:

- with your Xbox controller, enable bits for the motors should match test instruction, magnitudes always set, only one command per test
- with your GuliKit controller, enable bits for the motors should mismatch the test instructions, magnitudes should match start and stopping of the motors, one command per start/stop

Let me see the output. I checked the code multiple times, I should not but I may have messed up somewhere.

kakra, 2024-12-04:
Also, you can try to replicate the testing steps with hidraw and - if you see any differences - test how to fix it.

RX0FA, 2024-12-04:
I have installed the module from the current PR, restarted the computer and reloaded the module with modprobe.

```
$ cat /sys/module/hid_xpadneo/parameters/debug_hid
Y
```

This is the dmesg output of the Xbox controller

```
[ 1719.534689] xpadneo 0005:045E:0B20.000A: BLE firmware version 5.22
[ 1719.534694] xpadneo 0005:045E:0B20.000A: pretending XB1S Windows wireless mode (changed PID from 0x0B20 to 0x028E)
[ 1719.534696] xpadneo 0005:045E:0B20.000A: working around wrong SDL2 mappings (changed version from 0x00000522 to 0x00001130)
[ 1719.534699] xpadneo 0005:045E:0B20.000A: report descriptor size: 283 bytes
[ 1719.534701] xpadneo 0005:045E:0B20.000A: fixing up Rx axis
[ 1719.534703] xpadneo 0005:045E:0B20.000A: fixing up Ry axis
[ 1719.534704] xpadneo 0005:045E:0B20.000A: fixing up Z axis
[ 1719.534705] xpadneo 0005:045E:0B20.000A: fixing up Rz axis
[ 1719.534707] xpadneo 0005:045E:0B20.000A: fixing up button mapping
[ 1719.534762] xpadneo 0005:045E:0B20.000A: gamepad detected
[ 1719.534764] xpadneo 0005:045E:0B20.000A: enabling compliance with Linux Gamepad Specification
[ 1719.534805] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B20.000A/input/input37
[ 1719.534922] xpadneo 0005:045E:0B20.000A: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on e0:e1:a9:...
[ 1719.534950] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B20.000A/input/input38
[ 1719.534976] xpadneo 0005:045E:0B20.000A: consumer control added
[ 1719.535002] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B20.000A/input/input39
[ 1719.535030] xpadneo 0005:045E:0B20.000A: keyboard added
[ 1719.535034] xpadneo 0005:045E:0B20.000A: controller quirks: 0x00000050
[ 1719.535035] xpadneo xpadneo_welcome_rumble start
[ 1719.535037] xpadneo 0005:045E:0B20.000A: testing weak motor: sustain 50ms release 50ms loop 2 wait 30ms
[ 1719.535039] xpadneo 0005:045E:0B20.000A: HID debug: len 9 rumble cmd 0x03 motors left 0 right 0 strong 0 weak 1 magnitude left 10 right 10 strong 20 weak 40 pulse sustain 50ms release 50ms loop 2
[ 1719.566110] Bluetooth: hci0: unexpected cc 0x0c7c length: 1 < 3
[ 1719.865262] xpadneo 0005:045E:0B20.000A: testing strong motor: sustain 50ms release 50ms loop 2 wait 30ms
[ 1719.865265] xpadneo 0005:045E:0B20.000A: HID debug: len 9 rumble cmd 0x03 motors left 0 right 0 strong 1 weak 0 magnitude left 10 right 10 strong 20 weak 40 pulse sustain 50ms release 50ms loop 2
[ 1720.195489] xpadneo 0005:045E:0B20.000A: testing trigger motors: sustain 50ms release 50ms loop 2 wait 30ms
[ 1720.195492] xpadneo 0005:045E:0B20.000A: HID debug: len 9 rumble cmd 0x03 motors left 1 right 1 strong 0 weak 0 magnitude left 10 right 10 strong 20 weak 40 pulse sustain 50ms release 50ms loop 2
[ 1720.525715] xpadneo xpadneo_welcome_rumble took 990ms
[ 1720.525718] xpadneo 0005:045E:0B20.000A: Xbox Wireless Controller [44:16:22:...] connected
```

Guilikit dmesg output

```
[ 2043.200498] xpadneo 0005:045E:02E0.000B: buggy firmware detected, please upgrade to the latest version
[ 2043.200503] xpadneo 0005:045E:02E0.000B: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[ 2043.200506] xpadneo 0005:045E:02E0.000B: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[ 2043.200508] xpadneo 0005:045E:02E0.000B: report descriptor size: 307 bytes
[ 2043.200510] xpadneo 0005:045E:02E0.000B: fixing up report descriptor size
[ 2043.200592] xpadneo 0005:045E:02E0.000B: battery detected
[ 2043.200595] xpadneo 0005:045E:02E0.000B: gamepad detected
[ 2043.200596] xpadneo 0005:045E:02E0.000B: enabling compliance with Linux Gamepad Specification
[ 2043.200639] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.000B/input/input40
[ 2043.200794] xpadneo 0005:045E:02E0.000B: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on e0:e1:a9:...
[ 2043.200825] input: GuliKit Controller XW Consumer Control as /devices/virtual/misc/uhid/0005:045E:02E0.000B/input/input41
[ 2043.200945] xpadneo 0005:045E:02E0.000B: consumer control added
[ 2043.201004] input: GuliKit Controller XW Keyboard as /devices/virtual/misc/uhid/0005:045E:02E0.000B/input/input42
[ 2043.201073] xpadneo 0005:045E:02E0.000B: keyboard added
[ 2043.201076] xpadneo 0005:045E:02E0.000B: controller quirks: 0x00000103
[ 2043.201079] xpadneo xpadneo_welcome_rumble start
[ 2043.201080] xpadneo 0005:045E:02E0.000B: testing weak motor: sustain 00ms release 00ms loop 0 wait 30ms
[ 2043.201082] xpadneo 0005:045E:02E0.000B: HID debug: len 9 rumble cmd 0x03 motors left 0 right 1 strong 0 weak 0 magnitude left 10 right 10 strong 20 weak 40 pulse sustain 0ms release 0ms loop 0
[ 2043.501327] xpadneo 0005:045E:02E0.000B: HID debug: len 9 rumble cmd 0x03 motors left 0 right 1 strong 0 weak 0 magnitude left 10 right 10 strong 20 weak 0 pulse sustain 0ms release 0ms loop 0
[ 2043.531356] xpadneo 0005:045E:02E0.000B: testing strong motor: sustain 00ms release 00ms loop 0 wait 30ms
[ 2043.531359] xpadneo 0005:045E:02E0.000B: HID debug: len 9 rumble cmd 0x03 motors left 1 right 0 strong 0 weak 0 magnitude left 10 right 10 strong 20 weak 40 pulse sustain 0ms release 0ms loop 0
[ 2043.831562] xpadneo 0005:045E:02E0.000B: HID debug: len 9 rumble cmd 0x03 motors left 1 right 0 strong 0 weak 0 magnitude left 10 right 10 strong 0 weak 40 pulse sustain 0ms release 0ms loop 0
[ 2043.861587] xpadneo xpadneo_welcome_rumble took 660ms
[ 2043.861590] xpadneo 0005:045E:02E0.000B: GuliKit Controller XW [98:b6:ec:...] connected
```

> Also, you can try to replicate the testing steps with hidraw and - if you see any differences - test how to fix it.

I've performed the same tests with the new driver, and results are exactly the same.
btw `./hidraw` does not trigger extra dmesg messages when the module is loaded with `debug_hid=1`

kakra, 2024-12-04:
> I've performed the same tests with the new driver, and results are exactly the same.
> btw `./hidraw` does not trigger extra dmesg messages when the module is loaded with `debug_hid=1`

Yes, because it uses HID directly on the raw level, bypassing any drivers.

So in the log, we can see that it sends a strong=0 and weak=0 to stop the corresponding motor, also setting the bit for left and right because your motor mask is swapped.

Can you test if `modprobe hid_debug=1 quirks=98:b6:ec:XX:YY:ZZ+4` changes anything?

RX0FA, 2024-12-05:
When the quirk is applied and the controller connected, the test rumbles are done as usual, then ~3 seconds later the controller rumbles one last time for some milliseconds. I can not see dmesg logs related to this last rumble, but based on the feeling and strength, it probably lasts 500ms and uses both motors.

As of right now, I'm happy with the result, unless you consider this last rumble to be indication of a bigger issue.

Here are the module params
```
$ cat /sys/module/hid_xpadneo/parameters/debug_hid
Y
$ cat /sys/module/hid_xpadneo/parameters/quirks
98:b6:ec:**:**:**+4
```

And the Gulikit dmesg output
```
[ 8206.597009] xpadneo 0005:045E:02E0.0016: buggy firmware detected, please upgrade to the latest version
[ 8206.597013] xpadneo 0005:045E:02E0.0016: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[ 8206.597016] xpadneo 0005:045E:02E0.0016: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[ 8206.597019] xpadneo 0005:045E:02E0.0016: report descriptor size: 307 bytes
[ 8206.597020] xpadneo 0005:045E:02E0.0016: fixing up report descriptor size
[ 8206.597109] xpadneo 0005:045E:02E0.0016: battery detected
[ 8206.597112] xpadneo 0005:045E:02E0.0016: gamepad detected
[ 8206.597113] xpadneo 0005:045E:02E0.0016: enabling compliance with Linux Gamepad Specification
[ 8206.597155] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.0016/input/input71
[ 8206.597273] xpadneo 0005:045E:02E0.0016: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on e0:e1:a9:**:**:**
[ 8206.597304] input: GuliKit Controller XW Consumer Control as /devices/virtual/misc/uhid/0005:045E:02E0.0016/input/input72
[ 8206.597332] xpadneo 0005:045E:02E0.0016: consumer control added
[ 8206.597360] input: GuliKit Controller XW Keyboard as /devices/virtual/misc/uhid/0005:045E:02E0.0016/input/input73
[ 8206.597386] xpadneo 0005:045E:02E0.0016: keyboard added
[ 8206.597390] xpadneo 0005:045E:02E0.0016: quirks added: 98:b6:ec:**:**:** flags 0x00000004
[ 8206.597392] xpadneo 0005:045E:02E0.0016: controller quirks: 0x00000107
[ 8206.597393] xpadneo xpadneo_welcome_rumble start
[ 8206.597395] xpadneo 0005:045E:02E0.0016: testing weak motor: sustain 00ms release 00ms loop 0 wait 30ms
[ 8206.597397] xpadneo 0005:045E:02E0.0016: HID debug: len 9 rumble cmd 0x03 motors left 1 right 1 strong 0 weak 0 magnitude left 0 right 0 strong 0 weak 40 pulse sustain 0ms release 0ms loop 0
[ 8206.897600] xpadneo 0005:045E:02E0.0016: HID debug: len 9 rumble cmd 0x03 motors left 1 right 1 strong 0 weak 0 magnitude left 0 right 0 strong 0 weak 0 pulse sustain 0ms release 0ms loop 0
[ 8206.927629] xpadneo 0005:045E:02E0.0016: testing strong motor: sustain 00ms release 00ms loop 0 wait 30ms
[ 8206.927632] xpadneo 0005:045E:02E0.0016: HID debug: len 9 rumble cmd 0x03 motors left 1 right 1 strong 0 weak 0 magnitude left 0 right 0 strong 20 weak 0 pulse sustain 0ms release 0ms loop 0
[ 8207.227837] xpadneo 0005:045E:02E0.0016: HID debug: len 9 rumble cmd 0x03 motors left 1 right 1 strong 0 weak 0 magnitude left 0 right 0 strong 0 weak 0 pulse sustain 0ms release 0ms loop 0
[ 8207.257863] xpadneo xpadneo_welcome_rumble took 660ms
[ 8207.257866] xpadneo 0005:045E:02E0.0016: GuliKit Controller XW [98:b6:ec:**:**:**] connected
```

kakra, 2024-12-05:
> I can not see dmesg logs related to this last rumble, but based on the feeling and strength, it probably lasts 500ms and uses both motors.

This is probably caused by the GuliKit firmware itself. I think I remember that it does this after Bluetooth connect. And that's why there's nothing in dmesg about it. I think I have a KK2 somewhere donated by the community. I can check it. But this check won't work for the quirk flags because yours clearly behaves differently, and I implemented support for the previous model with the donated hardware, and it worked properly.

That issue put aside, according to your tests it looks like the firmware of your controller is really messed up and somehow supports the motor enable bits but then again, it doesn't always obey them, and it would be better if we ship support for this by using "quirks=MAC:7". Can you update the PR? It fixes an issue with quirks overriding when heuristics apply. And then test with `modprobe debug_hid=1 quirks=MAC:7` (not `+`, just `:` to reset the quirks to value 7).

If it works, I'll leave the swap quirk in code, it seems useful, especially if a future KK model does things differently. I'll redo the PR then.

After adjusting the code, I'd like you to try a final test with no quirk flags in modprobe, and also test some games.

RX0FA, 2024-12-05:
I've pulled the latest changes from the PR and reinstalled the xpadneo, then loaded the module with `modprobe hid-xpadneo debug_hid=1 quirks=98:b6:ec:**:**:**:7`

I can confirm the rumble stops as expected.

Here is the dmesg output
```
[11359.150749] xpadneo 0005:045E:02E0.001F: buggy firmware detected, please upgrade to the latest version
[11359.150754] xpadneo 0005:045E:02E0.001F: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[11359.150757] xpadneo 0005:045E:02E0.001F: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[11359.150760] xpadneo 0005:045E:02E0.001F: report descriptor size: 307 bytes
[11359.150762] xpadneo 0005:045E:02E0.001F: fixing up report descriptor size
[11359.150844] xpadneo 0005:045E:02E0.001F: battery detected
[11359.150846] xpadneo 0005:045E:02E0.001F: gamepad detected
[11359.150848] xpadneo 0005:045E:02E0.001F: enabling compliance with Linux Gamepad Specification
[11359.150889] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.001F/input/input96
[11359.151002] xpadneo 0005:045E:02E0.001F: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on e0:e1:a9:**:**:**
[11359.151033] input: GuliKit Controller XW Consumer Control as /devices/virtual/misc/uhid/0005:045E:02E0.001F/input/input97
[11359.151057] xpadneo 0005:045E:02E0.001F: consumer control added
[11359.151085] input: GuliKit Controller XW Keyboard as /devices/virtual/misc/uhid/0005:045E:02E0.001F/input/input98
[11359.151108] xpadneo 0005:045E:02E0.001F: keyboard added
[11359.151111] xpadneo 0005:045E:02E0.001F: quirks override: 98:b6:ec:**:**:**
[11359.151113] xpadneo 0005:045E:02E0.001F: controller quirks: 0x00000007
[11359.151115] xpadneo xpadneo_welcome_rumble start
[11359.151116] xpadneo 0005:045E:02E0.001F: testing weak motor: sustain 00ms release 00ms loop 0 wait 30ms
[11359.151119] xpadneo 0005:045E:02E0.001F: HID debug: len 9 rumble cmd 0x03 motors left 0 right 0 strong 1 weak 1 magnitude left 0 right 0 strong 0 weak 40 pulse sustain 0ms release 0ms loop 0
[11359.451334] xpadneo 0005:045E:02E0.001F: HID debug: len 9 rumble cmd 0x03 motors left 0 right 0 strong 1 weak 1 magnitude left 0 right 0 strong 0 weak 0 pulse sustain 0ms release 0ms loop 0
[11359.481360] xpadneo 0005:045E:02E0.001F: testing strong motor: sustain 00ms release 00ms loop 0 wait 30ms
[11359.481362] xpadneo 0005:045E:02E0.001F: HID debug: len 9 rumble cmd 0x03 motors left 0 right 0 strong 1 weak 1 magnitude left 0 right 0 strong 20 weak 0 pulse sustain 0ms release 0ms loop 0
[11359.781566] xpadneo 0005:045E:02E0.001F: HID debug: len 9 rumble cmd 0x03 motors left 0 right 0 strong 1 weak 1 magnitude left 0 right 0 strong 0 weak 0 pulse sustain 0ms release 0ms loop 0
[11359.811591] xpadneo xpadneo_welcome_rumble took 660ms
[11359.811594] xpadneo 0005:045E:02E0.001F: GuliKit Controller XW [98:b6:ec:**:**:**] connected
```

kakra, 2024-12-05:
Thanks for testing. If it still works the same with the currently PR, then I consider this fixed and merge it.

RX0FA, 2024-12-06:
I installed the driver from the updated PR, restarted the computer, and connected the Gulikit controller, after the welcome rumbles, rumble stops as expected (i.e. works the same as before).

Also, I did some extra tests and it works well on all scenarios:
* Retro games from different platforms: rumble OK
* Native linux game: rumble OK
* Online gamepad tester: rumble OK

This is the dmesg output and the quirk now seems to be applied automatically.
```
[  199.572027] hid-generic 0005:045E:02E0.0007: unknown main item tag 0x0
[  199.572145] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input28
[  199.572269] hid-generic 0005:045E:02E0.0007: input,hidraw6: BLUETOOTH HID v9.03 Gamepad [GuliKit Controller XW] on e0:e1:a9:**:**:**
[  199.577720] loaded hid-xpadneo pr510v4
[  199.663637] xpadneo 0005:045E:02E0.0007: buggy firmware detected, please upgrade to the latest version
[  199.663642] xpadneo 0005:045E:02E0.0007: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[  199.663645] xpadneo 0005:045E:02E0.0007: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[  199.663648] xpadneo 0005:045E:02E0.0007: report descriptor size: 307 bytes
[  199.663651] xpadneo 0005:045E:02E0.0007: fixing up report descriptor size
[  199.663716] xpadneo 0005:045E:02E0.0007: battery detected
[  199.663718] xpadneo 0005:045E:02E0.0007: gamepad detected
[  199.663720] xpadneo 0005:045E:02E0.0007: enabling compliance with Linux Gamepad Specification
[  199.663759] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input29
[  199.663875] xpadneo 0005:045E:02E0.0007: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on e0:e1:a9:**:**:**
[  199.663914] input: GuliKit Controller XW Consumer Control as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input30
[  199.663966] xpadneo 0005:045E:02E0.0007: consumer control added
[  199.664005] input: GuliKit Controller XW Keyboard as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input31
[  199.664043] xpadneo 0005:045E:02E0.0007: keyboard added
[  199.664047] xpadneo 0005:045E:02E0.0007: controller quirks: 0x00000107
[  199.664050] xpadneo xpadneo_welcome_rumble start
[  199.664052] xpadneo 0005:045E:02E0.0007: testing weak motor: sustain 0ms release 0ms loop 0 wait 30ms
[  199.994501] xpadneo 0005:045E:02E0.0007: testing strong motor: sustain 0ms release 0ms loop 0 wait 30ms
[  200.325859] xpadneo xpadneo_welcome_rumble took 660ms
[  200.325863] xpadneo 0005:045E:02E0.0007: GuliKit Controller XW [98:b6:ec:**:**:**] connected
[  200.374472] xpadneo 0005:045E:02E0.0007: reverting to original version (changed version from 0x00001130 to 0x00000903)
[  200.374476] xpadneo 0005:045E:02E0.0007: reverting to original product (changed PID from 0x028E to 0x02E0)
[  200.520296] xpadneo 0005:045E:02E0.0007: buggy firmware detected, please upgrade to the latest version
[  200.520299] xpadneo 0005:045E:02E0.0007: pretending XB1S Windows wireless mode (changed PID from 0x02E0 to 0x028E)
[  200.520302] xpadneo 0005:045E:02E0.0007: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00001130)
[  200.520304] xpadneo 0005:045E:02E0.0007: report descriptor size: 307 bytes
[  200.520306] xpadneo 0005:045E:02E0.0007: fixing up report descriptor size
[  200.520367] xpadneo 0005:045E:02E0.0007: battery detected
[  200.520369] xpadneo 0005:045E:02E0.0007: gamepad detected
[  200.520370] xpadneo 0005:045E:02E0.0007: enabling compliance with Linux Gamepad Specification
[  200.520419] input: GuliKit Controller XW as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input32
[  200.520522] xpadneo 0005:045E:02E0.0007: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [GuliKit Controller XW] on e0:e1:a9:**:**:**
[  200.520550] input: GuliKit Controller XW Consumer Control as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input33
[  200.520577] xpadneo 0005:045E:02E0.0007: consumer control added
[  200.520603] input: GuliKit Controller XW Keyboard as /devices/virtual/misc/uhid/0005:045E:02E0.0007/input/input34
[  200.520634] xpadneo 0005:045E:02E0.0007: keyboard added
[  200.520637] xpadneo 0005:045E:02E0.0007: controller quirks: 0x00000107
[  200.520639] xpadneo xpadneo_welcome_rumble start
[  200.520640] xpadneo 0005:045E:02E0.0007: testing weak motor: sustain 0ms release 0ms loop 0 wait 30ms
[  200.850848] xpadneo 0005:045E:02E0.0007: testing strong motor: sustain 0ms release 0ms loop 0 wait 30ms
[  201.181060] xpadneo xpadneo_welcome_rumble took 660ms
[  201.181065] xpadneo 0005:045E:02E0.0007: GuliKit Controller XW [98:b6:ec:**:**:**] connected
[  221.748142] xpadneo 0005:045E:02E0.0007: throttling rumble reprogramming
```

I think this issue can be considered fixed. Thanks a lot for your support.

closed 2024-12-06 by kakra