atar-axis/xpadneo #491: Xpadneo not surviving kernel updates on Ubuntu 22.04 / Kernel 6.5.0?

ipkpjersi, 2024-08-10:
## Version of xpadneo
v0.9.6

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [x] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [x] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

Before, at least I think pre Linux kernel 6.x, it used to survive kernel updates and RPCS3 would continue to work properly with evdev mode. Now it requires a reinstall of xpadneo for evdev mode to start working properly again. The joysticks are out of range and the buttons are mapped incorrectly, etc after every kernel update.

## Steps to Reproduce

1. Install xpadneo 0.9.6.
2. Update kernel to latest version through apt-get update / apt-get dist-upgrade.

## Expected Behavior

I expect xpadneo to survive kernel updates like it used to and for RPCS3 to work as expected with evdev mode.

## Screenshots / GIFs / Videos

[image]

## System Information

```console
# uname -a
Linux fast-cottage 6.5.0-45-generic #45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Jul 15 16:40:02 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0
```

## Controller and Bluetooth Information

[xpadneo-lsusb.txt](https://github.com/user-attachments/files/16570374/xpadneo-lsusb.txt)
[xpadneo-dmesg.txt](https://github.com/user-attachments/files/16570375/xpadneo-dmesg.txt)
[xpadneo-btmon.txt.gz](https://github.com/user-attachments/files/16570376/xpadneo-btmon.txt.gz)

## Additional Context

The weird thing is, the controller works as expected with SDL mode, just not with evdev mode, evdev mode requires a reinstall of xpadneo:
[image]

[image]

If you need any additional information, please let me know.

kakra, 2024-08-11:
The hidraw device bypasses kernel drivers, user-space like SDL will read the raw commands from the device. So for such software, it will work regardless of xpadneo being installed.

This looks like our dkms integration may be somewhat broken. I'll look into it.

ipkpjersi, 2024-08-11:
I'm not sure if it helps but here's my `dkms status`:

```
sudo dkms status
hid-xpadneo/v0.9.6, 6.5.0-41-generic, x86_64: installed
hid-xpadneo/v0.9.6, 6.5.0-45-generic, x86_64: installed
nvidia/535.183.06, 6.5.0-41-generic, x86_64: installed
nvidia/535.183.06, 6.5.0-45-generic, x86_64: installed
openrazer-driver/3.8.0, 6.5.0-41-generic, x86_64: installed
openrazer-driver/3.8.0, 6.5.0-45-generic, x86_64: installed
zfs/2.2.5, 6.5.0-45-generic, x86_64: installed
```

It does have it installed for both my older and newer kernel, not sure if that matters, probably not.

kakra, 2024-12-18:
Please test #517

ipkpjersi, 2024-12-19:
I tried it, but it doesn't seem to install:
```
 ~/xpadneo-dkms-integration git:(rework/dkms-integration) sudo ./install.sh
sed: can't read lib/../VERSION: No such file or directory
ERROR: If you install from a git repository, you may need to run
ERROR: sudo git config --global --add safe.directory $PWD
ERROR: to trust this git checkout for the root user.
ERROR: See https://github.com/atar-axis/xpadneo/issues/346
WARNING: kernel uhid module not found, controller firmware 5.x will not be supported
* deploying DKMS package
fatal: No names found, cannot describe anything.
make: *** [Makefile:33: VERSION] Error 128
HINT: Try uninstalling xpadneo first
```

I have xpadneo uninstalled already.

ipkpjersi, 2024-12-19:
That worked, I installed that branch on my desktop and my Lenovo P14s Gen1 laptop, I won't know if it fixed it until there's a new kernel update though. I'll post here how it goes once I get a kernel update.

kakra, 2024-12-19:
> That worked, I installed that branch on my desktop and my Lenovo P14s Gen1 laptop, I won't know if it fixed it until there's a new kernel update though. I'll post here how it goes once I get a kernel update.

As far as I understand, you can reboot into an older kernel and see if it automatically rebuilds xpadneo for that kernel.

ipkpjersi, 2024-12-20:
I am on commit 28435e8, and just tested booting into an older kernel on my Linux Mint 22 laptop. I checked `sudo dkms status` and it looks like hid-xpadneo was not re-built for the older kernel version. I'm not sure if you're still working on this or if it's possible I actually do need to wait for upgrading to a new kernel, but either way, I just thought I would let you know.

kakra, 2024-12-20:
But dkms.conf in /var/lib/dkms/xpadneo...version should have autoinstall enabled... Could you check?

ipkpjersi, 2024-12-20:
It does seem to be enabled, I guess it just didn't work, or maybe I have to wait until there's a new kernel update.
```
cat /var/lib/dkms/hid-xpadneo/v0.9-200-g28435e8/source/dkms.conf 
# vim:syntax=bash

PACKAGE_NAME="hid-xpadneo"
PACKAGE_VERSION="v0.9-200-g28435e8"

BUILT_MODULE_NAME[0]="hid-xpadneo"
BUILT_MODULE_LOCATION[0]="src"
DEST_MODULE_LOCATION[0]="/kernel/drivers/hid"

MAKE[0]="make -C '${kernel_source_dir}' M='${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build/src' VERSION='${PACKAGE_VERSION}' modules"
CLEAN="make -C '${kernel_source_dir}' M='${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build/src' clean"

BUILD_EXCLUSIVE_KERNEL_MIN="4.18.0"
BUILD_EXCLUSIVE_CONFIG="CONFIG_HID CONFIG_INPUT_FF_MEMLESS CONFIG_POWER_SUPPLY"

AUTOINSTALL="Y"

POST_INSTALL="dkms.post_install"
POST_REMOVE="dkms.post_remove"
```

ipkpjersi, 2024-12-21:
For what it's worth, I just tried the latest commit 0304a5c, and for some reason both install.sh and uninstall.sh (both running as a normal user and running with sudo before the script name) no longer do anything for me. 

I'm not sure if I'm doing something wrong or what but just thought I would pass this along.

kakra, 2024-12-21:
> No output of any kind.

Yeah, I've seen this behavior myself randomly during testing. I've not found what's causing it: It just won't show an error message.

kakra, 2024-12-21:
@ipkpjersi Meh... Found it. What a sneaky bug: It depended on a success condition in the options parser and sneaked in during rebase. It should be fixed now.

kakra, 2024-12-21:
Please check with the latest PR, it should fix the autoinstall. You should be able to run `sudo dkms autoinstall`.

ipkpjersi, 2024-12-21:
It appears to be working, it looks like I am on commit 3b489bf, I rebooted into an older kernel, did `sudo dkms status` and it showed hid-xpadneo only for the current kernel, then I did `sudo dkms autoinstall`, it built the hid-xpadneo and now `sudo dkms status` also shows hid-xpadneo for the older kernel too. 

I can only assume that installing a newer kernel version fires off `sudo dkms autoinstall` which means this issue is likely now fixed.

kakra, 2024-12-21:
> I can only assume that installing a newer kernel version fires off `sudo dkms autoinstall` which means this issue is likely now fixed.

Yes, this should be how it works if I understand it correctly. Thanks for confirmation. Finally! \o/ yay

closed 2024-12-23 by kakra

ipkpjersi, 2025-01-17:
Not sure if this is worth mentioning, but, I just cloned the latest version from master branch and then tried running an uninstall so I could install the latest version, but the Makefile looks like it returned an error, not sure if it's a problem or not, but just thought I would mention it here.

Here are the logs:
```
git clone https://github.com/atar-axis/xpadneo
Cloning into 'xpadneo'...
remote: Enumerating objects: 5003, done.
remote: Counting objects: 100% (917/917), done.
remote: Compressing objects: 100% (278/278), done.
remote: Total 5003 (delta 721), reused 646 (delta 639), pack-reused 4086 (from 1)
Receiving objects: 100% (5003/5003), 20.85 MiB | 43.67 MiB/s, done.
Resolving deltas: 100% (3284/3284), done.

cd xpadneo
sudo ./uninstall.sh
* unloading current driver module
* looking for registered instances
found 1 registered instance(s) on your system
Module hid-xpadneo-v0.9-200-g28435e8 for kernel 6.8.0-50-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

hid-xpadneo.ko:
 - Uninstallation
   - Deleting from: /lib/modules/6.8.0-50-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

Running the post_remove script:
Uninstalling ERTM override...
Uninstalling modalias database...
Uninstalling udev rules...
Reloading udev...
depmod...
Module hid-xpadneo-v0.9-200-g28435e8 for kernel 6.8.0-51-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

hid-xpadneo.ko:
 - Uninstallation
   - Deleting from: /lib/modules/6.8.0-51-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

Running the post_remove script:
Uninstalling ERTM override...
Uninstalling modalias database...
Uninstalling udev rules...
Reloading udev...
depmod...
Deleting module hid-xpadneo-v0.9-200-g28435e8 completely from the DKMS tree.
rmdir: failed to remove '/usr/share/doc/xpadneo': No such file or directory
make: *** [Makefile:55: uninstall] Error 1
```

I then ran the uninstaller again, and I guess it seems fine:
```
sudo ./uninstall.sh
* unloading current driver module
modprobe: FATAL: Module hid_xpadneo not found.
* looking for registered instances
found 0 registered instance(s) on your system
```

Hopefully this rmdir was just some noise and it can be ignored safely.

Cheers.

ipkpjersi, 2025-01-17:
New comment instead of edits, I noticed that it looks like maybe 0.9.6 *was* technically surviving kernel updates according to DKMS maybe, it just wasn't working in rpcs3 for evdev only but SDL mode in rpcs3 was working fine, either way here is my DKMS status I noticed that I just checked on my secondary PC:
```
~/xpadneo-0.9.7  dkms status
hid-xpadneo/v0.9.6, 6.8.0-49-generic, x86_64: installed
hid-xpadneo/v0.9.6, 6.8.0-51-generic, x86_64: installed
```

Either way, I've installed 0.9.7 now and we'll see if this issue pops up again or if it's truly fixed.

kakra, 2025-01-17:
> Kernel preparation unnecessary for this kernel

This just means the kernel is new enough to properly support ERTM with Xbox controllers.

The new installer should be more robust and works a bit cleaner wrt/ kernel updates and rebuild as part of the boot process. I think surviving updates was never the core problem, we rather have a problem that some distributions do not properly boot if something fails during recompile by DKMS.

SDL mode (if you mean hidraw mode) should always work no matter the driver because xpadneo does not provide the hidraw device, neither does it do anything with Bluetooth, it is a HID driver only.