atar-axis/xpadneo #492: rtl8761b (tp link ub500) compatibility report

andrew-ld, 2024-08-31:
## Version of xpadneo
archlinux aur dkms 0.9.6-1

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

### Context

hi, i am opening this issue just to report the current status of compatibility between the xbox controller and the rtl8761b bluetooth receiver.

currently with the mainline kernel (6.10.6) and latest linux-firmware commit (2024-08-30)  it disconnect and reconnect randomly.

this problem is not of bluez, nor of xpadneo, nor of the hardware itself.

the problem is a regression with the latest firmware, you go back to a previous version it works perfectly. i put on this repo (https://github.com/andrew-ld/rtl8761b-firmware) the latest working version of the firmware.

i also uploaded a package on aur https://aur.archlinux.org/packages/rtl8761b-firmware

CosmicHeron, 2024-09-04:
Can confirm the same problems with the dongle. And ASUS BT500 one as well, since they use the same chip. Looking at Realtek's Open Source repository on GitHub, the last time the driver was updated was 5 years ago so I'm not sure if they'll ever go back and fix it?

Are there even any Bluetooth 5.0 dongles that work with Linux and *don't* use this chip?

puleglot, 2024-09-04:
I've submitted firmware taken from the latest Windows driver package. Works fine for me. Not sure it will be accepted though.
https://lore.kernel.org/linux-firmware/20240904132742.2861526-1-alexander@tsoy.me/T/#u

kakra, 2024-09-04:
> I've submitted firmware taken from the latest Windows driver package.

It probably doesn't work this way because firmware as part of driver packages is distributed under restrictive licenses. There's a similar problem with the xone driver which needs a firmware blob only available in the Xbox controller Windows driver.

Also, kernel devs cannot take it if it is not signed by official distributors.

But I pretty much like the approach of trying things out and confirming that some release fixes some bugs. This is a good confirmation of a known bug we can document.

andrew-ld, 2024-09-04:
at least kernel developers could revert to the latest working version, it seems to fix even other devices https://github.com/andrew-ld/rtl8761b-firmware/issues/1

puleglot, 2024-09-04:
> Also, kernel devs cannot take it if it is not signed by official distributors.

I've CC'ed an author of previous commits, so I hope to get some feedback at least.

closed 2024-12-21 by github-actions[bot]