atar-axis/xpadneo #494: Support for 8BitDo Ultimate Wired Controller for Xbox and Windows (Hall Effect Joystick)

simpliston, 2024-09-04:
## Version of xpadneo
0.9.6-1

## Controller Model

- [x] Other: 8BitDo Ultimate Wired Controller for Xbox and Windows (Hall Effect Joystick) https://shop.8bitdo.com/products/8bitdo-ultimate-wired-controller-for-xbox-and-windows

Vendor id `2dc8`, product id `2016`, id version `0114`, id bustype `0003`

### Connection mode

- [x] USB cable (not yet supported)

## Describe your feature request

- [x] I'd like to see support for a specific model

**Describe the solution you'd like**
I'd like to see xpadneo add the extra device id and needed quirks for that gamepad.

**Describe alternatives you've considered**
xpad, xboxdrv, [custom udev rules from this snippet](https://gist.github.com/ammuench/0dcf14faf4e3b000020992612a2711e2?permalink_comment_id=4817756)

kakra, 2024-09-22:
USB devices are not supported by xpadneo currently but we plan to do so in the future.

amarraff, 2024-10-24:
If this controller is supported, I'm hoping it fixes the issues with mine as well; I'm having an issue with the 8BitDo Ultimate C and 2C (found [here)](https://www.amazon.com/dp/B0D72TNXSC/?coliid=I2KJ0RJ9R9YQUR&colid=GIM2ZYNYHXMV&ref_=list_c_wl_lv_ov_lig_dp_it&th=1).

For both controllers, DirectInput mode works great.

When either controller is in xInput mode, LT inputs RS left/right, and RS left/right inputs RS up/down. So, there seems to be some issue with how xpadneo is interpreting the xInput signals that 8BitDo controllers send.

Worth noting that Steam corrects this issue, I haven't had any issues playing games with either controller in xInput mode. It's only with apps outside of Steam that try to read the xInput.

tr1xem, 2025-07-05:
Sir Anyprogress on this?

kakra, 2025-07-05:
> **Describe the solution you'd like**
> I'd like to see xpadneo add the extra device id and needed quirks for that gamepad.

Repeating: xpadneo does not support USB controllers currently, and it's not as easy as adding a device ID. If Xbox controllers connect via Bluetooth, they run as a HID device. xpadneo doesn't even support Bluetooth - it supports HID, and HID only. Controllers connected via USB are usually never HID devices. Xbox controllers use the GIP protocol. It's similar in structure but it operates on a completely different device bus in the kernel with incompatible interfaces.

You guys are currently probably using xpad or xboxdrv. You need to ask there to fix the mappings. You cannot bind xpadneo to this device, even not with udev rules.