atar-axis/xpadneo #497: Trying to set up xpadneo on Raspberry Pi 2 running buster with kernel 6.6

defanator, 2024-09-28:
## Version of xpadneo

```
root@raspberrypi:/ # dkms status
hid-xpadneo, v0.9-175-g70ef8ee, 6.6.51-v7+, armv7l: installed
```

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller (I guess - under the battery it says "Xbox Wireless Controller Model 1914")
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [x] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [x] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
[... 689 lines cut ...]

With the 6.6.51 kernel (current setup I'm struggling with) I am seeing this:
```
# systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2024-09-28 13:23:51 BST; 14min ago
     Docs: man:bluetoothd(8)
 Main PID: 410 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 2055)
   CGroup: /system.slice/bluetooth.service
           └─410 /usr/lib/bluetooth/bluetoothd

Sep 28 13:23:51 raspberrypi systemd[1]: Starting Bluetooth service...
Sep 28 13:23:51 raspberrypi bluetoothd[410]: Bluetooth daemon 5.50
Sep 28 13:23:51 raspberrypi systemd[1]: Started Bluetooth service.
Sep 28 13:23:51 raspberrypi bluetoothd[410]: Starting SDP server
Sep 28 13:23:51 raspberrypi bluetoothd[410]: Bluetooth management interface 1.22 initialized
Sep 28 13:23:51 raspberrypi bluetoothd[410]: Sap driver initialization failed.
Sep 28 13:23:51 raspberrypi bluetoothd[410]: sap-server: Operation not permitted (1)
Sep 28 13:23:51 raspberrypi bluetoothd[410]: Failed to set privacy: Rejected (0x0b)
Sep 28 13:31:31 raspberrypi bluetoothd[410]: input-hog profile accept failed for F4:6A:D7:FC:E2:4C
Sep 28 13:31:32 raspberrypi bluetoothd[410]: input-hog profile accept failed for F4:6A:D7:FC:E2:4C
```

defanator, 2024-09-28:
Here's the module build log + `modprobe` errors afterwards:

```
xx@raspberrypi:~/git/xpadneo $ sudo ./install.sh 
* creating dkms.conf
* adding hid-xpadneo-v0.9-175-g70ef8ee folder to /usr/src
* installing module (using DKMS)

Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee/source ->
                 /usr/src/hid-xpadneo-v0.9-175-g70ef8ee

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=6.6.51-v7+ -C /lib/modules/6.6.51-v7+/build M=/var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee/build/src VERSION=v0.9-175-g70ef8ee modules.......
cleaning build area....

DKMS: build completed.

hid-xpadneo.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.6.51-v7+/kernel/drivers/hid/

Running the post_install script:
Not disabling ERTM, kernel version doesn't require it...
Installing modalias database...
Installing udev rules...
Reloading udev...

depmod.......

DKMS: install completed.

xx@raspberrypi:~/git/xpadneo $ sudo dkms status
hid-xpadneo, v0.9-175-g70ef8ee, 6.6.51-v7+, armv7l: installed

xx@raspberrypi:~/git/xpadneo $ sudo modprobe -v hid-xpadneo
insmod /lib/modules/6.6.51-v7+/kernel/drivers/input/ff-memless.ko.xz 
insmod /lib/modules/6.6.51-v7+/kernel/drivers/hid/hid-xpadneo.ko.xz 
modprobe: ERROR: could not insert 'hid_xpadneo': Exec format error

xx@raspberrypi:~/git/xpadneo $ dmesg -T | grep -i -- 'xpadneo'
[Sat Sep 28 14:39:26 2024] module hid_xpadneo: .gnu.linkonce.this_module section size must match the kernel's built struct module size at run time
```

defanator, 2024-09-28:
I guess this is the issue - see `vermagic` still pointing to older kernel for some reason:

```
xx@raspberrypi:~/git/xpadneo $ modinfo hid-xpadneo
filename:       /lib/modules/6.6.51-v7+/kernel/drivers/hid/hid-xpadneo.ko.xz
version:        v0.9-175-g70ef8ee
description:    Linux kernel driver for Xbox ONE S+ gamepads (BT), incl. FF
author:         Kai Krakow <kai@kaishome.de>
author:         Florian Dollinger <dollinger.florian@gmx.de>
license:        GPL
srcversion:     2CDA9CB681685D13C24373F
alias:          hid:b0005g*v0000045Ep00000B13
alias:          hid:b0005g*v0000045Ep00000B22
alias:          hid:b0005g*v0000045Ep00000B05
alias:          hid:b0005g*v0000045Ep00000B20
alias:          hid:b0005g*v0000045Ep000002FD
alias:          hid:b0005g*v0000045Ep000002E0
depends:        ff-memless
name:           hid_xpadneo
vermagic:       5.10.103-v7+ SMP mod_unload modversions ARMv7 p2v8 
parm:           trigger_rumble_mode:(u8) Trigger rumble mode. 0: pressure, 2: disable. (byte)
parm:           rumble_attenuation:(u8) Attenuate the rumble strength: all[,triggers] 0 (none, full rumble) to 100 (max, no rumble). (array of byte)
parm:           ff_connect_notify:(bool) Connection notification using force feedback. 1: enable, 0: disable. (bool)
parm:           gamepad_compliance:(bool) Adhere to Linux Gamepad Specification by using signed axis values. 1: enable, 0: disable. (bool)
parm:           disable_deadzones:(bool) Disable dead zone handling for raw processing by Wine/Proton, confuses joydev. 0: disable, 1: enable. (bool)
parm:           enable_rolling_axis:(bool) Enable rolling axis by combining both triggers, out of spec for many games. (deprecated) 0: disable, 1: enable. (bool)
parm:           disable_shift_mode:(bool) Disable use Xbox logo button as shift. Will prohibit profile switching when enabled. 0: disable, 1: enable. (bool)
parm:           quirks:(string) Override or change device quirks, specify as: "MAC1{:,+,-}quirks1[,...16]", MAC format = 11:22:33:44:55:66, no pulse parameters = 1, no trigger rumble = 2, no motor masking = 4, use Linux button mappings = 16, use Nintendo mappings = 32, use Share button mappings = 64, reversed motor masking = 128 (array of charp)
```

defanator, 2024-09-28:
Tried to point `dkms` to the kernel source location explicitly with this:

```
xx@raspberrypi:~/git/xpadneo $ git diff
diff --git a/install.sh b/install.sh
index 27df47f..11cbc82 100755
--- a/install.sh
+++ b/install.sh
@@ -1,4 +1,4 @@
-#!/bin/bash -e
+#!/bin/bash -ex
 
 cd "$(dirname "$0")" || exit 1
 source "lib/verbose.sh"
@@ -23,7 +23,7 @@ if [[ -z "${INSTALLED[*]}" ]]; then
     cp --recursive "${V[@]}" hid-xpadneo/. "/usr/src/hid-xpadneo-${VERSION}/."
 
     echo "* installing module (using DKMS)"
-    dkms install "${V[*]}" "hid-xpadneo/${VERSION}" --force || cat_dkms_make_log
+    dkms install "${V[*]}" "--kernelsourcedir=/root/linux-315314059f92c13bc7d34b6aaff7527aca68457c" "hid-xpadneo/${VERSION}" --force || cat_dkms_make_log
 
 else
 
```

But it didn't work:

```
xx@raspberrypi:~/git/xpadneo $ sudo ./install.sh 
++ dirname ./install.sh
+ cd .
+ source lib/verbose.sh
++ case "$1" in
+ source lib/installer.sh
++ '[' 0 -ne 0 ']'
+++ git rev-parse --show-toplevel
++ GIT_ROOT=/home/xx/git/xpadneo
+++ __version
+++ git describe --tags --dirty
++ VERSION=v0.9-175-g70ef8ee-dirty
+++ type -p dkms
++ DKMS_BIN=/usr/sbin/dkms
++ : /usr/sbin/dkms
++ INSTALLED=($(get_dkms_versions_installed))
+++ get_dkms_versions_installed
+++ /usr/sbin/dkms status
+++ tr -s ':,/ ' ' '
+++ awk -- '$1 == "hid-xpadneo" { print $2 }'
+++ sort -nu
+ [[ ! -d /sys/devices/virtual/misc/uhid ]]
+ [[ -z '' ]]
+ set -e
+ echo '* creating dkms.conf'
* creating dkms.conf
+ sed 's/"@DO_NOT_CHANGE@"/"v0.9-175-g70ef8ee-dirty"/g'
+ echo '* adding hid-xpadneo-v0.9-175-g70ef8ee-dirty folder to /usr/src'
* adding hid-xpadneo-v0.9-175-g70ef8ee-dirty folder to /usr/src
+ mkdir -p /usr/src/hid-xpadneo-v0.9-175-g70ef8ee-dirty
+ cp --recursive hid-xpadneo/. /usr/src/hid-xpadneo-v0.9-175-g70ef8ee-dirty/.
+ echo '* installing module (using DKMS)'
* installing module (using DKMS)
+ dkms install '' --kernelsourcedir=/root/linux-315314059f92c13bc7d34b6aaff7527aca68457c hid-xpadneo/v0.9-175-g70ef8ee-dirty --force

Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee-dirty/source ->
                 /usr/src/hid-xpadneo-v0.9-175-g70ef8ee-dirty

DKMS: add completed.

Preparing kernel 6.6.51-v7+ for module build:
(This is not compiling a kernel, just preparing kernel symbols)
[... 52 lines cut ...]
                 from /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee-dirty/build/src/xpadneo/keyboard.c:7:
./include/uapi/asm-generic/int-ll64.h:12:10: fatal error: asm/bitsperlong.h: No such file or directory
 #include <asm/bitsperlong.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:280: /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee-dirty/build/src/xpadneo/core.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:280: /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee-dirty/build/src/xpadneo/consumer.o] Error 1
make[1]: *** [scripts/Makefile.build:280: /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee-dirty/build/src/xpadneo/keyboard.o] Error 1
In file included from ./include/asm-generic/int-ll64.h:11,
                 from ./arch/arm/include/uapi/asm/types.h:5,
                 from ./include/uapi/linux/types.h:5,
                 from ./include/linux/types.h:6,
                 from ./include/linux/limits.h:6,
                 from ./include/linux/kernel.h:7,
                 from ./include/linux/delay.h:22,
                 from /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee-dirty/build/src/xpadneo.c:11:
./include/uapi/asm-generic/int-ll64.h:12:10: fatal error: asm/bitsperlong.h: No such file or directory
 #include <asm/bitsperlong.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:280: /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee-dirty/build/src/xpadneo.o] Error 1
make: *** [Makefile:1825: /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee-dirty/build/src] Error 2
make: Leaving directory '/root/linux-315314059f92c13bc7d34b6aaff7527aca68457c'
```

defanator, 2024-09-28:
FWIW, got it working with the same version of xpadneo running the same Pi 2 under bookworm with kernel 6.6.31+rpt-rpi-v7 (got a completely fresh installation on separate SD card for testing).

Controller paired successfully without any noticeable issues.

Relevant info - module details:

```
root@raspberrypi:/# modinfo hid-xpadneo
filename:       /lib/modules/6.6.31+rpt-rpi-v7/updates/dkms/hid-xpadneo.ko.xz
version:        v0.9-175-g70ef8ee
description:    Linux kernel driver for Xbox ONE S+ gamepads (BT), incl. FF
author:         Kai Krakow <kai@kaishome.de>
author:         Florian Dollinger <dollinger.florian@gmx.de>
license:        GPL
srcversion:     A714E2DD109B5FD7A0B20B9
alias:          hid:b0005g*v0000045Ep00000B13
alias:          hid:b0005g*v0000045Ep00000B22
alias:          hid:b0005g*v0000045Ep00000B05
alias:          hid:b0005g*v0000045Ep00000B20
alias:          hid:b0005g*v0000045Ep000002FD
alias:          hid:b0005g*v0000045Ep000002E0
depends:        ff-memless
name:           hid_xpadneo
vermagic:       6.6.31+rpt-rpi-v7 SMP mod_unload modversions ARMv7 p2v8 
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        42:B4:FB:2B:45:63:7C:95:C8:E6:DB:FA:88:DB:11:7C:C5:B9:C6:B6
sig_hashalgo:   sha512
signature:      3D:66:44:C6:EB:07:1A:CC:B3:D3:91:A0:07:05:E6:85:13:96:24:EA:
		11:B9:47:DF:33:6C:03:1E:A6:9B:93:1D:5D:3D:C0:4E:D4:7C:67:B2:
		79:0D:39:0A:60:CA:2D:35:31:66:44:EE:59:24:68:93:01:C0:B5:07:
		8E:BC:70:B0:04:74:3D:0B:7E:25:4A:51:69:6A:93:39:C4:48:26:F5:
		DD:38:F0:FF:EF:6C:0C:06:6A:FA:23:DD:83:B5:D4:2D:9B:6E:32:10:
		E0:1B:52:3B:36:38:E4:CD:0D:22:2E:D7:6C:FE:0B:FA:E8:22:D9:21:
		FC:EA:3E:30:9B:61:50:CB:99:83:B5:C4:DF:72:36:09:64:F6:5E:49:
		B5:8D:6A:61:7D:0D:BF:35:35:EA:C6:6F:B1:F1:17:94:23:44:E0:7D:
		6E:61:35:B8:75:B4:69:F5:62:AA:C3:56:AC:E4:DD:AC:5E:5A:02:2D:
		A1:56:4E:70:02:A0:C2:87:CA:8C:CF:CA:6E:2E:6D:D1:63:C3:E0:BC:
		B1:75:81:91:B6:88:26:D9:E9:59:DC:6C:84:08:B5:E3:3E:74:E9:A5:
		62:9A:ED:CA:98:0F:A0:82:59:32:D5:38:BB:5C:F7:01:10:B8:B3:E7:
		95:F4:31:76:99:C3:9B:DC:67:80:8E:1D:7F:D1:0E:AF
parm:           trigger_rumble_mode:(u8) Trigger rumble mode. 0: pressure, 2: disable. (byte)
parm:           rumble_attenuation:(u8) Attenuate the rumble strength: all[,triggers] 0 (none, full rumble) to 100 (max, no rumble). (array of byte)
parm:           ff_connect_notify:(bool) Connection notification using force feedback. 1: enable, 0: disable. (bool)
parm:           gamepad_compliance:(bool) Adhere to Linux Gamepad Specification by using signed axis values. 1: enable, 0: disable. (bool)
parm:           disable_deadzones:(bool) Disable dead zone handling for raw processing by Wine/Proton, confuses joydev. 0: disable, 1: enable. (bool)
parm:           enable_rolling_axis:(bool) Enable rolling axis by combining both triggers, out of spec for many games. (deprecated) 0: disable, 1: enable. (bool)
parm:           disable_shift_mode:(bool) Disable use Xbox logo button as shift. Will prohibit profile switching when enabled. 0: disable, 1: enable. (bool)
parm:           quirks:(string) Override or change device quirks, specify as: "MAC1{:,+,-}quirks1[,...16]", MAC format = 11:22:33:44:55:66, no pulse parameters = 1, no trigger rumble = 2, no motor masking = 4, use Linux button mappings = 16, use Nintendo mappings = 32, use Share button mappings = 64, reversed motor masking = 128 (array of charp)
```

from bluetoothctl:

```
[Xbox Wireless Controller]# info
Device F4:6A:D7:FC:E2:4C (public)
	Name: Xbox Wireless Controller
	Alias: Xbox Wireless Controller
	Appearance: 0x03c4
	Icon: input-gaming
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00000001-5f60-4c4f-9c83-a7953298d40d)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v045Ep0B13d0522
	ManufacturerData Key: 0x0006
	ManufacturerData Value:
  03 00 80                                         ...             
	Battery Percentage: 0x44 (68)
```

Also, on this kernel I didn't have to play with disabling ERTM or doing any other custom stuff, things just worked out of the box.

The `hid-xpadneo` kernel module is not being loaded at start, but appears in `lsmod` output once controller is paired (perhaps this is how things should work with udev and stuff?)

Would still love to see it working under buster...

defanator, 2024-09-28:
TWIMC, if you're on buster and trying to use rpi-source to get latest kernel sources matching your updated kernel, you'll likely need this one - https://github.com/RPi-Distro/rpi-source/pull/35.

kaburkett, 2024-10-24:
I encounter the same issue I am trying to bypass, but different versions of some things, however vermagic still 5.10 matching your problematic example exactly 

When pairing Xbox Series X it also throws:
```
Failed to pair: org.bluez.Error.AuthenticationCanceled
```

``` 
pi@retropie:~ $ uname -a
Linux retropie 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

pi@retropie:~ $ dkms status
hid-xpadneo, v0.9.6, 5.10.103-v7l+, armv7l: installed
xpad, 0.4, 5.10.103-v7l+, armv7l: installed (original_module exists)
```

But my vermagic version is the same: ( 5.10.103-v7l+ SMP)

``` 
pi@retropie:~ $ modinfo hid-xpadneo
filename:       /lib/modules/5.10.103-v7l+/kernel/drivers/hid/hid-xpadneo.ko
version:        v0.9.6
description:    Linux kernel driver for Xbox ONE S+ gamepads (BT), incl. FF
author:         Kai Krakow <kai@kaishome.de>
author:         Florian Dollinger <dollinger.florian@gmx.de>
license:        GPL
srcversion:     C4A17E802EF117E0A044F94
alias:          hid:b0005g*v0000045Ep00000B13
alias:          hid:b0005g*v0000045Ep00000B22
alias:          hid:b0005g*v0000045Ep00000B05
alias:          hid:b0005g*v0000045Ep00000B20
alias:          hid:b0005g*v0000045Ep000002E0
alias:          hid:b0005g*v0000045Ep000002FD
depends:        ff-memless
name:           hid_xpadneo
vermagic:       5.10.103-v7l+ SMP mod_unload modversions ARMv7 p2v8 
parm:           trigger_rumble_mode:(u8) Trigger rumble mode. 0: pressure, 1: directional (deprecated), 2: disable. (byte)
parm:           rumble_attenuation:(u8) Attenuate the rumble strength: all[,triggers] 0 (none, full rumble) to 100 (max, no rumble). (array of byte)
parm:           ff_connect_notify:(bool) Connection notification using force feedback. 1: enable, 0: disable. (bool)
parm:           gamepad_compliance:(bool) Adhere to Linux Gamepad Specification by using signed axis values. 1: enable, 0: disable. (bool)
parm:           disable_deadzones:(bool) Disable dead zone handling for raw processing by Wine/Proton, confuses joydev. 0: disable, 1: enable. (bool)
parm:           quirks:(string) Override device quirks, specify as: "MAC1:quirks1[,...16]", MAC format = 11:22:33:44:55:66, no pulse parameters = 1, no trigger rumble = 2, no motor masking = 4, hardware profile switch = 8, use Linux button mappings = 16, use Nintendo mappings = 32, use Share button mappings = 64, reversed motor masking = 128 (array of charp)
```

I followed your lead and wiped the sd, starting with fresh brandnew version of raspian, installed retropie and then xpadneo ontop of that, 

When trying to connect now, the status is continually flipping:

```
CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
[CHG] Device 40:8E:2C:4F:9B:7D Connected: yes
[CHG] Device 40:8E:2C:4F:9B:7D Connected: no
```

no messages in `dmesg`

I try using hid-xpadneo to get some output but get the module not found error like I think you might be mentioning here:

> The hid-xpadneo kernel module is not being loaded at start, but appears in lsmod output once controller is paired (perhaps this is how things should work with udev and stuff?)

kaburkett, 2024-10-25:
For me, the issue was partly because my kernel was version 8, and when running the two commands I got a complaint. 
` lsmod | grep hid_xpadneo ` would return nothing

` lsmod | grep hid_xpadneo ` would complain there is no xpadneo module in kernel 8. So I had to modify my pi config.txt to boot to kernel 7l

```
[all]
kernel=kernel7l.img
```

Then xpadneo was running, but it still took me a lot of connection attempts to get them paired. Finally when they reported as paired I would reboot the bluetooth service

` sudo systemctl restart bluetooth ` without rebooting and this would make the controllers available in the retropie input config section

I also followed a reddit user's guide to modify some additional things:

1.Enabling uhid

First we need to enable the UserspaceHID if we don't do that the Controller won't connect properly and begins to connect and disconnect in a cycle and the Xbox Button will keep flashing

All you need to do is edit/create /etc/bluetooth/input.conf with your editor of choice (I prefer nano) In there paste this:
```
# Enable HID protocol handling in userspace input profile
# Defaults to false (HIDP handled in HIDP kernel module)
UserspaceHID=true
```
2. Edit the main bluetooth config

We need to set some settings in the main bluetooth config file so xpadneo will work as intended and we also need to address the input lag Problem all you need to do is editing /etc/bluetooth/main.conf pasting in this:
```
[General]
Privacy = device
JustWorksRepairing = always
Class = 0x000100
FastConnectable = true

[LE]
MinConnectionInterval=7
MaxConnectionInterval=9
ConnectionLatency=0
```
3. Installing xpadneo

Install bluez bluez-utils and dkms beforehand

git clone https://github.com/atar-axis/xpadneo.git
cd xpadneo
sudo ./install.sh

Afterwards Reboot your system

4. Pairing the Series Controller with the PC
```
sudo bluetoothctl
scan on    
(stop the scan as soon as you see that "Xbox Wireless Controller" is popping up)
scan off (copy the Mac Address of the controller)
pair <MAC Address you've copied>
trust <MAC Address you've copied>
connect <MAC Address you've copied>
```

kakra, 2026-03-26:
I wonder if this is still an issue with modern distributions on Raspberry Pi. I think some even provide xpadneo packages already.