atar-axis/xpadneo #498: error install

RaulKong898, 2024-10-05:
raulkong898@fedora:~/xpadneo$ sudo ./install.sh
* creating dkms.conf
* adding hid-xpadneo-v0.9-175-g70ef8ee folder to /usr/src
* installing module (using DKMS)

Sign command: /lib/modules/6.12.0-0.rc1.17.fc42.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee/source -> /usr/src/hid-xpadneo-v0.9-175-g70ef8ee

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=6.12.0-0.rc1.17.fc42.x86_64 -C /lib/modules/6.12.0-0.rc1.17.fc42.x86_64/build M=/var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee/build/src VERSION=v0.9-175-g70ef8ee modules
Error! Bad return status for module build on kernel: 6.12.0-0.rc1.17.fc42.x86_64 (x86_64)
Consult /var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee/build/make.log for more information.

kakra, 2024-10-05:
Please attach this file: `/var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee/build/make.log`.

RaulKong898, 2024-10-06:
> Please attach this file: `/var/lib/dkms/hid-xpadneo/v0.9-175-g70ef8ee/build/make.log`.

[make.log](https://github.com/user-attachments/files/17269551/make.log)

kode54, 2024-10-08:
I can attach a similar make.log. Also from a 6.12-rc kernel, rc2 to be precise.

[make.log](https://github.com/user-attachments/files/17290274/make.log)

kakra, 2024-10-08:
Okay, this looks like an incompatibility with kernel 6.12. I'll fix this. Thanks for reporting.

TarsiSurdi, 2024-10-26:
Can confirm that there is an issue with the kernel module with this version of the kernel (6.12). Just tried it out on Fedora 41 Beta and it fails to build.

hithere-at, 2024-10-27:
I think i have found a workaround. You can change the code from
```
static u8 *xpadneo_report_fixup(struct hid_device *hdev, u8 *rdesc, unsigned int *rsize)
```

kakra, 2024-11-18:
I've queued some patches here: https://github.com/kakra/xpadneo/commits/queue/for-0.9/

Feel free to test those. I yet want to finalize the remaining hardware support commits before releasing a final v0.9. From there on, v0.10 will restructure a lot of code so it will become impossible to backport patches from v0.10 to v0.9. So I want to let that settle first.

ionenwks, 2024-11-18:
> I've queued some patches here: https://github.com/kakra/xpadneo/commits/queue/for-0.9/

Seems to be missing commit 242e9b46bb's addition of `#include <linux/version.h>` to xpadneo.h which is needed for the 6.12 commit (I've added that to my own backport in Gentoo).

Beside that seems fine at a glance, albeit I don't really use/test 0.9.x much myself (I've been using the master branch without issues for a while for my own gamepad -- which is updated to latest every kernel bumps I do).

kakra, 2024-11-18:
> Seems to be missing commit [242e9b4](https://github.com/atar-axis/xpadneo/commit/242e9b46bb39d3e59d03f1ca670abd0001a21e68)'s addition of `#include <linux/version.h>` to xpadneo.h which is needed for the 6.12 commit (I've added that to my own backport in Gentoo).

Good find. For some reason, the same patch builds just fine in master. I will fix that.

closed 2024-11-18 by kakra

smilodon-gihub, 2025-04-04:
hi. it seems im having this problem since kernel 6.13.8 on fedora 41. It makes the kernel panic on boot.

heres the [make.log](https://github.com/user-attachments/files/19607303/make.log)

ndbhamvani, 2025-08-21:
I can confirm that I noticed something similar to what @smilodon-gihub mentioned. 
On login screen, after reboot following driver installation, the screen froze and the fans started making noise. 
**Kernel:** 6.15.10
**OS:** Fedora 42