atar-axis/xpadneo #501: Xbox One S and Xbox Elite 2 Controllers connect using USB and Bluetooth but aren't recognized by any programs

coolcoons, 2024-10-30:
## Version of xpadneo
v0.9-175-g70ef8ee

## Controller Model

- [X] Xbox One S controller
- [X] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [X] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [X] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [X] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [X] Steam Proton games are having issues
- [X] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [X] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [X] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [X] Other software is having issues (describe software and issues below)
- [X] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [X] I've tried disabling or running without above mentioned software
- [X] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [X] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

Both controllers will connect to this computer on Fedora via Bluetooth and USB cable but neither will do anything or be recognized by any programs including Steam and evtest while the xpadneo is installed.
The Xbox One S Controller functions perfectly while xpadneo isn't installed and shows up in Steam and evtest.
The Xbox Elite 2 Controller will connect to the computer while xpadneo isn't installed but still doesn't do anything and Steam and evtest don't recognize it.
[... 121 lines cut ...]
[  +0.000762] [<    0.000762>] microsoft 0005:045E:0B20.0014: input,hidraw12: BLUETOOTH HID v5.17 Gamepad [Xbox Wireless Controller] on f4:3b:d8:6d:86:ce
[  +0.036120] [<    0.036120>] Loading of module with unavailable key is rejected
[ +19.125230] [<   19.125230>] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B22.0015/input/input53
[  +0.000822] [<    0.000822>] microsoft 0005:045E:0B22.0015: input,hidraw12: BLUETOOTH HID v5.22 Gamepad [Xbox Wireless Controller] on f4:3b:d8:6d:86:ce
[  +0.007654] [<    0.007654>] Loading of module with unavailable key is rejected
[ +22.979988] [<   22.979988>] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B20.0016/input/input54
[  +0.000261] [<    0.000261>] microsoft 0005:045E:0B20.0016: input,hidraw12: BLUETOOTH HID v5.17 Gamepad [Xbox Wireless Controller] on f4:3b:d8:6d:86:ce
[  +0.001189] [<    0.001189>] Loading of module with unavailable key is rejected
[  +5.096053] [<    5.096053>] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B22.0017/input/input55
[  +0.002151] [<    0.002151>] microsoft 0005:045E:0B22.0017: input,hidraw12: BLUETOOTH HID v5.22 Gamepad [Xbox Wireless Controller] on f4:3b:d8:6d:86:ce
[  +0.002915] [<    0.002915>] Loading of module with unavailable key is rejected
```

```
#zgrep UHID /boot/config-6.11.5-300.fc41.x86_64

CONFIG_UHID=m
```

## Controller and Bluetooth Information

Xbox One S controller C8:3F:26:5A:12:28
Xbox Elite 2 controller 98:7A:14:29:18:59
[dmesg egrep -i 'hid input xpadneo' tee xpadneo-dmesg.md](https://github.com/user-attachments/files/17564730/dmesg.egrep.-i.hid.input.xpadneo.tee.xpadneo-dmesg.md)
[sudo btmon tee xpadneo-btmon.md](https://github.com/user-attachments/files/17564734/sudo.btmon.tee.xpadneo-btmon.md)

kakra, 2024-10-31:
According to your dmesg snippet, xpadneo doesn't load at all, probably due to missing module signing:
> [  +0.002915] [<    0.002915>] Loading of module with unavailable key is rejected

This should be done by your distribution tooling if you use dkms for installation. If you install the module manually, it's up to you. It looks like you're installing via dkms, and dkms properly did sign the module...

You should probably ask your distribution support what may go wrong here.

Currently, your controller is supported through the `hid-microsoft` built-in kernel module. It should just work with Steam, evtest etc. xpadneo just adds additional functions, can fix some mappings for older games, and adds rumble support for some models where the native kernel doesn't enable rumble support. All your controllers are fully supported by native kernel drivers.

You may see multiple problems...

Here's a hint about Bluetooth:

- Your description tells me that you may see Bluetooth compatibility problems with your controllers
- Try updating all your models to the latest firmware (using an Xbox console, a Windows machine, or a Windows VM)
- The kernel may not properly support your Bluetooth chipset if it works in Windows
- Your distribution may ship a firmware for your Bluetooth chipset which has bugs with certain devices
- If running KDE with kdeconnect, Xbox controller pairing may be disfunctional or the connection is unstable

Here's a hint about your installation:

- QMK interferes with xpadneo if trying to use the controller with Steam (may be fixed by an upcoming patch in my patch queue), but in your case, xpadneo doesn't load at all

Additionally, unless the controller shows up in `evtest`, the Bluetooth pairing hasn't been properly finished. That's not an issue with xpadneo because xpadneo (and neither hid-microsoft or hid-generic) do anything with Bluetooth. The xpadneo driver doesn't know anything about Bluetooth devices, the Bluetooth layer just gives us a HID device if it was properly paired, otherwise we never know about it in xpadneo.

coolcoons, 2024-10-31:
Thank you! I was really spinning my wheels and didn't understand what was going wrong. That really helps to explain what's going on behind the scenes so I can do further troubleshooting and/or reach out to the Fedora community. https://discussion.fedoraproject.org/tag/workstation

closed 2024-10-31 by coolcoons