atar-axis/xpadneo #502: Support 8bitdo Ultimate 2C (2.4GHz)

MastaG, 2024-11-02:
## Version of xpadneo

## Controller Model

- [ ] Other: 8bitdo Ultimate 2C

### Connection mode

- [ ] It comes with its own dongle, probably Xinput

## Describe your feature request

- [ ] I'd like to see support for a specific model

**Is your feature request related to a problem? Please describe.**

The controller works fine, however xpadneo will try and set the Nintendo layout.
However this controller uses the Xbox layout.
This means that the action buttons are reversed.

**Describe the solution you'd like**

Please remove the quirk for this model that sets the Nintendo layout.
Most 8bitdo controllers are like that, but not this one.

**Describe alternatives you've considered**

**Additional context**

kakra, 2024-11-02:
You can reverse the buttons via a module option.

You didn't give enough details so we could automate this via a built-in quirk flag.

MastaG, 2024-11-02:
Yes I know, I'll post proper logs tomorrow.