atar-axis/xpadneo #503: Supprt PDP Victrix Pro BFG Xbox controller

8bitbuddhist, 2024-11-02:
## Version of xpadneo
0.9.6

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [X] Other: PDP Victrix Pro BFG - Xbox edition

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Describe your feature request

- [X] I'd like to see support for a specific model
- [ ] I'd like to see support for a specific hardware feature
- [ ] I'd like to see support for a specific software feature

**Is your feature request related to a problem? Please describe.**
This controller connects and works over Bluetooth, but it doesn't use the xpadneo driver. Instead, it uses the hid-generic driver. Almost everything works except for rumble. I tried creating udev rules to unbind and rebind the controller without success. It would be great to have xpadneo automatically detect and bind the controller on connection.

**Describe the solution you'd like**
When connecting the controller over Bluetooth, it uses xpadneo instead of the hid-generic driver

**Describe alternatives you've considered**
Creating udev rules (unsuccessful, but still trying)

**Additional context**
udevadm output:

```sh
looking at device '/devices/virtual/misc/uhid/0005:0E6F:024B.0005/input/input21':
    KERNEL=="input21"
    SUBSYSTEM=="input"
    DRIVER==""
    ATTR{capabilities/abs}=="30627"
    ATTR{capabilities/ev}=="1b"
    ATTR{capabilities/ff}=="0"
    ATTR{capabilities/key}=="ffff000000000000 0 8000000000 0 0"
    ATTR{capabilities/led}=="0"
    ATTR{capabilities/msc}=="10"
    ATTR{capabilities/rel}=="0"
    ATTR{capabilities/snd}=="0"
    ATTR{capabilities/sw}=="0"
    ATTR{id/bustype}=="0005"
    ATTR{id/product}=="024b"
    ATTR{id/vendor}=="0e6f"
    ATTR{id/version}=="0300"
    ATTR{inhibited}=="0"
    ATTR{name}=="Victrix Pro BFG"
    ATTR{phys}=="ac:50:de:9f:ab:88"
    ATTR{power/async}=="disabled"
    ATTR{power/control}=="auto"
    ATTR{power/runtime_active_kids}=="0"
    ATTR{power/runtime_active_time}=="0"
    ATTR{power/runtime_enabled}=="disabled"
    ATTR{power/runtime_status}=="unsupported"
    ATTR{power/runtime_suspended_time}=="0"
    ATTR{power/runtime_usage}=="0"
    ATTR{properties}=="0"
    ATTR{uniq}=="00:34:30:47:37:ab"
```

kakra, 2024-11-03:
Binding won't work if xpadneo itself won't recognize the device. You cannot just force a driver to use any "random" device. You'd need to add it to the source code, too. This will only work if the HID descriptor is compatible. Can you dump it?

8bitbuddhist, 2024-11-10:
In hindsight, I probably should've known that 😅 I just assumed it would be treated like an Xbox controller. I'll see if I can get and post a descriptor dump.

closed 2024-12-21 by github-actions[bot]