atar-axis/xpadneo #504: ./configure.sh: WARNING: hid-xpadneo not loaded, configured anyways - trigger FF - Ubuntu 24.10

ashleyhinton, 2024-11-20:
## Version of xpadneo
0.9-178

## Controller Model
0005:045E:0B22.000A

- [ ] Xbox One S controller
- [X ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [X] Steam Proton games are having issues
- [X] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [X] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [X] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [X] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [X] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [X] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [X] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
Trigger rumble not working, configure.sh complains hid-xpadneo not loaded yet the controller generally works otherwise.

Rumble in general sometimes seems hit & miss, and actually better if I enable Steam Input, so I'm wondering is there something deeper going on rather than just the trigger rumble. However _evtest_, _fftest_, steam, and an online controller test website I googled are happy with the controller buttons including the bottom paddles and general rumble.

`sudo ./configure.sh -m 0`

`./configure.sh: trigger_rumble_mode: parameter written to /etc/modprobe.d/99-xpadneo-options.conf`

`./configure.sh: WARNING: hid-xpadneo not loaded, configured anyways`

Have followed the instructions to install xpadneo, I think!, have reviewed it and gone through the steps again.
Am reasonably competent with command line and am certainly able to understand what to look for if instructed.

I noticed when I first installed xpadneo that it warned me kernel uhid not found, controller firmware 5.x will not be supported.
I removed it using the uninstall script and re-installed after `sudo modprobe uhid` although I think I'm right in saying that it didn't matter as the controller worked (as in seen, could play games, the buttons worked) without uhid.

The only thing I haven't done is check the controller firmware is the latest, but I'm suspecting because its telling me the kernel module isn't loaded? that the controller firmware is not the main issue. However, it is loaded - I think - because the controller works.

`sudo modinfo hid_xpadneo`

`filename:       /lib/modules/6.11.0-9-generic/updates/dkms/hid-xpadneo.ko.zst
version:        v0.9-178-gbe65dbb
description:    Linux kernel driver for Xbox ONE S+ gamepads (BT), incl. FF
author:         Kai Krakow <kai@kaishome.de>
author:         Florian Dollinger <dollinger.florian@gmx.de>
license:        GPL
srcversion:     E7CF7D39F46AF18ED28BBD7
alias:          hid:b0005g*v0000045Ep00000B13
alias:          hid:b0005g*v0000045Ep00000B22
alias:          hid:b0005g*v0000045Ep00000B05
alias:          hid:b0005g*v0000045Ep00000B20
alias:          hid:b0005g*v0000045Ep000002FD
alias:          hid:b0005g*v0000045Ep000002E0
depends:        hid,ff-memless
retpoline:      Y
name:           hid_xpadneo
vermagic:       6.11.0-9-generic SMP preempt mod_unload modversions 
sig_id:         PKCS#7
signer:         Biggie Secure Boot Module Signature key
sig_key:        (reacted unless you need it?)
sig_hashalgo:   sha512
signature:      (redacted unless you need it?)
parm:           trigger_rumble_mode:(u8) Trigger rumble mode. 0: pressure, 2: disable. (byte)
parm:           rumble_attenuation:(u8) Attenuate the rumble strength: all[,triggers] 0 (none, full rumble) to 100 (max, no rumble). (array of byte)
parm:           ff_connect_notify:(bool) Connection notification using force feedback. 1: enable, 0: disable. (bool)
parm:           gamepad_compliance:(bool) Adhere to Linux Gamepad Specification by using signed axis values. 1: enable, 0: disable. (bool)
parm:           disable_deadzones:(bool) Disable dead zone handling for raw processing by Wine/Proton, confuses joydev. 0: disable, 1: enable. (bool)
parm:           enable_rolling_axis:(bool) Enable rolling axis by combining both triggers, out of spec for many games. (deprecated) 0: disable, 1: enable. (bool)
parm:           disable_shift_mode:(bool) Disable use Xbox logo button as shift. Will prohibit profile switching when enabled. 0: disable, 1: enable. (bool)
parm:           quirks:(string) Override or change device quirks, specify as: "MAC1{:,+,-}quirks1[,...16]", MAC format = 11:22:33:44:55:66, no pulse parameters = 1, no trigger rumble = 2, no motor masking = 4, use Linux button mappings = 16, use Nintendo mappings = 32, use Share button mappings = 64, reversed motor masking = 128 (array of charp)
`

## Steps to Reproduce
`sudo apt install dkms linux-headers-`uname -r`  `
`sudo apt install git`
`git clone https://github.com/atar-axis/xpadneo.git`
`cd xpadneo`
`sudo modprobe uhid`
`sudo ./install.sh`

`sudo ./configure.sh -m 0`

## Expected Behavior
Trigger rumble configurable/working.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a

Linux Biggie 6.11.0-9-generic #9-Ubuntu SMP PREEMPT_DYNAMIC Mon Oct 14 13:19:59 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory

```

## Controller and Bluetooth Information

## Additional Context
I think I need to know whether the kernel module is really loaded properly and if not track down why not, or indeed am I missing something obvious? None of this is game-breaking, the controller works, and I also wanted to say a big thank you for the driver - being able to use my nice comfortable controller on Linux is awesome.

[xpadneo-btmon.txt](https://github.com/user-attachments/files/17830700/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/user-attachments/files/17830704/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/user-attachments/files/17830705/xpadneo-lsusb.txt)

[install-sh-output.txt](https://github.com/user-attachments/files/17830716/install-sh-output.txt)
[fftest-output.txt](https://github.com/user-attachments/files/17830718/fftest-output.txt)

kakra, 2024-11-20:
uhid isn't exactly a requirement of xpadneo but rather bluez to provide HID input with BLE devices. That said, an Xbox controller is only running in BLE mode with firmware 5 or higher.  Your model XBE2 comes with firmware 5 out of the box. The early versions had some Bluetooth stability problem, that's why xpadneo logs a warning if the firmware version is known to have problems. And combined with the uhid requirement for BLE devices in bluez, it's why it says "firmware 5.x will not be supported".

However, your distribution may have changed bluez standard configuration to use the old non-uhid model even with BLE. We cannot really test for any possible combination. If it works, even with uhid not loaded, then it's fine. We will keep the warning anyways. I'd rather see a bug report saying "or there's this message but it works even without uhid" than a report saying "oh it doesn't work" only to find that uhid is missing.

The configure script won't load the driver but it can still configure it. And the driver will only load after the controller has been connected. If you run the configure script after connecting the controller, there should be no warning.

According to dmesg, xpadneo is loaded and working: It grabs the device.

So this leaves us with what issue exactly? I think this issue is rather a question... ;-) Feel free to join our Discord (link in the main page of this project).

Thank you for your nice words in the end. Always appreciated. If this answered your questions, feel free to close.

ashleyhinton, 2024-11-22:
Thanks for replying. That all makes sense, and xpadneo does indeed grab the device, what I thought was an error when using ./configure.sh   .. isn't ;-)
Interestingly I don't think I played any games with trigger FF when using Windows. When I tried it yesterday I didn't actually like it so have switched it off again. I am getting some random disconnects but thats bluetooth/dongle (TP-Link UB400) related, I'm sure, and not xpadneo. Ultimate fix maybe is a different controller with a 2.4GHz dongle but I don't want to do that as I really like my XBE2 (f/w up to date as of yesterday via a win10 machine) - its a perfectly comfortable controller for me and I'm mostly gaming on the tv via steam big picture these days :-)  thanks again, I will close.

closed 2024-11-22 by ashleyhinton

kakra, 2024-11-22:
> Interestingly I don't think I played any games with trigger FF when using Windows. When I tried it yesterday I didn't actually like it so have switched it off again.

xpadneo currently doesn't support native trigger FF from games because the kernel itself doesn't support this (future kernels plan to implement that, I've yet to check whether 6.12 supports it). So what xpadneo does instead: It synthesizes the effect by using a combination of the main motors to apply an effect to the trigger motors derived from the applied pressure. This will not match any native effect by any means but it can create a somewhat convincing effect in some games which even don't support trigger force feedback.

> I am getting some random disconnects but thats bluetooth/dongle (TP-Link UB400) related, I'm sure, and not xpadneo.

Yes, true. xpadneo itself doesn't interact with Bluetooth at all. We are just operating on the HID driver which itself is created by the in-kernel Bluetooth driver (for non-BLE controllers) or the bluez user-space daemon (for BLE controller which all 5.x firmware controllers are).

If you want to use the original Xbox dongle (which uses a proprietary Wifi protocol at 5 or 2.4 GHz), this will work with your controller and can be obtained separately. But then you'll need to use the xone driver instead for now, as xpadneo doesn't support this yet.