atar-axis/xpadneo #509: Could xpadneo cause Kwin/x11 freezes?

CaptainCoward, 2024-11-25:
I'm sorry i know that might be a stupid question. Though i lately did a System-Update to Kernel 6.12. It hardly updated any packages but xpadneo and ever since i seem to get random freezes in Retroarch. What's odd about that is that i have nothing indicating any issues in my logs nor when running dmesg. When i tab out i see that the Taskbar (Plasma) isn't working anymore. Though I'm still able to start programs with hotkeys. I can just asume Kwin/x11 crashed or bugged out.
I started suspecting some relation with xpadneo due to the fact that it worked pre 6.12 and it didn't happen yet without using my gamepad. I'm not really good in such things so i don't know if it is theoretical possible that there could be some connection?

kakra, 2024-11-26:
Most likely, your issue is not with kwin but with plasma-shell instead.

Freezes won't be caused by xpadneo directly. However, in the past we found hints that kdeconnectd may interact badly with some devices connected via Bluetooth. But usually, that caused problems with connection stability.

Try stopping/killing kdeconnectd and see if it solves your problem. If it does, please report to KDE instead. If it doesn't, your problem is probably not related to xpadneo.

Personally, I'm seeing freezes in plasma-shell due to filesystem IO over network. If you mount or use any network shares, check if the problem still exists if you avoid using those for a while. If it fixes your problem, report to KDE.

One more issue I've seen from time to time is the weather applet causing plasma-shell freezes. That's also most likely related to network IO.

On console, you can use alt+sysrq+w to dump a list of tasks that are stuck in uninterruptible state. This MAY give a clue of what's going on. But I don't think this is your problem.

There's a slight chance that plasma chokes on battery reporting which xpadneo implements. But we are doing nothing special here. If you disable the battery applet, this could solve your problem. But most likely, this is not a problem of xpadneo because we are just reporting to the power supply subsystem in der kernel, user-space doesn't directly interact with xpadneo.

According to your description, I'd most likely point to kdeconnectd first. OTOH, something in kernel 6.12 might work differently and it's just coincidence that you happen to see it while using the gamepad... Maybe because you're using a game or emulator at the same time?

If you need more help, I suggest reaching out to our Discord channel. This report is most likely not a problem in xpadneo itself. So I tend to closing it. I'll set status to auto-close after some time of inactivity. If we find exact hints that xpadneo is directly causing the problem, we can re-open this issue.

CaptainCoward, 2024-11-26:
Hey thank you for your detailed response. I'll have a look. So far i was just switching to Wayland and the Issue didn't happen again. I might see how that goes. If it happens again i'll try killing  kdeconnectd. Thanks a bunch :)

kakra, 2024-12-23:
Hmm, well, there's a chance that xpadneo is detected as an input driver in X11. But I doubt that this could cause any issues for kwin. Maybe post your `xorg.log`.

CaptainCoward, 2024-12-23:
Hey, I haven't had the issue again ever since I switched to Wayland. I mean, it could have been one of the kernels as well. I've now already updated the system/kernel twice after it happened but i'll post the Xorg.log file anyway in case it is still helpful in any way. Though honestly, I don't want to waste your time on this, as it seems to work fine on Wayland, and I don't even know if there actually is or was an issue with xpadneo. As I mentioned in my original post, I was basically just trying to track down what might have caused the issue and was thinking maybe there could be some relation. 

[xorg.txt](https://github.com/user-attachments/files/18233256/xorg.txt)

kakra, 2024-12-23:
I don't see any indication of the controller being used as an input device. Let's close it then.

CaptainCoward, 2024-12-24:
Alright Ty :)

closed 2024-12-24 by kakra