atar-axis/xpadneo #511: How do I make the controller use key presses so it works with most things?

realeatham, 2024-12-02:
I paired my (Xbox series X|S) controller, but only the share button seems to affect my apps like firefox and my desktop environment. I used some online gamepad tester and it seems to work, so I assume that the gamepad uses some system for gamepads, not key presses, which makes sense. However, I am not playing games and nothing else seems to support these, so a way to make the controller use key and mouse presses would be nice.

Ideas for mapping the controller to keyboard presses:

dpad > arrow keys
left stick > scrollwheel
right stick > mouse
left trigger > left click
right trigger > right click
xbox button > super key
3 lines > escape (maybe)
Not sure about what to do with xyab, LB and RB, and the button with 2 rectangles intercecting. Maybe just left as gamepad keys?

kakra, 2024-12-02:
There's a mouse emulation aka "couch mode" in preparation which will be merged with v0.10. The gamepad will send mouse and keyboard input in this mode.

The share button works because the gamepad actually sends the F12 keypress. This will be configurable, probably with v0.11.

Check this PR for mouse mode: https://github.com/atar-axis/xpadneo/pull/282. I could need some testers.

kakra, 2024-12-02:
Please ask in our Discord how to use a pull request for testing. The link is in the readme.

realeatham, 2024-12-03:
Can you just say how here? I don't want to join a discord server just to test the driver, and I don't even have discord on this computer

kakra, 2024-12-03:
I'd rather add documentation for that instead of digging this inside issues. Stay tuned.

kakra, 2025-12-18:
Moved to v0.11 because this better fits with planned customizations.