atar-axis/xpadneo #512: [support request] Could xpadneo affect other USB controllers?

bserem, 2024-12-05:
## Version of xpadneo
Latest

## Controller Model

- [x] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [x] Other: Logitech G920 Wheel

### Connection mode

- [x] Bluetooth connection -> For One S controller
- [x] USB cable (not yet supported) -> For G920
- [ ] Xbox Dongle connection (not yet supported)

## Description

I connected a Logitech G920 wheel to my linux box (via USB). I can see it in https://github.com/berarma/oversteer and I can game (tried one game for the moment) under steam. All is looking ok.
However, even though I can use it in the game ("grid autosport") I **can not use it inside Steam Big Picture**. Steam reports `no controllers found`.

I doubt it has to do with xpadneo, but it doesn't hurt to ask.
Could xpadneo affect detection of a USB device by Steam?

Thanks

kakra, 2024-12-05:
Steam probably doesn't show devices detected as wheels. xpadneo should not affect this but you didn't provide details like dmesg or lsusb. xpadneo will only support HID controllers showing as Microsoft vendor, no matter if connected via USB or Bluetooth - it's HID that matters. I don't expect the wheel to show as Microsoft vendor, so xpadneo won't pick it up.

If you show the requested log outputs, we can be sure.

bserem, 2024-12-06:
**dmesg**
```
[  212.099898] usb 3-1.3: new full-speed USB device number 3 using xhci_hcd
[  212.204733] usb 3-1.3: New USB device found, idVendor=046d, idProduct=c261, bcdDevice=96.03
[  212.204752] usb 3-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  212.204761] usb 3-1.3: Product: G920 Driving Force Racing Wheel for Xbox One
[  212.204768] usb 3-1.3: Manufacturer: Logitech
[  212.204775] usb 3-1.3: SerialNumber: 0000bd712ae44044
[  213.435388] usb 3-1.3: USB disconnect, device number 3
[  213.741905] usb 3-1.3: new full-speed USB device number 4 using xhci_hcd
[  213.844276] usb 3-1.3: New USB device found, idVendor=046d, idProduct=c262, bcdDevice=96.03
[  213.844297] usb 3-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  213.844306] usb 3-1.3: Product: G920 Driving Force Racing Wheel for Xbox One
[  213.844312] usb 3-1.3: Manufacturer: Logitech
[  213.844319] usb 3-1.3: SerialNumber: 0000bd712ae44044
[  213.882671] hid: raw HID events driver (C) Jiri Kosina
[  213.886858] usbcore: registered new interface driver usbhid
[  213.886862] usbhid: USB HID core driver
[  213.959024] input: Logitech G920 Driving Force Racing Wheel as /devices/pci0000:00/0000:00:1c.4/0000:04:00.0/0000:05:01.0/0000:07:00.0/0000:08:02.0/0000:0c:00.0/usb3/3-1/3-1.3/3-1.3:1.0/0003:046D:C262.0001/input/input23
[  213.960113] logitech-hidpp-device 0003:046D:C262.0001: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Logitech G920 Driving Force Racing Wheel] on usb-0000:0c:00.0-1.3/input0
[  213.963665] logitech-hidpp-device 0003:046D:C262.0001: HID++ 4.2 device connected.
[  213.992079] logitech-hidpp-device 0003:046D:C262.0001: Force feedback support loaded (firmware release 3).
```

**lsusb**
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 003: ID 04f2:b604 Chicony Electronics Co., Ltd Integrated Camera (1280x720@30)
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. Card Reader
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 004: ID 046d:c262 Logitech, Inc. G920 Driving Force Racing Wheel
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 002: ID 05e3:0620 Genesys Logic, Inc. GL3523 Hub
```

@kakra thanks for the response. The device does identify properly as Logitech, I pasted the log outputs above for you. It does say it's HID. From your comment I understand this doesn't matter though, as it is not a `Microsoft` device.

It could easily be an issue with Steam itself. I just couldn't know where to start looking for this.

Thanks!

kakra, 2024-12-06:
Yes, according to your logs, your device is grabbed by the driver `logitech-hidpp-device` and not `hid-xpadneo`. So xpadneo has no impact here.

Most likely Steam Input just ignores the device. You may be able to solve it via udev by tagging it as a game input device, or Steam Input itself ignores it just for being a wheel.

closed 2024-12-06 by kakra