atar-axis/xpadneo #513: Xbox Wireless Controller doesn't rumble but is successfully probed by driver

d-k-c, 2024-12-06:
## Version of xpadneo

`[   82.760417] loaded hid-xpadneo v0.9-185-g33408d5`

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [X] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [X] Steam Input (enabled by default via Steam Desktop client)
  - I've installed and then removed the desktop client, so there might be some artifacts?
- [X] Steam Link (usually via Raspberry Pi or other micro computers)
  - Installed manually with flatpack
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [X] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [X] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [X] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [X] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [X] I don't have this tool

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [X] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [X] I can code and I want to help

## Describe the Bug

I got a pair of Xbox Series S controller and they don't rumble at all. In the troubleshooting guide, it states that this kind of failure can be caused by the wrong driver being loaded, but I can see in dmesg that xpadneo handles the controller (I had to manually modprobe uhid and hid-xpadneo, and then restarted bluez):

## Steps to Reproduce

Turn on the controller, it connects but doesn't rumble.

## Expected Behavior

Rumble on connection.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a

Linux mediotop 5.10.0-33-amd64 #1 SMP Debian 5.10.226-1 (2024-10-03) x86_64 GNU/Linux
```

```
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
```

[cksum.txt](https://github.com/user-attachments/files/18043602/cksum.txt)

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/user-attachments/files/18043562/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/user-attachments/files/18043588/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/user-attachments/files/18043589/xpadneo-lsusb.txt)

## Additional Context

The controller works fine and I can play games with it, only the rumbling seems to be affected. I'm probably missing something in my setup but at this point I've exhausted all the possibilities I found in the troubleshooting guide and on the Internet.

sdlea, 2024-12-07:
> [  138.626474] xpadneo 0005:045E:0B13.0005: enabling heuristic GameSir Nova quirks
> Xbox Wireless Controller [ac:8e:bd:22:2f:f4] connected
According to your dmesg output, we probably have the same issue.
Maybe you can temporarily change XPADNEO_OUI_MASK_GAMESIR_NOVA to another value rebuild and reload the module, then test again.

kakra, 2024-12-07:
Thanks for reporting, both of you. Microsoft seems to use a lot more OUIs by now then we originally knew, so my heuristics prove to be very weak. I'll try to combat that with listing well known genuine but conflicting Microsoft OUIs with a new quirk flag to bypass heuristics.

closed 2024-12-07 by kakra

sdlea, 2024-12-07:
Thank YOU for your FIX. Glad that I can make a little bit of contribution.

d-k-c, 2024-12-08:
I can confirm that the controller now rumbles for me as well! Thank you so much for the quick fix!

cedarrapidsboy, 2024-12-18:
> Thanks for reporting, both of you. Microsoft seems to use a lot more OUIs by now than we originally knew, so my heuristics prove to be very weak. I'll try to combat that with listing well known genuine but conflicting Microsoft OUIs with a new quirk flag to bypass heuristics.

I added a new line for my own Xbox Series X|S controller. Thanks for this discussion, now it works in Moonlight.

`DEVICE_OUI_QUIRK("3C:FA:06", XPADNEO_QUIRK_NO_HEURISTICS),`