atar-axis/xpadneo #515: Xbox Series X controller not working on Lenvo P14s Gen 1

ipkpjersi, 2024-12-09:
## Version of xpadneo

0.9.6 FINAL

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [X] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [X] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [x] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [x] I don't use native games or did not try
- [X] Other software is having issues (describe software and issues below)
- [X] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [x] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [x] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [X] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

I can pair my Xbox Series X controller with bluetooth and have it connect, and it stays connected, but it doesn't show in `sudo evtest` and it doesn't appear in Steam games via wine/Proton, RPCS3, etc. I tried with the same Bluetooth adapter and Xbox controller on my desktop and it works fine on my desktop, but not on my laptop. I also tried multiple other Bluetooth adapters as well.

The reason why I think it's a driver issue / issue with xpadneo is because the same controller and same Bluetooth adapter work on my desktop but not on my laptop, and because the controller can pair, connect and stay connected, it just doesn't function in games and doesn't get recognized as an input device.

## Steps to Reproduce

1. Install latest xpadneo 0.9.6 FINAL
2. Connect any Bluetooth adapter or use internal Bluetooth adapter to Lenovo P14S Gen 1.
3. Try to load any games or `sudo evtest` or anything, the controller doesn't show up.

## Expected Behavior

I expect my controller to behave like it does on my desktop. Using the same bluetooth adapter on my desktop, I can pair and connect to the controller and use it in Steam games via wine/Proton, RPCS3, etc. 

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

```

## Controller and Bluetooth Information
FWIW, these logs are not from the known-working adapter as I'm currently using that adapter on my desktop, but these are from an Avantree DG45 which I would think should work too.

[xpadneo-btmon.txt](https://github.com/user-attachments/files/18065728/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/user-attachments/files/18065729/xpadneo-dmesg.txt)

[xpadneo-lsusb.txt](https://github.com/user-attachments/files/18065730/xpadneo-lsusb.txt)

## Additional Context

If you need any additional information, please let me know. Thanks as always for this amazing project!

kakra, 2024-12-09:
Please try to unpair (or at least turn off) the Sonova device and see if it does work then. We've seen some issues with (probably) bandwidth reservation conflicting with the Xbox controller connection. If the controller then connects, you can turn the audio device back on to check if it still connects. If it does, that's your workaround.

Since this works on different hardware, it's most likely a driver or firmware issue with a different OS version. It looks like Intel Bluetooth chipsets are affected by this.

ipkpjersi, 2024-12-09:
Hi,

Sorry, I'm not sure what a Sonova device is. When searching for Bluetooth devices for pairing I saw a 65 inch TV but I don't have one (only 75 and 55), so I'm guessing maybe both devices are from my neighbour, even though I live in a house maybe Bluetooth is really powerful after all...

In blueman-manager for example, I only see one device. I can see the Xbox Wireless Controller is connected, and it says that it is a "Gamepad", it can connect when powering the controller on, and it stays connected never disconnecting even once until I manually disconnect/turn off my controller, it's just it's like my laptop doesn't see it as an input device for some reason.

The fact that it's likely a third party bug like a driver or firmware issue is disheartening, I was hoping I was just overlooking something simple here and maybe I was missing a flag I could try or something.

In terms of Intel Bluetooth chipsets, I'm taking a look at my lsusb log and it looks like the adapter I'm currently testing is a Realtek, not an Intel unless I'm mistaken?

Since it's likely a third party bug, do you know where I could post my issues to possibly get more help about it? Normally I use Ubuntu where everything usually just works :tm: and I'm trying out Linux Mint now, and I was kind of surprised to see that I'm having problems with something that was working so well for me on Ubuntu when Linux Mint is based off of Ubuntu.

Thanks again.

ipkpjersi, 2024-12-09:
Apologies for the double post, but I just realized that I wasn't actually using 0.9.6 final, I was actually using the master branch. When I ran the uninstaller to remove the master version, I noticed this:
```
sudo ./uninstall.sh
* unloading current driver module
modprobe: FATAL: Module hid_xpadneo not found.
* looking for registered instances
found 1 registered instance(s) on your system
* v0.9-190-ged56962
  * uninstalling and removing v0.9-190-ged56962 from DKMS
Module hid-xpadneo-v0.9-190-ged56962 for kernel 6.8.0-38-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

hid-xpadneo.ko.zst:
 - Uninstallation
   - Deleting from: /lib/modules/6.8.0-38-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

Running the post_remove script:
Uninstalling ERTM override...
Uninstalling modalias database...
Uninstalling udev rules...
Reloading udev...
depmod...
Deleting module hid-xpadneo-v0.9-190-ged56962 completely from the DKMS tree.
  * removing v0.9-190-ged56962 folder from /usr/src
```

It looks like there's no module hid_xpadneo. Isn't this a kernel-level driver, though? That's probably not good, right?

I ran the installer for 0.9.6:
```
sudo ./install.sh  
* creating dkms.conf
* adding hid-xpadneo-v0.9.6 folder to /usr/src
* installing module (using DKMS)
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der
Creating symlink /var/lib/dkms/hid-xpadneo/v0.9.6/source -> /usr/src/hid-xpadneo-v0.9.6

Building module:
Cleaning build area...
make -j16 KERNELRELEASE=6.8.0-49-generic -C /lib/modules/6.8.0-49-generic/build M=/var/lib/dkms/hid-xpadneo/v0.9.6/build/src VERSION=v0.9.6 modules...
Signing module /var/lib/dkms/hid-xpadneo/v0.9.6/build/src/hid-xpadneo.ko
Cleaning build area...

hid-xpadneo.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.0-49-generic/updates/dkms/

Running the post_install script:
Not disabling ERTM, kernel version doesn't require it...
Installing modalias database...
Installing udev rules...
Reloading udev...
depmod...
```

I manually loaded hid_xpadneo just as a test, and then opened up RPCS3, and then suddenly, my Xbox controller works with SDL and evdev!

So either it's a regression, or something didn't work properly the first time I installed it.

kakra, 2024-12-09:
If the driver does not autoload, check if the udev rules are properly in place...

> In terms of Intel Bluetooth chipsets, I'm taking a look at my lsusb log and it looks like the adapter I'm currently testing is a Realtek, not an Intel unless I'm mistaken?

Well, according to your bluez log, it's Intel:
> `= Index Info: C0:3C:59:1E:B2:65 (Intel Corp.)                   [hci0] 0.405699`

But you're right, there are multiple adapters, and it looks like you're trying to pair via:
> `= Index Info: 00:E0:4C:3.. (Realtek Semiconductor Corporation)  [hci2] 0.405692`

Just in case you're seeing problems, retry with the Intel HCI.

I'm currently looking into problems with DKMS. There are reports which say that DKMS stops working properly if stale driver data of xpadneo is left in `/var/lib/dkms`. You can try to clean up there.

This one is raising an eyebrow:
```
sudo ./uninstall.sh
* unloading current driver module
modprobe: FATAL: Module hid_xpadneo not found.
```

Aha... unloading does not work because it's not loaded. But then:
```
* v0.9-190-ged56962
  * uninstalling and removing v0.9-190-ged56962 from DKMS
Module hid-xpadneo-v0.9-190-ged56962 for kernel 6.8.0-38-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.
```

Now it claims the module has been active?

Well... There may be problems in xpadneo but Bluetooth connectivity is never a problem of xpadneo itself. And your logs look like it simply won't connect. If the real issue is that xpadneo doesn't properly install and load, that's something we should look into.

Can you retry the logs after manually loading xpadneo? (if autoloading is still an issue after cleaning up)

kakra, 2024-12-09:
BTW: Using master is fine, that version is almost finished, most bug fixes are in and I'm near to release v0.9.7 before merging the new features of v0.10.

ipkpjersi, 2024-12-09:
It seems both xpadneo master and 0.9.6 are working fine for me, after uninstalling and installing, I can even reboot my computer and it works automatically. It was connecting via Bluetooth just fine and staying connected too, so I think maybe the logs were just lying to us lol or something else was going on.

I'm not sure how much more my logs will help given that I'm not able to replicate my issue I had earlier, everything seems to be working as expected now.

I think I can probably just keep an eye on kernel updates and see if it survives kernel updates or if I run into this issue again. We could probably even close this issue for now tbh since I can't seem to replicate this issue anymore.

ipkpjersi, 2024-12-10:
I'll go ahead and close this for now since I can't seem to replicate the issue anymore. I'll re-open this if I run into it again. Cheers.

closed 2024-12-10 by ipkpjersi