atar-axis/xpadneo #519: Inconsistencies when using multiple 8BitDo Pro 2 Contollers

starmatt, 2025-01-07:
## Version of xpadneo

`xpadneo-dkms-0.9.7-1` from AUR.

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [X] Other: 8BitDo Pro 2 Controllers

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [X] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [X] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [X] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [X] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [X] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [X] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [X] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

Hello ! First thanks for the work on xpadneo, it solves the input lag issues I have with my 8BitPro 2 Controllers (via on-board bluetooth on my Framework 13 laptop), but there is something weird about connecting multiple controllers (I own 3).
First I need to preface something, I _think_ that with the latest firmware, when connecting a 8BitDo Pro 2 controller in _windows_ mode (the back switch pointing to X), the ABXY buttons are reversed by default (as to mimic an xbox controller configuration), [now the driver's documentation](https://atar-axis.github.io/xpadneo/#8bitdo-controllers) says that it also swaps these buttons, which means that the default behaviour mimics a Switch Pro controller configuration (adding the quirks fixes that).
The core of my problem is that when connecting multiple controllers, they don't behave the same way, and I'm not sure why and what is happening:
1. The first controller that is connected will have its ABXY buttons working as described above. <-- To me, that's the expected behaviour.
2. The other controllers that are connected afterwards, and that until I reboot the computer, their buttons wont be reversed by default and use standard Switch Pro configuration. If I disconnect the first controller and reconnect it, it will also use standard Switch Pro configuration.

One other inconsistency I can see is that, in KDE game controller settings, the first controller that gets connected is recognized as a Xbox Series S controller, and all other controllers are recognized as Xbox 360 controllers.

Thank you for your attention and help ! I'll edit this post as I continue searching and trying to fix this issue.

## Steps to Reproduce

Connect multiple 8BitDo Pro 2 controllers.

## Expected Behavior

All the controllers to behave the same way.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux arch-fw 6.12.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 02 Jan 2025 22:52:26 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01  u.................%.
000000b4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000000c8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
000000dc: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
000000f0: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000104: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000118: 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75  .u........... ..&..u
0000012c: 08 95 01 81 02 c0                                            ......
3445511648 1458
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/user-attachments/files/18331327/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/user-attachments/files/18331350/xpadneo-dmesg.txt)

Bus 003 Device 004: ID 8087:0032 Intel Corp. AX210 Bluetooth

[xpadneo-lsusb.txt](https://github.com/user-attachments/files/18331368/xpadneo-lsusb.txt)

## Additional Context

kakra, 2025-01-07:
The inconsistency is not the different controller, it's the different device API: Do not compare hidraw with evdev. They work differently when it comes to order of axis and buttons. There's no inconsistency that I can see here.

closed 2025-01-07 by kakra

starmatt, 2025-01-07:
Thanks for answering, could you explain why the controllers use different APIs then, shouldn't they all use the same ?

kakra, 2025-01-07:
Each controller should appear twice or three times: hidraw (hidrawXX), evdev (eventXX), joydev (jsXX). You can only compare controllers using the same interface type.

If you're missing controllers on some interface types, your problem is most likely due to having Qmk or OpenRGB installed which mess with the hidraw device permissions. xpadneo devices should never be visible on hidraw. Or one of your device is not connected via xpadneo...

starmatt, 2025-01-07:
According to what I can understand from `dmesg`, they should be :P
But I'll look into it, thanks.

kakra, 2025-01-07:
xpadneo should not be forced to load at boot if it is managed by dkms, otherwise the boot process may hang. I know that this is a recommendation that popped up in one or another forum, but it autoloads via udev (tho, this may create a double rumble on first connect).

Subsequent connects won't have to load xpadneo until you reboot. The second device should automatically bind to xpadneo.

Usually, the first connect always first binds to `hid-generic` or `hid-microsoft`, then udev loads the module and rebinds the device, as you can see by the first `xpadneo XXXX: input,hidraw` line after the `microsoft XXXX: input,hidraw` line.

Adding a second controller should immediately bind to xpadneo then. If it doesn't but to a different driver instead, xpadneo doesn't detect it. xpadneo only supports HID device, raw USB-input won't work (bus 3). USB-HID would work, tho. But I don't know any Xbox-compatible driver which uses HID on USB. Bus 5 (Bluetooth) is always HID.

You can monitor this process of detection and binding with `udevadm monitor -p`.

I wonder why in your dmesg the controller seems to move from hidraw4 to hidraw5... Or did you connect two controllers? It looks like there are two different firmwares, and only the newer firmware properly binds to xpadneo.

You can see the bindings via `ls -ld /sys/class/input/*/*/*/driver`. udev rebinds the device by first writing the device path to the unbind file in the driver subdirectory, and then writing it to the bind file of the hid-xpadneo driver instead.

starmatt, 2025-01-07:
Alright, after trying a bunch of random stuff, it turns my issue stemed from how I was trying to add the quirk flags as described in the documentation: https://atar-axis.github.io/xpadneo/#8bitdo-controllers

I was trying to add them to a file in the folder `/etc/modprobe.d/` as per the archlinux wiki specifies, since adding them to `/etc/modprobe.conf` did nothing. Removing that file fixed the issue, and all controllers connect correctly then.

Here the content of that file:
```
# /etc/modprobe.d/xpadneo.conf
options hid_xpadneo quirks=E4:17:D8:94:84:E7+32,E4:17:D8:D1:E9:A8+32,E4:17:D8:E6:DC:C6+32
```

So I'm still trying to debug this, but I'm getting closer.

Thanks a lot @kakra for all your help!

kakra, 2025-01-07:
Ideally, quirks should be handled automatically and only used as documented, if auto-detection does not work.

starmatt, 2025-01-07:
Sure, but I as described in the original post, when you connect a 8BitPro 2 in PC mode, the ABXY buttons are already swapped by default. And then xpad neo also swaps them, and they end up A = B, X = Y, etc. (and from reading the doc I don't think that's the intended behaviour). I was hoping that setting the quirk flag would fix that.

kakra, 2025-01-07:
Then we should add a quirk flag. But we should open a new issue for that matter.

kakra, 2025-01-07:
> as per the archlinux wiki specifies

Ah wait, I see why: The 0.9 branch does not support the `+` syntax to add additional quirks. So you need to manually combine the quirk flags with your intended quirk and provide the result as `00:11:22:33:44:55:quirks`, see `modinfo hid-xpadneo`: if will show a list of possible parameters.