atar-axis/xpadneo #524: Xbox One S Controller Disconnects Randomly With TP-Link UB400 Bluetooth Adapter

DCeric, 2025-03-05:
If I need to get a log tell me the command. Anyways the issue is that every now and then the controller disconnects and reconnects randomly. I have it set to trusted and the Bluetooth set to always visible. I've been trying to figure out the issue and what needs to be done.

kakra, 2025-03-05:
Please send at least the output of dmesg after connecting your controller. Note however, xpadneo itself cannot change anything about the disconnects, it's an issue from another driver layer. But dmesg may provide some hints to help.

DCeric, 2025-03-06:
It's not allowing me to send it for some reason. It is a rather large log.

DCeric, 2025-03-06:
`[25640.240351] init: Service 'vendor.hwcomposer-2-1' (pid 5073) exited with status 1
[25640.240381] init: Sending signal 9 to service 'vendor.hwcomposer-2-1' (pid 5073) process group...
[25640.240457] libprocessgroup: Successfully killed process cgroup uid 1000 pid 5073 in 0ms
[25640.240510] init: Sending signal 9 to service 'surfaceflinger' (pid 130) process group...
[25640.240562] libprocessgroup: Successfully killed process cgroup uid 1000 pid 130 in 0ms
[25645.219526] init: starting service 'vendor.hwcomposer-2-1'...
[25645.222523] libprocessgroup: Failed to make and chown /acct/uid_1000: Read-only file system
[25645.222620] init: createProcessGroup(1000, 5076) failed for service 'vendor.hwcomposer-2-1': Read-only file system
[25645.249626] init: Service 'vendor.hwcomposer-2-1' (pid 5076) exited with status 1
[25645.249646] init: Sending signal 9 to service 'vendor.hwcomposer-2-1' (pid 5076) process group...
[25645.249694] libprocessgroup: Successfully killed process cgroup uid 1000 pid 5076 in 0ms
[25645.249729] init: Sending signal 9 to service 'surfaceflinger' (pid 130) process group...
[25645.249759] libprocessgroup: Successfully killed process cgroup uid 1000 pid 130 in 0ms
[25646.864642] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B20.0005/input/input26
[25646.864846] hid-generic 0005:045E:0B20.0005: input,hidraw2: BLUETOOTH HID v5.23 Gamepad [Xbox Wireless Controller] on 30:68:93:93:33:76
[25646.869318] loaded hid-xpadneo v0.9-214-g8d20a23
[25646.896454] xpadneo 0005:045E:0B20.0005: BLE firmware version 5.23
[25646.896466] xpadneo 0005:045E:0B20.0005: pretending XB1S Windows wireless mode (changed PID from 0x0B20 to 0x028E)
[25646.896470] xpadneo 0005:045E:0B20.0005: working around wrong SDL2 mappings (changed version from 0x00000523 to 0x00001130)
[25646.896477] xpadneo 0005:045E:0B20.0005: report descriptor size: 283 bytes
[25646.896481] xpadneo 0005:045E:0B20.0005: fixing up Rx axis
[25646.896484] xpadneo 0005:045E:0B20.0005: fixing up Ry axis
[25646.896486] xpadneo 0005:045E:0B20.0005: fixing up Z axis
[25646.896489] xpadneo 0005:045E:0B20.0005: fixing up Rz axis
[25646.896492] xpadneo 0005:045E:0B20.0005: fixing up button mapping
[25646.896907] xpadneo 0005:045E:0B20.0005: gamepad detected
[25646.896912] xpadneo 0005:045E:0B20.0005: enabling compliance with Linux Gamepad Specification
[25646.897042] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B20.0005/input/input27
[25646.897435] xpadneo 0005:045E:0B20.0005: input,hidraw2: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on 30:68:93:93:33:76
[25646.897522] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B20.0005/input/input28
[25646.897614] xpadneo 0005:045E:0B20.0005: consumer control added
[25646.897696] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B20.0005/input/input29
[25646.897786] xpadneo 0005:045E:0B20.0005: keyboard added
[25646.897795] xpadneo 0005:045E:0B20.0005: controller quirks: 0x00000050
[25646.897799] xpadneo xpadneo_welcome_rumble start
[25646.897802] xpadneo 0005:045E:0B20.0005: testing weak motor: sustain 50ms release 50ms loop 2 wait 30ms
[25647.227868] xpadneo 0005:045E:0B20.0005: testing strong motor: sustain 50ms release 50ms loop 2 wait 30ms
[25647.558009] xpadneo 0005:045E:0B20.0005: testing trigger motors: sustain 50ms release 50ms loop 2 wait 30ms
[25647.888102] xpadneo xpadneo_welcome_rumble took 991ms
[25647.888155] xpadneo 0005:045E:0B20.0005: Xbox Wireless Controller [98:7a:14:a5:bc:d9] connected
[25647.923228] xpadneo 0005:045E:0B20.0005: reverting to original version (changed version from 0x00001130 to 0x00000523)
[25647.923244] xpadneo 0005:045E:0B20.0005: reverting to original product (changed PID from 0x028E to 0x0B20)
[25647.964192] xpadneo 0005:045E:0B20.0005: BLE firmware version 5.23
[25647.964197] xpadneo 0005:045E:0B20.0005: pretending XB1S Windows wireless mode (changed PID from 0x0B20 to 0x028E)
[25647.964200] xpadneo 0005:045E:0B20.0005: working around wrong SDL2 mappings (changed version from 0x00000523 to 0x00001130)
[25647.964204] xpadneo 0005:045E:0B20.0005: report descriptor size: 283 bytes
[25647.964207] xpadneo 0005:045E:0B20.0005: fixing up Rx axis
[25647.964208] xpadneo 0005:045E:0B20.0005: fixing up Ry axis
[25647.964210] xpadneo 0005:045E:0B20.0005: fixing up Z axis
[25647.964211] xpadneo 0005:045E:0B20.0005: fixing up Rz axis
[25647.964212] xpadneo 0005:045E:0B20.0005: fixing up button mapping
[25647.964422] xpadneo 0005:045E:0B20.0005: gamepad detected
[25647.964424] xpadneo 0005:045E:0B20.0005: enabling compliance with Linux Gamepad Specification
[25647.964481] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B20.0005/input/input30
[25647.964657] xpadneo 0005:045E:0B20.0005: input,hidraw2: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on 30:68:93:93:33:76
[25647.964697] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B20.0005/input/input31
[25647.964737] xpadneo 0005:045E:0B20.0005: consumer control added
[25647.964772] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B20.0005/input/input32
[25647.964821] xpadneo 0005:045E:0B20.0005: keyboard added
[25647.964825] xpadneo 0005:045E:0B20.0005: controller quirks: 0x00000050
[25647.964827] xpadneo xpadneo_welcome_rumble start
[... cut ...]
[25650.254995] init: Service 'vendor.hwcomposer-2-1' (pid 5079) exited with status 1
[25650.255008] init: Sending signal 9 to service 'vendor.hwcomposer-2-1' (pid 5079) process group...
[25650.255041] libprocessgroup: Successfully killed process cgroup uid 1000 pid 5079 in 0ms
[25650.255064] init: Sending signal 9 to service 'surfaceflinger' (pid 130) process group...
[25650.255083] libprocessgroup: Successfully killed process cgroup uid 1000 pid 130 in 0ms
[25655.235731] init: starting service 'vendor.hwcomposer-2-1'...
`

kakra, 2025-03-06:
I wanted to see this:
```
[25646.896454] xpadneo 0005:045E:0B20.0005: BLE firmware version 5.23
```

This firmware version should be fine and stable with your Bluetooth dongle.

```
[25648.955141] xpadneo xpadneo_welcome_rumble took 991ms
```
Timing is quite good. It should take 990ms exactly but this is just 1ms off. I don't think that indicates any issue with the Bluetooth drivers.

> There is multiples of the following:

Probably unrelated but you may want to look into it. I've never seen such a message. Is this a system running snaps? You should probably ask your distribution forum about these...

According to your description and the logs, I'd say you should try setting Bluetooth parameters:
https://github.com/atar-axis/xpadneo/blob/master/docs/TROUBLESHOOTING.md

Especially "High Latency or Lost Button Events with Bluetooth LE" (improves connection stability) and "Gamepad Does not Connect at All, Runs A Reconnect Loop, or Immediately Disconnects" (improves reconnect behavior) may help your problem.

To apply the changed parameters to the device connection, you may need to remove the pairing from your Bluetooth applet (delete it, don't just disconnect it), and pair the device freshly - otherwise it may use cached parameters.

This is mostly everything we can provide from xpadneo as troubleshooting steps.

DCeric, 2025-03-06:
So what should I need to type into terminal?

DCeric, 2025-03-06:
I only installed snaps for MakeMKV cause that was the only way I could get it to work with Linux Mint 22.

kakra, 2025-03-06:
> So what should I need to type into terminal?

Follow the troubleshooting instructions I've linked. There's no magic command you could type into the terminal. If you need more help with troubleshooting, ask our Discord community. But please read those instructions first.

You may also try AI, I've prepared it for you:
https://chatgpt.com/share/67c9f3d6-5944-8008-a08b-b5c0ca4f5b4c

DCeric, 2025-03-06:
Thanks for the help I'll see if the one mentioned in the troubleshooting doc helps

DCeric, 2025-05-23:
The issue was related to KDE Connect. Since uninstalling it, it works perfectly now.

closed 2025-05-23 by DCeric

kakra, 2025-05-30:
> The issue was related to KDE Connect. Since uninstalling it, it works perfectly now.

Yep, this is known. If you are able to build kdeconnect without Bluetooth support, it works just fine (it works perfectly fine without it if your phone joins the same wifi anyways). The problem basically is that kdeconnect spams the Bluetooth radio with discovery requests every few seconds. During this time, some time-critical packets aren't properly exchanged, and that is when the Xbox firmware disconnects (it probably shouldn't but that's the way it is).