atar-axis/xpadneo #525: Controller not recognized even after connecting

The-Pezman, 2025-03-15:
## Version of xpadneo
The most recent (literally just downloaded it)

## Controller Model

- [ ] Xbox Elite 2 controller

### Connection mode
- [ ] Bluetooth connection

## Installed Software
None of the ones listed

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

It is not recognized at the OS level

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] It does not work at all

## Describe the Bug
I have installed the software and connected the controller, but the OS does not recognize the controller at all

## Steps to Reproduce
Install xpadneo, restart system, pair controller

## Expected Behavior
I would expect to see the controller listed under the "game controller" Kubuntu settings menu

## Screenshots / GIFs / Videos
[image]

## System Information

```console
# uname -a

```
[image]

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

```
[image]

## Controller and Bluetooth Information

I wasn't really clear on how to do these steps. Please let me know if they end up being necessary but I've skipped them for now.

## Additional Context

Sch1nken, 2025-03-18:
I have a very similiar (if not the same) issue. Controller was working fine until it didn't. I can't really say when the issue started, since I haven't used a controller on this system for a few weeks.

Xbox One Controller
Version: 0.9.7 (reinstalled several times throughout testing). 

> uname -a
> Linux schinken-pc 6.13.7-1-MANJARO #1 SMP PREEMPT_DYNAMIC Thu, 13 Mar 2025 14:03:23 +0000 x86_64 GNU/Linux

Firmware updated via Windows 10 (`dmesg` told me to update :).

- Controller worked fine before
- Connection works flawlessly (no PIN, not disconnects, it just connects and vibrates, dmesg prints xpadneo stuff)
- Gamepad is not detected in KDE Game Controller Window
- Gamepad is not detected in Games. Tried in Godot Game Engine and a Web-based Gamepad Tester (which worked without issues before)
- I can see inputs coming in via `evtest` (didn't check if they were the correct inputs, but all buttons and axes respond)
- So somehow they're not translated to something the system/some gamepad api can understand?

I also tried returning to a lower kernel version (6.12.19) but no luck.

Might also be interesting: I did fiddle around with connecting connecting a Wiimote via Bluetooth inbetween it working and not working (might be the cause, but doesn't have to be). I did have to change some bluetooth configs on my system. But I am pretty sure I put everything back uninstall all modules and driver files as well as specific udev rules for the wiimote. (For bluetooth settings I checked with `paccat bluez -- main.conf` as well as input.conf).

kakra, 2025-03-18:
> Version: 0.9.7 (reinstalled several times throughout testing).

@Sch1nken Reinstalling doesn't affect the connection at all: xpadneo doesn't handle the connection, it even doesn't know about Bluetooth.

Connection problems are rather located in the Bluetooth stack (bluez, BT firmware, BT drivers, chipset).

Your dmesg log shows that the connection seems stable. Did the controller rumble? If yes, the connection generally works in both directions (we got info, and we sent data).

If not, the connection dropped, and you should check the BT stack instead.

Now try `evtest`, select the controller by typing its number.

If it is not listed, the connection is gone again. Check your BT stack. If it's a permission issue, `sudo evtest` should see the controller. Then see below.

If it is listed, generate some input.

Does it generate a constant flow of input data during continuous input? Does it show button presses immediately and precisely (no bounces, no missing or late events)? Then everything is good. If your software is still not seeing the controller, your issue is not xpadneo itself but maybe permissions or missing libraries.

If input data stuffers, drops, or misses events, then your BT connection is unstable. Check your BT stack.

We also have some troubleshooting instructions in the docs folder.

kakra, 2025-03-18:
> Screenshots

@The-Pezman Your screenshot shows no output for running `install.sh`. Usually, I'd expect to see some output.

As a first step, please try:
```bash
cd ~/Desktop/xpadneo
sudo ./install.sh -v
```

Let me know if it worked, and what output you got.

The-Pezman, 2025-03-18:
I got:
./install.sh: invalid option -- 'v'

I downloaded xpadneo again, tried the same thing and got the same result.

[image]

kakra, 2025-03-18:
Sorry, I got it wrong. Use `sudo ./install.sh --verbose`.

The-Pezman, 2025-03-18:
* verbose mode enabled
+ source lib/installer.sh
++ '[' -n '' ']'
++ '[' 0 -ne 0 ']'
+++ git rev-parse --show-toplevel
++ GIT_ROOT=/home/onair/Desktop/xpadneo
+++ __version
+++ git describe --tags --dirty
++ VERSION=v0.9-214-g8d20a23
+++ type -p dkms
++ DKMS_BIN=

kakra, 2025-03-18:
dkms is not installed. I should probably add something to detect this and actually show an error message for it.

But which distribution are you using? As an example, on Gentoo, you usually don't use dkms although it is available: xpadneo would be installed differently here. On most other distributions, dkms is the way to go.

The-Pezman, 2025-03-19:
Ah geez, you're right. I completely misread the directions, and thought the directions that did need DKMS were for those that didn't.

However, my problems are not over. While I did install the necessary packages and xpadneo itself also installs, now the controller constantly connects/disconnects, which persists even after a restart.

[image]

Sch1nken, 2025-03-19:
> Now try `evtest`, select the controller by typing its number.
> 
> If it is not listed, the connection is gone again. Check your BT stack. If it's a permission issue, `sudo evtest` should see the controller. Then see below.
> 
> If it is listed, generate some input.
> 
> Does it generate a constant flow of input data during continuous input? Does it show button presses immediately and precisely (no bounces, no missing or late events)? Then everything is good. If your software is still not seeing the controller, your issue is not xpadneo itself but maybe permissions or missing libraries.
> 
> If input data stutters, drops, or misses events, then your BT connection is unstable. Check your BT stack.

Inputs both through evtest and jstest where working without issues, it was just that other programs did not "see" the gamepad. I found the issue. As stated earlier I tried using a wiimote by using some old packages and drivers. One of those packages was "moltengamepad". Even though tracing the udev rules with udevadm did not show moltengamepad interacting in any way, it somehow still interfered. 

I noticed that `sudo evtest` could see the controller while `evtest` could not. I then checked `ls -la /dev/input` and saw that the group of both event19 (Xbox Gamepad) and js0 where set to `gamepad` and not `input`. Quick removal of moltengamepad and the unused group from /etc/group fixed it for me. 

Maybe it will help someone else as well (at least check the groups/permissions of your `/dev/input` files :).

kakra, 2025-03-19:
Okay so both of you saw different problems. Good that one is solved, and we should document checking permissions and identify conflicting software.

Now, for the issue of @The-Pezman: Repeating disconnects and reconnects are usually a problem with your BT stack. Either, the controller firmware isn't updated, or your chipset is generally incompatible (Xbox controllers have problems with certain chipsets). We have a list of tested chipsets and documented some tunables for bluez in the docs folder.

The-Pezman, 2025-03-20:
I see the docs folder, but I'm not sure what to look for in there, what a tunable is, or what specific actions to take.

kakra, 2025-03-20:
@The-Pezman Then please go to your Discord community server. There should be people to take you through the process step by step.

kakra, 2025-03-20:
It's right here:
[image]

kakra, 2025-12-15:
Yes, this is a known problem and the installer warns about it. I still do not know why distributions don't ship it built into the kernel. The problem is: bluez should pull the module in if it detects a device connecting that uses uhid for input. It's not something that xpadneo should do because we are not a Bluetooth driver but a HID driver. If bluez doesn't provide HID to us as intended, the controller cannot work. xpadneo doesn't even know that the controller connected via Bluetooth unless uhid provides the device.