atar-axis/xpadneo #529: Rumble not working

Laozi1527, 2025-04-06:
## Version of xpadneo

v0.9.7 HO-HO-HO EDITION

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [X] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [X] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

Rumble does not work via Bluetooth. Even though

- evtest shows FF_RUMBLE
- logs show the xpadneo welcome rumble (no rumble happend)
- python script with evdev does not trigger rumble with no error at compiling
- rumble works over USB (exactly the same python script)
- c_hidraw test works completly fine over bluetooth

## Steps to Reproduce

1. use bluetooth to connect the controller
2. execute this python script:

```python3
from evdev import InputDevice, ecodes, ff

dev = InputDevice('/dev/input/event4')

rumble = ff.Rumble(strong_magnitude=0xFFFF, weak_magnitude=0xFFFF)  

effect_type = ff.EffectType(ff_rumble_effect=rumble)

effect = ff.Effect(
    ecodes.FF_RUMBLE,  
    -1,  
    0,  
    ff.Trigger(0, 0),  
    ff.Replay(duration_ms, 0),  
    effect_type  
)

effect_id = dev.upload_effect(effect)

dev.write(ecodes.EV_FF, effect_id, 1)

import time
time.sleep(1)

dev.erase_effect(effect_id)
```

3. No error, no rumble
4. run c_hidraw  ->  works just fine
5. connect via USB  ->  works just fine

## Expected Behavior

rumble triggering just like with c_hidraw or when connected with USB

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux dev 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

## Controller and Bluetooth Information

I have no dongle because I'm on a rasp and use the build in bluetooth

[xpadneo-btmon.txt](https://github.com/user-attachments/files/19621908/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/user-attachments/files/19621909/xpadneo-dmesg.txt)

## Additional Context

The problem appears specific to evdev with FF. Reading inputs over evdev works fine.

kakra, 2025-04-06:
Please check if another software is masking your controller, e.g. Steam Input may take over the controller and may not support rumble through evdev.

Please also check if you have options set for rumble attenuation in xpadneo: hidraw will ignore those but the internal rumble test and the evdev interface probably won't ignore those.

That said, one of the next xpadneo versions will drop support for ff_memless and only support rumble: we really don't need all the emulated force feedback types (sawtooth, sine, square...) because no game uses it. Every game implements just plain rumble, and that will get rid of some aliasing effects we get with ff_memless due to its internal emulation timer.

Laozi1527, 2025-04-06:
The os is completely clean with its only purpose to get rumble running (No other software). I have rumble attenuation set to zero and I think as you can see, everything looks alright in dmesg. The error still remain exclusive to bluetooth, so I think evdev isn't the problem. I have tried many things and have no idea what the problem is. Hope you have an idea!

Laozi1527, 2025-04-06:
Just noticed that it applies the wrong quirks: [ 3125.292485] xpadneo 0005:045E:0B13.0005: enabling heuristic GameSir Nova quirks

kakra, 2025-04-06:
Ah yes... oh well, another one of those false detections. Please try the patch from the referenced commit above.

kakra, 2025-04-07:
Will be closed via commit, reopening...

closed 2025-04-07 by kakra