atar-axis/xpadneo #536: Xbox One S|X controller disconnects sporadically and then reconnects

poperigby, 2025-04-27:
## Version of xpadneo

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
  - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
  - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
  - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
  - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
  - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
  - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [x] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [x] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
Every once in a while, my controller will disconnect. The light will stay solid but the controller won't respond to input, then the light will start blinking and the controller will reconnect.

## Steps to Reproduce
1. Connect the controller over Bluetooth
2. Keep it connected

## Expected Behavior
The controller never disconnects unless I manually disconnect it.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux snowy 6.14.2 #1-NixOS SMP PREEMPT_DYNAMIC Thu Apr 10 12:44:49 UTC 2025 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
2986910699 1363
```

## Controller and Bluetooth Information
[xpadneo-lsusb.txt](https://github.com/user-attachments/files/19932420/xpadneo-lsusb.txt)
[xpadneo-dmesg.txt](https://github.com/user-attachments/files/19932421/xpadneo-dmesg.txt)
[xpadneo-btmon.txt](https://github.com/user-attachments/files/19932422/xpadneo-btmon.txt)

## Additional Context

kakra, 2025-05-10:
This is most likely a problem with your Bluetooth chipset, or another interfering device or Bluetooth radio. It seems you're using two different Bluetooth chipset. Please try disabling or disconnecting one and try with the other, and vice verse.

poperigby, 2025-05-16:
That seems to have been the issue. Switching to just using my motherboard radio fixed it. Thanks!

closed 2025-05-16 by poperigby