atar-axis/xpadneo #539: Xbox One S controller not detected by AntiMicroX nor Bottles.

StarShine1A, 2025-05-10:
## Version of xpadneo

current master branch (git clone https://github.com/atar-axis/xpadneo.git)

## Controller Model

- [ x] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [ x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ x] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
  - [ x] I don't use Steam or did not try
- [ x] games running through Lutris, wine and/or Bottles are having issues
  - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
  - [x ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
  - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
  - [ x] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
  - [ x] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ x] I've tried disabling or running without above mentioned software
- [ x] It does not work at all
- [ x] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

I have two xbox one s controllers and both fail to be detected by either Bottles or AntiMicroX, even though the light stops blinking on both after connecting (and sometimes right after pairing). 
After pairing and trusting the controller, AntiMicroX detects them for a split second after connecting to them then nothing, even though the controller is still connected by bluetooth.
The controllers never rumbled.
I must say though that when pressing any key on the connected controller, I noticed they were detected by btmon.

It used to work on a previous version that I do not remember, although even then the Input was never siphoned directly to Bottles but had to be mapped by AntiMicroX even though the games supported the controller.

## Steps to Reproduce

git clone https://github.com/atar-axis/xpadneo.git
cd xpadneo
sudo ./install.sh
sudo bluetoothctl
[bluetooth]# scan on
[bluetooth]# pair <MAC>
[bluetooth]# trust <MAC>
[bluetooth]# connect <MAC>

MAC was required and I used DKMS as I use secureboot, having scan on or off during pairing didn't make a difference.

## Expected Behavior

Rumbling on [bluetooth]# connect <MAC> and the detection of the controller by AntiMicroX and Bottles.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux starpc 6.14.4-200.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Apr 25 15:45:16 UTC 2025 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/user-attachments/files/20138717/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/user-attachments/files/20138718/xpadneo-dmesg.txt)

## Additional Context

When connecting to the Controller I get this in the Logs app:
``` 1:16:14 PM (udev-worker): 0005:045E:0B20.0006: /etc/udev/rules.d/60-xpadneo.rules:2 Failed to write ATTR{/sys/devices/virtual/misc/uhid/0005:045E:0B20.0006/[drivers/hid:xpadneo]bind}="0005:045E:0B20.0006", ignoring: No such file or directory
 1:10:02 PM bluetoothd: Failed to confirm name for hci0: Failed (0x03)
12:45:57 PM (udev-worker): 0005:045E:02FD.0005: /etc/udev/rules.d/60-xpadneo.rules:2 Failed to write ATTR{/sys/devices/virtual/misc/uhid/0005:045E:02FD.0005/[drivers/hid:xpadneo]bind}="0005:045E:02FD.0005", ignoring: No such file or directory
12:45:40 PM bluetoothd: extract_int: Unexpected end of packet
12:43:51 PM (udev-worker): 0005:045E:02FD.0004: /etc/udev/rules.d/60-xpadneo.rules:2 Failed to write ATTR{/sys/devices/virtual/misc/uhid/0005:045E:02FD.0004/[drivers/hid:xpadneo]bind}="0005:045E:02FD.0004", ignoring: No such file or directory
12:43:33 PM bluetoothd: extract_int: Unexpected end of packet
12:35:11 PM (udev-worker): 0005:045E:02FD.0003: /etc/udev/rules.d/60-xpadneo.rules:2 Failed to write ATTR{/sys/devices/virtual/misc/uhid/0005:045E:02FD.0003/[drivers/hid:xpadneo]bind}="0005:045E:02FD.0003", ignoring: No such file or directory
12:34:47 PM bluetoothd: profiles/input/device.c:control_connect_cb() connect to 5C:BA:37:0E:A9:E2: Host is down (112)
12:34:10 PM (udev-worker): 0005:045E:02FD.0002: /etc/udev/rules.d/60-xpadneo.rules:2 Failed to write ATTR{/sys/devices/virtual/misc/uhid/0005:045E:02FD.0002/[drivers/hid:xpadneo]bind}="0005:045E:02FD.0002", ignoring: No such file or directory
12:33:40 PM bluetoothd: extract_int: Unexpected end of packet```

kakra, 2025-05-10:
According to dmesg, xpadneo doesn't even pickup the controller, and the udev rules reflect that. The problem is the error `hci0: Failed`. This indicates that your Bluetooth chipset cannot pair the controller.

In many instances, this can be fixed by updating the controller firmware. We have instructions made by a user here:
https://github.com/atar-axis/xpadneo/issues/472

If you need more help on updating the firmware, please join our Discord.

But it is also possible that your Bluetooth chipset is simply incompatible. We have a list of verified dongles and chipsets here:
https://github.com/atar-axis/xpadneo/blob/master/docs/BT_DONGLES.md

Unless dmesg even doesn't show the controller connected properly to xpadneo, we cannot really help here. The issue is outside of xpadneo.

StarShine1A, 2025-05-10:
So after updating the firmware of both Xbox One S controllers, I tried again to pair and connect them to the computer again.

``` 
 7:47:23 PM kernel: Bluetooth: hci0: ACL packet for unknown connection handle 3837
 7:45:10 PM kernel: Loading of module with unavailable key is rejected
  7:45:10 PM kernel: hid-generic 0005:045E:0B20.0003: input,hidraw1: BLUETOOTH HID v5.23 Gamepad [Xbox Wireless Controller] on 28:d0:43:13:82:2e
 7:45:10 PM kernel: input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B20.0003/input/input29
 7:43:31 PM kernel: Loading of module with unavailable key is rejected
 7:43:31 PM kernel: hid-generic 0005:045E:0B20.0002: input,hidraw1: BLUETOOTH HID v5.23 Gamepad [Xbox Wireless Controller] on 28:d0:43:13:82:2e
 7:43:31 PM kernel: input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B20.0002/input/input25
 7:42:28 PM kernel: evm: overlay not supported
 7:41:36 PM kernel: Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
 7:41:16 PM kernel: rfkill: input handler disabled
 7:41:16 PM kernel: Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
 7:41:15 PM kernel: Bluetooth: RFCOMM ver 1.11
 7:41:06 PM kernel: wlo1: associated
 7:41:03 PM kernel: Bluetooth: MGMT ver 1.23
 7:41:03 PM kernel: r8169 0000:05:00.0 eno1: Link is Down
 7:41:02 PM kernel: nvidia 0000:01:00.0: [drm] Cannot find any crtc or sizes
 7:41:02 PM kernel: [drm] Initialized nvidia-drm 0.0.0 for 0000:01:00.0 on minor 0
 7:41:02 PM kernel: Generic FE-GE Realtek PHY r8169-0-500:00: attached PHY driver (mii_bus:phy_addr=r8169-0-500:00, irq=MAC)
 7:41:02 PM kernel: NET: Registered PF_QIPCRTR protocol family
 7:41:02 PM kernel: mt7921e 0000:04:00.0 wlo1: renamed from wlan0
 7:41:02 PM kernel: block nvme0n1: No UUID available providing old NGUID
 7:41:02 PM kernel: Bluetooth: BNEP socket layer initialized
````

```
 7:47:23 PM kernel: Bluetooth: hci0: ACL packet for unknown connection handle 3837
 7:45:10 PM (udev-worker): 0005:045E:0B20.0003: /etc/udev/rules.d/60-xpadneo.rules:2 Failed to write ATTR{/sys/devices/virtual/misc/uhid/0005:045E:0B20.0003/[drivers/hid:xpadneo]bind}="0005:045E:0B20.0003", ignoring: No such file or directory
```

Is it dead?

kakra, 2025-05-11:
Please completely remove the device from Bluetooth first, potentially by stopping the bluez service, then purging the contents of `/var/lib/bluetooth`. Since the controller is "half paired" to the PC, pair it to another device like your smartphone first before retrying on the PC.

Here's another possible problem:
```
Loading of module with unavailable key is rejected
```

It looks like your system is using secure boot. To use xpadneo, you have to sign the modules yourself or use DKMS to install it. If this message comes from trying to load xpadneo, then xpadneo hasn't been properly signed.

You can verify by doing:
```bash
lsmod | grep xpadneo # verify it is not loaded
modprobe hid-xpadneo # try to load it
lsmod | grep xpadneo # verify is has been loaded
dmesg | tail -20 # check for any recent errors
```

StarShine1A, 2025-05-11:
```(base) star@starpc:~$ lsmod | grep xpadneo 
(base) star@starpc:~$ modprobe hid-xpadneo
modprobe: ERROR: could not insert 'hid_xpadneo': Operation not permitted
(base) star@starpc:~$ sudo lsmod | grep xpadneo 
(base) star@starpc:~$ sudo modprobe hid-xpadneo
modprobe: ERROR: could not insert 'hid_xpadneo': Key was rejected by service
(base) star@starpc:~$ sudo lsmod | grep xpadneo 
(base) star@starpc:~$ dmesg | tail -20 
dmesg: read kernel buffer failed: Operation not permitted```

In case it might be useful to know, I had to use DKMS before to sign my NVIDIA Driver.
I also used sudo ./install.sh while installing XPADNEO as recommended in the Docs as I am using secureboot.

kakra, 2025-05-11:
Keys are provided by you or your distribution, not by xpadneo. If you (or DKMS) created a key for xpadneo, you probably have to add it to some trusted keyring.

The failing modprobe command already shows that the key used to sign the module has been rejected. You need to ask your distribution support how to fix this. Also, run `sudo -i` to run sudo in interactive mode: then you don't need to specify `sudo` each time. You can leave this mode with `exit`. Your system seems to be extremely restricted, not even `dmesg` is allowed for ordinary users. Your system will probably cause more problems when trying to run games (either reduced performance or crashes due to permission restrictions).

I'm sorry to say that, but handling the special cases for your distribution is not the scope of this issue tracker. You can try asking for help in our Discord, but most likely you need to solve problems in your distribution before xpadneo even comes into play.

closed 2025-05-23 by StarShine1A