atar-axis/xpadneo #545: Controller won't connect

YLlywelyn, 2025-06-04:
## Version of xpadneo

v0.9-222-gcd25680

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [X] Other: xbox one controller

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
  - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
  - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
  - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
  - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
  - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
  - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [X] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

Controller constantly connects and disconnects until eventually turning off.

## Steps to Reproduce

Attempt to connect to the controller via gui or command line.

## Expected Behavior

I expect that the software will allow me to use my bluetooth controller via bluetooth.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
6.8.0-60-generic #63-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 15 19:04:15 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/user-attachments/files/20599884/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/user-attachments/files/20599903/xpadneo-dmesg.txt)

## Additional Context

kakra, 2025-06-05:
Please check our troubleshooting guide: https://atar-axis.github.io/xpadneo/#troubleshooting

lantgabor, 2025-06-06:
@kakra Havent solved it yet myself but thanks for the info. What you wrote down here is super useful would it be possible to add all of these to the faq/docs?

kakra, 2025-12-06:
Please post the logs from dmesg for both cases...