atar-axis/xpadneo #546: Gamepad doesn't rumble after connecting

AzyWng, 2025-06-23:
## Version of xpadneo
hid-xpadneo-v0.9-222-gcd25680

## Controller Model

- [X] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
  - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
  - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
  - [ ] I don't use native games or did not try
- [X] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
  - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
  - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
  - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [X] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [X] I probably didn't figure it all out but it's too early to give up
- [X] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
The gamepad previously would disconnect right after connecting and vice versa. However, now that I've updated the firmware, the gamepad connects but doesn't rumble, and I can't seem to get any input from it.

When installing xpadneo, the first time I was prompted to set up a new Machine Owner Key, but I forgot to do so (I accidentally selected on the wrong option after restarting and didn't know how to get back to the screen).

## Steps to Reproduce
1. Install xpadneo without setting up a new Machine Owner Key
2. Connect gamepad to computer
3. Gamepad connects but doesn't rumble, and button inputs don't get a response.

## Expected Behavior
I expected the gamepad to rumble upon connecting and that I'd be able to use it afterwards.

## System Information

```console
# uname -a
Linux AFrameWork 6.11.0-26-generic #26~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Apr 17 19:20:47 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory

```

kakra, 2025-06-23:
Sorry, please retry with `dmesg | grep -E 'xpadneo|hid' | tee dmesg.log`.

AzyWng, 2025-06-24:
Here's the output of `dmesg | grep -E 'xpadneo|hid' | tee dmesg.log`
[dmesg.log](https://github.com/user-attachments/files/20886779/dmesg.log)

kakra, 2025-06-24:
According to the log, your controller doesn't even show up as a raw HID device. This is independent of whether xpadneo works or doesn't. Your problem is in the Bluetooth stack, most likely some settings in bluez need to be adjusted, or your chipset in incompatible with the controller.

So this is not our bug, it's in the Bluetooth stack.

We can still provide some initial help:

1. Our Discord may be able to guide you through the troubleshooting steps (see "Chat" badge in main page)
2. We have some Bluetooth troubleshooting steps in the docs folder, also a list of compatible tested chipsets
3. Your bluez daemon may have trouble with cached attributes or pairing the controller at all, you may need to adjust settings and then maybe clear the attribute cache in `/var/lib/bluetooth`

You can also look in the collection of previous issues to see if your problem has been already solved:
https://github.com/users/atar-axis/projects/2

Sorry, I cannot provide extensive support here because this is a repeating issue, and it's even not in our driver. Please follow the above steps. You may re-open this report if the troubleshooting steps reveal an issue within xpadneo itself, but according to your log, xpadneo isn't even involved yet because the Bluetooth layer hasn't properly connected to the device.

AzyWng, 2025-06-24:
Hold up, I think I made a mistake when I did the command previously. When I'd previously posted the dmesg.log I'd forgotten to actually turn on and connect my controller. That's probably why it didn't show up in the dmesg.log. It's showing up now after I turn the controller on and connect it.

[dmesg.log](https://github.com/user-attachments/files/20889218/dmesg.log)