atar-axis/xpadneo #549: XBE2 recognized as 360 controller in Steam and everywhere else

Epicness7131, 2025-07-01:
I am on Fedora 42, the driver works amazing and fully works, however, there is one issue, the controller is universally recognized a xbox 360 controller rather then the XBE2, meaning the back paddles are not properly supported or shown when testing/using. When xpadneo is uninstalled the controller is recognized as XBE2 again, meaning this is a issue/quirk with the driver.  This is my only issue with the driver and everything else works fine, controller rumbles when connected battery is shown etc. If the controller is intended to be recognized as a 360 controller it should probably be made more clear in the documentation.

[image]

[image]

```
[ 2599.671255] xpadneo 0005:045E:0B22.000A: input,hidraw6: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on 34:6f:24:da:1f:00
[ 2599.671260] xpadneo 0005:045E:0B22.000A: controller quirks: 0x00000050
[ 2599.671262] xpadneo xpadneo_welcome_rumble start
[ 2599.671264] xpadneo 0005:045E:0B22.000A: testing weak motor: sustain 50ms release 50ms loop 2 wait 30ms
[ 2600.001520] xpadneo 0005:045E:0B22.000A: testing strong motor: sustain 50ms release 50ms loop 2 wait 30ms
[ 2600.331764] xpadneo 0005:045E:0B22.000A: testing trigger motors: sustain 50ms release 50ms loop 2 wait 30ms
[ 2600.662007] xpadneo xpadneo_welcome_rumble took 991ms
[ 2600.662014] xpadneo 0005:045E:0B22.000A: Xbox Wireless Controller [f4:6a:d7:44:67:cf] connected
[ 2600.834553] xpadneo 0005:045E:0B22.000A: mapping profiles detected
[ 2600.834558] xpadneo 0005:045E:0B22.000A: switching profile to 1
[ 2824.121217] xpadneo 0005:045E:0B22.000A: switching profile to 2
[ 2825.599214] xpadneo 0005:045E:0B22.000A: switching profile to 3
[ 2825.921472] xpadneo 0005:045E:0B22.000A: switching profile to 0
[ 2844.525495] xpadneo 0005:045E:0B22.000A: reverting to original version (changed version from 0x00001130 to 0x00000523)
[ 2844.525501] xpadneo 0005:045E:0B22.000A: reverting to original product (changed PID from 0x028E to 0x0B22)
[ 2848.690753] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B22.000B/input/input46
[ 2848.691064] microsoft 0005:045E:0B22.000B: input,hidraw6: BLUETOOTH HID v5.23 Gamepad [Xbox Wireless Controller] on 34:6f:24:da:1f:00
[ 2848.774746] xpadneo 0005:045E:0B22.000B: BLE firmware version 5.23
[ 2848.774751] xpadneo 0005:045E:0B22.000B: pretending XB1S Windows wireless mode (changed PID from 0x0B22 to 0x028E)
[ 2848.774753] xpadneo 0005:045E:0B22.000B: working around wrong SDL2 mappings (changed version from 0x00000523 to 0x00001130)
[ 2848.774756] xpadneo 0005:045E:0B22.000B: report descriptor size: 464 bytes
[ 2848.774758] xpadneo 0005:045E:0B22.000B: fixing up Rx axis
[ 2848.774760] xpadneo 0005:045E:0B22.000B: fixing up Ry axis
[ 2848.774762] xpadneo 0005:045E:0B22.000B: fixing up Z axis
[ 2848.774763] xpadneo 0005:045E:0B22.000B: fixing up Rz axis
[ 2848.774765] xpadneo 0005:045E:0B22.000B: fixing up button mapping
[ 2848.774871] xpadneo 0005:045E:0B22.000B: gamepad detected
[ 2848.774873] xpadneo 0005:045E:0B22.000B: enabling compliance with Linux Gamepad Specification
[ 2848.774926] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B22.000B/input/input47
[ 2848.775010] xpadneo 0005:045E:0B22.000B: consumer control detected
[ 2848.775045] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B22.000B/input/input48
[ 2848.775082] xpadneo 0005:045E:0B22.000B: keyboard detected
[ 2848.775117] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B22.000B/input/input49
[ 2848.775195] xpadneo 0005:045E:0B22.000B: input,hidraw7: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on 34:6f:24:da:1f:00

```

Terrorwolf01, 2026-09-10:
Hi,
I have the same issue. Any update or Timeline on this?
Thanks in Advance

kakra, 2026-09-15:
The share button is exposed as a keyboard device, the guide button is exposed as a consumer control device, see `sudo libinput list-devices`. For the share button, there isn't even an equivalent button in the gamepad bitmap in Linux, so it would either show up as a wrong button, or be misaligned among other buttons. This is by design.

Currently, the share buttons send `F12` hard-coded. xpadneo will make this configurable in a future version.

Showing as "Xbox 360" is a cosmetic issue, all buttons should work, and SDL still recognizes the paddle buttons. If you use Steam Input, the paddles may not work. Please note that we explicitly do not recommend using xpadneo with Steam Input currently. If you want to use Steam Input, rather stick with the built-in kernel driver using hidraw.

orbea, 2026-10-11:
@kakra Fwiw, there are some games that reportedly only have gamepad input with steam input (Romancing Saga Remastered) and as I am sure you are aware the in kernel driver is unreliable... I'm looking forward to this being resolved.