atar-axis/xpadneo #552: Make issue when building for new kernel

Wheelz108, 2025-08-04:
I'm getting the following issue when trying to build the module for a kernel upgrade.

```
DKMS (dkms-3.2.1) make.log for hid-xpadneo-v0.9-226/ga16acb0 for kernel 6.15.9-arch1-1 (x86_64)
Mon 04 Aug 2025 15:25:22 AEST

Building module(s)
# command: make -j16 KERNELRELEASE=6.15.9-arch1-1 -C '/usr/lib/modules/6.15.9-arch1-1/build' M='/var/lib/dkms/hid-xpadneo/v0.9-226-ga16acb0/build/src' VERSION='v0.9-226-ga16acb0' modules
make: Entering directory '/usr/lib/modules/6.15.9-arch1-1/build'
Makefile:199: *** specified external module directory "/var/lib/dkms/hid-xpadneo/v0.9-226-ga16acb0/build/src" does not exist.  Stop.
make: Leaving directory '/usr/lib/modules/6.15.9-arch1-1/build'

# exit code: 2
# elapsed time: 00:00:00
----------------------------------------------------------------
```

The directory it wants does indeed not exist on my machine, but installing after a restart (i.e `sudo modprobe uhid; ./install.sh`) works totally fine. 

Not sure how I would go about fixing this properly and submitting a pull request for said fix, so any help is greatly appreciated. Thanks!

kakra, 2025-08-04:
You probably need to install the needed kernel headers.

kakra, 2025-08-05:
Okay, maybe `/var/lib/dkms/hid-xpadneo/v0.9-226-ga16acb0/build/src` contains a stale symlink? Could you try re-installing xpadneo?

If this fixes it, that problem shouldn't happen in the first place. I'm pretty sure dkms should manage such a symlink.

Wheelz108, 2025-08-05:
Reinstalling via `sudo ./install.sh` does work as expected:

```
sudo ./install.sh
* deploying DKMS package
Deprecated feature: CLEAN (/usr/src/hid-xpadneo-v0.9-226-ga16acb0/dkms.conf)
Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-226-ga16acb0/source -> /usr/src/hid-xpadneo-v0.9-226-ga16acb0
* building module
Deprecated feature: CLEAN (/var/lib/dkms/hid-xpadneo/v0.9-226-ga16acb0/source/dkms.conf)
Sign command: /usr/lib/modules/6.15.9-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module(s)... done.
Signing module /var/lib/dkms/hid-xpadneo/v0.9-226-ga16acb0/build/src/hid-xpadneo.ko
* installing module
Deprecated feature: CLEAN (/var/lib/dkms/hid-xpadneo/v0.9-226-ga16acb0/source/dkms.conf)
Installing /usr/lib/modules/6.15.9-arch1-1/updates/dkms/hid-xpadneo.ko.zst
Running the post_install script:
Not disabling ERTM, kernel version doesn't require it...
Installing modalias database...
Installing udev rules...
Reloading udev...
Running depmod... done.
```

But reinstalling kernel headers still produces the same error:

```
sudo pacman -S dkms linux-headers bluez bluez-utils
warning: dkms-3.2.1-1 is up to date -- reinstalling
warning: linux-headers-6.15.9.arch1-1 is up to date -- reinstalling
warning: bluez-5.83-1 is up to date -- reinstalling
warning: bluez-utils-5.83-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (4) bluez-5.83-1  bluez-utils-5.83-1  dkms-3.2.1-1  linux-headers-6.15.9.arch1-1

Total Installed Size:  244.86 MiB
Net Upgrade Size:        0.00 MiB

:: Proceed with installation? [Y/n] 
(4/4) checking keys in keyring                                                                                  [###################################################################] 100%
(4/4) checking package integrity                                                                                [###################################################################] 100%
(4/4) loading package files                                                                                     [###################################################################] 100%
(4/4) checking for file conflicts                                                                               [###################################################################] 100%
(4/4) checking available disk space                                                                             [###################################################################] 100%
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
==> dkms remove --no-depmod hid-xpadneo/v0.9-226-ga16acb0 -k 6.15.9-arch1-1
:: Processing package changes...
(1/4) reinstalling dkms                                                                                         [###################################################################] 100%
(2/4) reinstalling linux-headers                                                                                [###################################################################] 100%
(3/4) reinstalling bluez                                                                                        [###################################################################] 100%
(4/4) reinstalling bluez-utils                                                                                  [###################################################################] 100%
:: Running post-transaction hooks...
(1/7) Reloading system manager configuration...
(2/7) Reloading user manager configuration...
(3/7) Restarting marked services...
(4/7) Arming ConditionNeedsUpdate...
(5/7) Updating module dependencies...
(6/7) Install DKMS modules
==> dkms install --no-depmod hid-xpadneo-v0.9-226/ga16acb0 -k 6.15.9-arch1-1
Deprecated feature: CLEAN (/var/lib/dkms/hid-xpadneo-v0.9-226/ga16acb0/source/dkms.conf)

Error! Bad return status for module build on kernel: 6.15.9-arch1-1 (x86_64)
Consult /var/lib/dkms/hid-xpadneo-v0.9-226/ga16acb0/build/make.log for more information.
==> WARNING: `dkms install --no-depmod hid-xpadneo-v0.9-226/ga16acb0 -k 6.15.9-arch1-1' exited 10
(7/7) Reloading system bus configuration...
```

`/var/lib/dkms/hid-xpadneo-v0.9-226/ga16acb0/build/make.log` for posterity:

```
DKMS (dkms-3.2.1) make.log for hid-xpadneo-v0.9-226/ga16acb0 for kernel 6.15.9-arch1-1 (x86_64)
Tue 05 Aug 2025 11:39:16 AEST

Building module(s)
# command: make -j16 KERNELRELEASE=6.15.9-arch1-1 -C '/usr/lib/modules/6.15.9-arch1-1/build' M='/var/lib/dkms/hid-xpadneo/v0.9-226-ga16acb0/build/src' VERSION='v0.9-226-ga16acb0' modules
make: Entering directory '/usr/lib/modules/6.15.9-arch1-1/build'
Makefile:199: *** specified external module directory "/var/lib/dkms/hid-xpadneo/v0.9-226-ga16acb0/build/src" does not exist.  Stop.
make: Leaving directory '/usr/lib/modules/6.15.9-arch1-1/build'

# exit code: 2
# elapsed time: 00:00:00
----------------------------------------------------------------
```

kakra, 2025-08-05:
Okay thanks. So the Arch trigger for reinstalling DKMS modules is different to what we do. I think this is enough context to try and reproduce this issue. Thank you very much.

Wheelz108, 2025-08-05:
Do let me know if you need more information from me at any point

chomskee17, 2025-08-15:
@Wheelz108 Did you just roll the kernel back for a workaround for this?

Wheelz108, 2025-08-16:
My workaround right now is manually reinstalling with the dkms module install script after a restart. The driver does work with new kernels, I just need to manually reinstall it rather than having it done as part of the upgrade process.

HectorDVR, 2026-09-13:
Reproduced on CachyOS with `dkms 3.4.3`, kernel `7.2.4-3-cachyos-bore`, driver `v0.11-pre-64-g9aa168e`. Same signature as reported here.

Stock pacman hook log:

```
==> dkms install --no-depmod hid-xpadneo-v0.11-pre-64/g9aa168e -k 7.2.4-3-cachyos-bore
Error! Bad return status for module build on kernel: 7.2.4-3-cachyos-bore (x86_64)
==> WARNING: `dkms install --no-depmod hid-xpadneo-v0.11-pre-64/g9aa168e -k 7.2.4-3-cachyos-bore' exited 10
```

`make.log`:

```
# command: make ... M='/var/lib/dkms/hid-xpadneo/v0.11-pre-64-g9aa168e/build/src' VERSION='v0.11-pre-64-g9aa168e' modules
Makefile:199: *** specified external module directory "/var/lib/dkms/hid-xpadneo/v0.11-pre-64/g9aa168e/build/src" does not exist.
```

Root cause (not headers, not stale symlink): `/usr/share/libalpm/scripts/dkms` parses the module as `([^/]+)-([^/]+)` over `/usr/src/hid-xpadneo-VERSION` in `all_nv_from_kver()` and in `main()`'s stdin parsing. `[^/]+` is greedy, so it splits at the LAST dash: `-m hid-xpadneo-v0.11-pre-64 -v g9aa168e`. Note the `M=` path in the log is even the *correct* one — the failure is the bogus tree path the hook registered. This is also Arch Flyspray FS#73194 (2021, still reproducing on dkms 3.4.3). Note the non-greedy fix proposed there is insufficient: with dashes in BOTH name (`hid-xpadneo`) and version (`v0.11-pre-64-g9aa168e`), neither first-dash nor last-dash split is correct. The only unambiguous source is `PACKAGE_NAME`/`PACKAGE_VERSION` from `dkms.conf`.

Workaround that holds on every kernel update: a `PostTransaction` hook running after `70-dkms-install` that reads `PACKAGE_NAME`/`PACKAGE_VERSION` via `grep` (never `source`, `dkms.conf` breaks under `set -u`) and calls `dkms install -m "$NAME" -v "$VER" -k "$kver"` explicitly, then removes the ghost `hid-xpadneo-v0.11-pre-64/g9aa168e: added` registration the stock hook leaves behind. Functional result `installed` on all kernels, `dkms status` stays clean.

Happy to test any candidate fix on this setup.

kakra, 2026-09-15:
This isn't a problem with our installer, it must be a problem in the distribution package:

```make
# LANG=C make install
{ [ -n "" ] && echo  || git describe --tags --dirty; } >VERSION
/usr/bin/make VERSION="v0.11-pre-70-gaad3597" -C hid-xpadneo dkms.conf
make[1]: Entering directory '/home/kakra/devel/git/src/github.com/kakra/xpadneo/hid-xpadneo'
sed 's/"@DO_NOT_CHANGE@"/"v0.11-pre-70-gaad3597"/g' <"dkms.conf.in" >"dkms.conf"
make[1]: Leaving directory '/home/kakra/devel/git/src/github.com/kakra/xpadneo/hid-xpadneo'
mkdir -p /etc/modprobe.d /etc/udev/rules.d /usr/share/doc/xpadneo
install -D -m 0644 -t /etc/modprobe.d hid-xpadneo/etc-modprobe.d/xpadneo.conf
install -D -m 0644 -t /etc/udev/rules.d hid-xpadneo/etc-udev-rules.d/60-xpadneo.rules hid-xpadneo/etc-udev-rules.d/70-xpadneo-disable-hidraw.rules
install -D -m 0644 -t /usr/share/doc/xpadneo NEWS.md docs/3P-BUGS.md docs/BT_DONGLES.md docs/CONFIGURATION.md docs/DEBUGGING.md docs/FIRMWARE_FLASHING.md docs/PACKAGING.md docs/README.md docs/SDL.md docs/TESTING.md docs/TROUBLESHOOTING.md
udevadm control --reload
dkms add hid-xpadneo
Creating symlink /var/lib/dkms/hid-xpadneo/v0.11-pre-70-gaad3597/source -> /usr/src/hid-xpadneo-v0.11-pre-70-gaad3597
rm VERSION

# dkms status
hid-xpadneo/v0.11-pre-70-gaad3597: added
```

Observe how it is supposed to split name and version with `/`, not `-` - and out installer *does* handle that correctly.

@HectorDVR basically confirms this, the problem is in `/usr/share/libalpm/scripts/dkms` which doesn't belong to xpadneo.

If the same problem comes up during boot or kernel installs, the problem is likely in the package tooling of the distribution. It should never use naive intermediate representations of `$PACKAGE-$VERSION` for dkms modules, it should always use `$PACKAGE/$VERSION`, so it never needs to split at the dash.

I think we can degrade this to a third-party bug now. Thanks, @HectorDVR, for the thorough explanation.

closed 2026-10-18 by kakra