atar-axis/xpadneo #558: Xbox One Wireless Controller with firmware 5.x not detected as input device (Bluetooth)

OrazioGrinzosiReal, 2025-09-06:
On Linux Mint 22.2 with kernel 6.8.0-79 and xpadneo v0.9-226, my Xbox Wireless Controller (Modalias: usb:v045Ep02FDd0903) connects via Bluetooth (bluetoothctl shows "Connected: yes"), but no input devices (/dev/input/js0 or /dev/input/event*) are created.

I’ve confirmed uhid is loaded, and xpadneo is correctly installed via DKMS. The controller works fine when connected via USB.

Is this firmware version (d0903) currently supported by xpadneo? Thanks in advance to anyone capable of helping.

kakra, 2025-09-07:
If uhid is loaded and it still doesn't work, bluez may ship with uhid disabled. If it still doesn't work, the controller probably doesn't fully handshake, try a different Bluetooth chipset: see our docs directory.

KyodaiKen, 2025-12-25:
I got the same issue, I didn't update the firmware on my controller and all of a sudden no input it recognized anymore when connected via BT. Via USB it works without any issue.

It seems like the pad is not being listed by libinput anymore. I am on Fedora 43 KDE.